Klipper3d/klipper #2503: Is it possible automatically disable the hotend fan during probing?

bjoern79de, 2020-02-14:
I have configured the hottend fan to be triggered by temperature, but the vibrations of the fan has a big impact to the repeatability of probing measures. Is there a possibility to turn the fan automatically off for each probing action?

KevinOConnor, 2020-02-14:
Currently, the only way to turn off the fan during probing is to issue a fan off command prior to probing.  This can be done manually or via various macros.

-Kevin

bjoern79de, 2020-02-14:
I've tried it using a macro and M107, but just the cooling-fan turned off - not the hottend-fan.
I guess, I would have to also remove the temperature trigger from the hotted-fan, but how to address both fans separately by gcode?

KevinOConnor, 2020-02-14:
There is currently no way to disable a fan configured as a heater_fan at runtime - except by disabling the heater and waiting for it to cool down below the configured heater_temp.

-Kevin

Hywelmartin, 2020-02-17:
Ad a relay that turn of the fan during probing

ld3300, 2020-02-18:
Would this be a kind of extension of #2486. It is a similar concept in controlling what is enabled on the machine during a probing move?

bjoern79de, 2020-02-18:
> Would this be a kind of extension of #2486. It is a similar concept in controlling what is enabled on the machine during a probing move?

Sounds good to me.

FanDjango, 2020-02-21:
> Would this be a kind of extension of #2486. It is a similar concept in controlling what is enabled on the machine during a probing move?

Noted.

My personal solution for this up to now was as follows:

I set the trigger temperature of my nozzle fan to say, 180°. I probe at 175°, so the fan is off. The few degrees lower than my printing temperatures of (example) 190° don't bother me for probe accuracy and it is a constant error anyway so it can be compensated. When the probe finishes, nozzle temp is increased to 190° and the fan starts because of that increase.

---------------

On the other hand:

I have investigated having a "[homing_fans]" section/module **similar** to "[homing_heaters]". Here are my findings and @KevinOConnor or others need to comment on this:

**Is it needed at all?** , because:

- It would need a `send_event` from `probe_multi_begin` and `..end` instead of `homing_move_begins` etc. There is no sense in turning the fan off for each single probe action, but rather as a bracket AROUND the probe actions, similarly to the logic for BL-Touch pin-ups and -downs in the "don't-stow-in-between-probes" mode.
- There would need to be a (configurable) sleep time after turning off the fan before vibrations die down.

My current tentative code attempt, which identifies the printers fan objects and so on is kludgy, turning off and restoring the fans states is also not so easy. If desired I would submit a preliminary PR for this and let everyone chime in.

KevinOConnor, 2020-02-24:
FWIW, it's fine if someone wishes to add this capability to the existing homing_heaters or to a new module.  It does seem a bit daring to me to automatically disable a nozzle cooling fan though..

-Kevin

bjoern79de, 2020-02-24:
> FWIW, it's fine if someone wishes to add this capability to the existing homing_heaters or to a new module. It does seem a bit daring to me to automatically disable a nozzle cooling fan though..
> 
> -Kevin

That's true. Maybe it's a better idea to replace the fan and use one which runs smoother.

KevinOConnor, 2020-03-21:
I'm going to close this as it looks like the original question was answered (there is no automatic way to turn off fans during homing).  A developer would need to implement it if desired.

-Kevin

closed 2020-03-21 by KevinOConnor