Klipper3d/klipper #2505: [FEATURE REQUEST] virtual pin for filament_switch_sensor

test3210-d, 2020-02-15:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4207834/klippy.log)
I whant add virtual runout sensor in filament width sensor for filament_switch_sensor. If diameter out of range change virtual pin state (like probe:z virtual endstop) . This feature possible to implementation in tsl1401 and hall based sensors. I don't know how implement virtual pin. May be somebody can help ?

Arksine, 2020-02-15:
While this is possible, I think the better option here would be for the filament width sensor to inherit the "BaseSensor" class in filament_switch_sensor.py.  Then when the width sensor detects a runout it can execute the proper callback.

test3210-d, 2020-02-15:
harder to configure by user , all option from filament_switch_sensor migrate to filament width section ? or not ?

Arksine, 2020-02-15:
To me it makes sense that all configuration for a filament sensor be in the same section.

KevinOConnor, 2020-02-15:
I agree that doing an `import filament_switch_sensor` and then directly using that code is preferable to virtual pins.

@Arksine - what do you think about converting the filament_switch_sensor code from a "base class" to a "helper class"?  That might make it easier to use in the hall and tsl1401 classes.

That said, if you are interested in how virtual pins are implemented, there is an example in klippy/extras/probe.py - see the use of the `setup_pin()` method.

-Kevin

KevinOConnor, 2020-02-15:
> what do you think about converting the filament_switch_sensor code from a "base class" to a "helper class"?

Specifically, I was thinking perhaps the `BaseSensor(object)` class could be converted to a `RunoutHelper` class, and SwitchSensor could be converted to something like:
```python
class SwitchSensor:
    def __init__(self, config):
        printer = config.get_printer()
        buttons = printer.try_load_module(config, 'buttons')
        switch_pin = config.get('switch_pin')
        buttons.register_buttons([switch_pin], self._button_handler)
        self.runout_helper = RunoutHelper(config)
    def _button_handler(self, eventtime, state):
        self.runout_helper.note_filament_present(state)
```

and then the RunoutHelper class could do all the work of discarding duplicate notifications, disabling events at printer startup, disabling events shortly after a previous event, disabling events on a SET_FILAMENT_SENSOR command, reporting status via GET_FILAMENT_SENSOR, etc.

That way, the hall and tsl1401 classes could just import this helper class and periodically call its `note_filament_present()` method.

Thoughts?
-Kevin

test3210-d, 2020-02-15:
Ok, but what about 
```
        switch_pin = config.get('switch_pin')
        buttons.register_buttons([switch_pin], self._button_handler)
```
i still must set some pin, as i think, or no ?

KevinOConnor, 2020-02-15:
> i still must set some pin, as i think, or no ?

My comments above were directed at @Arksine.

I'm proposing that the hall_filament_sensor.py code would add:
```
import filament_switch_sensor
...
self.runout_helper = filament_switch_sensor.RunoutHelper(config)
```
and then periodically call:
```
self.runout_helper.note_filament_present(self.lastFilamentWidthReading > 0.5)
```

It is just a proposal though, as it would require code changes to filament_switch_sensor.py.

-Kevin

Arksine, 2020-02-15:
@KevinOConnor I think its a good idea.  It would definitely reduce duplicate code.  I'll get working on it.

KevinOConnor, 2020-02-17:
FYI, the mentioned changes to filament_switch_sensor.py in https://github.com/KevinOConnor/klipper/issues/2505#issuecomment-586618603 were completed (commit 718da759) and thus it should now be possible to implement the hall_filament_sensor.py changes mentioned in https://github.com/KevinOConnor/klipper/issues/2505#issuecomment-586622668 .

-Kevin

closed 2020-02-18 by test3210-d