Klipper3d/klipper #2508: [FEATURE REQUEST] Reload macro without restarting klipper

kakou-fr, 2020-02-15:
Hi,

This is just a little FR.
Could it be possible to reload only the MACROS part of the configuration without restarting klipper/HOST ?

When i want to modify the behavior of a MACRO (when testing  multi-extruder gcode), I need stop cancel the print, reload HOST, and the restart the print.

Hywelmartin, 2020-02-16:
You can have parameters in macros.. if that helps

like 

https://github.com/KevinOConnor/klipper/blob/086c131bbfc28b7891b936b4630ffc0f34ca6e2c/config/sample-macros.cfg#L103

ld3300, 2020-02-16:
I use parameters and variables a lot when I am doing tests like that.

kakou-fr, 2020-02-17:
yes i use it also for some part, but other parts need to add or remove gcode.

KevinOConnor, 2020-02-24:
FWIW, my high-level feedback is that trying to reload part of the config file (instead of all of the config file as is done now on a RESTART command) seems to be a lot of work with not that much benefit.  Certainly, if someone wanted to do that they'd need to do the development for it and put together a proposal.

Cheers,
-Kevin

JohnEdwa, 2020-02-26:
Obviously there is no reason to do so if it still restarts, but he ability to just reload the config without restart would be a great, as the restart wipes the current unsaved bed mesh, cancels all heaters, and reboots the MCU firmware - quite annoying if you just want to tweak a macro or change your E-steps.

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like the conversation has concluded.

-Kevin

closed 2020-03-21 by KevinOConnor