Klipper3d/klipper #2509: Probe accuracy: always higher z?

rlesac, 2020-02-15:
Specs: Ender3 frame, SKR v1.3, genuine BLTouch v3.0, TMC2208 for xyz.

When I run the PROBE_ACCURACY, it almost always returns a higher z height.
Is that normal or expected?
If not normal, what could be the underlying issues?
[klippy.log](https://github.com/KevinOConnor/klipper/files/4208385/klippy.log)

Here are three consecutive runs:
 
PROBE_ACCURACY at X:100.000 Y:100.000 Z:12.230 (samples=10 retract=2.000 speed=0.5 lift_speed=0.5)
probe at 100.000,100.000 is z=2.362500
probe at 100.000,100.000 is z=2.362500
probe at 100.000,100.000 is z=2.365000
probe at 100.000,100.000 is z=2.365000
probe at 100.000,100.000 is z=2.367500
probe at 100.000,100.000 is z=2.367500
probe at 100.000,100.000 is z=2.370000
probe at 100.000,100.000 is z=2.367500
probe at 100.000,100.000 is z=2.370000
probe at 100.000,100.000 is z=2.370000
probe accuracy results: maximum 2.370000, minimum 2.362500, range 0.007500, average 2.366750, median 2.367500, standard deviation 0.002750

PROBE_ACCURACY at X:100.000 Y:100.000 Z:4.370 (samples=10 retract=2.000 speed=0.5 lift_speed=0.5)

probe at 100.000,100.000 is z=2.370000
probe at 100.000,100.000 is z=2.372500
probe at 100.000,100.000 is z=2.375000
probe at 100.000,100.000 is z=2.375000
probe at 100.000,100.000 is z=2.372500
probe at 100.000,100.000 is z=2.375000
probe at 100.000,100.000 is z=2.375000
probe at 100.000,100.000 is z=2.377500
probe at 100.000,100.000 is z=2.375000
probe at 100.000,100.000 is z=2.377500
probe accuracy results: maximum 2.377500, minimum 2.370000, range 0.007500, average 2.374500, median 2.375000, standard deviation 0.002179

PROBE_ACCURACY at X:100.000 Y:100.000 Z:4.378 (samples=10 retract=2.000 speed=0.5 lift_speed=0.5)

probe at 100.000,100.000 is z=2.380000
probe at 100.000,100.000 is z=2.385000
probe at 100.000,100.000 is z=2.382500
probe at 100.000,100.000 is z=2.380000
probe at 100.000,100.000 is z=2.380000
probe at 100.000,100.000 is z=2.382500
probe at 100.000,100.000 is z=2.382500
probe at 100.000,100.000 is z=2.385000
probe at 100.000,100.000 is z=2.385000
probe at 100.000,100.000 is z=2.385000
probe accuracy results: maximum 2.385000, minimum 2.380000, range 0.005000, average 2.382750, median 2.382500, standard deviation 0.002077

rlesac, 2020-02-16:
Reverted back to older Klipper, the results are a bit different, much better accuracy:
```
PROBE_ACCURACY at X:100.000 Y:100.000 Z:12.200 (samples=10 retract=2.000 speed=0.5

probe at 100.000,100.000 is z=2.352500
probe at 100.000,100.000 is z=2.350000
probe at 100.000,100.000 is z=2.352500
probe at 100.000,100.000 is z=2.352500
probe at 100.000,100.000 is z=2.355000
probe at 100.000,100.000 is z=2.355000
probe at 100.000,100.000 is z=2.355000
probe at 100.000,100.000 is z=2.355000
probe at 100.000,100.000 is z=2.355000
probe at 100.000,100.000 is z=2.355000
probe accuracy results: maximum 2.355000, minimum 2.350000, average 2.353750, median 2.355000, standard deviation 0.001677
PROBE_ACCURACY at X:100.000 Y:100.000 Z:4.355 (samples=10 retract=2.000 speed=0.5

probe at 100.000,100.000 is z=2.357500
probe at 100.000,100.000 is z=2.357500
probe at 100.000,100.000 is z=2.357500
probe at 100.000,100.000 is z=2.360000
probe at 100.000,100.000 is z=2.360000
probe at 100.000,100.000 is z=2.360000
probe at 100.000,100.000 is z=2.362500
probe at 100.000,100.000 is z=2.360000
probe at 100.000,100.000 is z=2.360000
probe at 100.000,100.000 is z=2.362500
probe accuracy results: maximum 2.362500, minimum 2.357500, average 2.359750, median 2.360000, standard deviation 0.001750
PROBE_ACCURACY at X:100.000 Y:100.000 Z:4.362 (samples=10 retract=2.000 speed=0.5

probe at 100.000,100.000 is z=2.362500
probe at 100.000,100.000 is z=2.362500
probe at 100.000,100.000 is z=2.362500
probe at 100.000,100.000 is z=2.365000
probe at 100.000,100.000 is z=2.365000
probe at 100.000,100.000 is z=2.365000
probe at 100.000,100.000 is z=2.365000
probe at 100.000,100.000 is z=2.365000
probe at 100.000,100.000 is z=2.365000
probe at 100.000,100.000 is z=2.367500
probe accuracy results: maximum 2.367500, minimum 2.362500, average 2.364500, median 2.365000, standard deviation 0.001500
```

KevinOConnor, 2020-02-24:
FWIW, as far as I know the code works as intended.  If the probe isn't reporting accurate values my best guess is that the probe isn't particularly accurate, or perhaps the Z axis is loose and/or has backlash.

-Kevin

rlesac, 2020-02-24:
I get you, though I only changed the Klipper version and got a bit different results, that were repeatable. 
With the latest Klipper, the standard deviation is always higher.

KevinOConnor, 2020-02-24:
I can't think of a change that would lead to what you are reporting.  What is the revision of the "older Klipper"?  If you have time, using `git bisect` to determine which revision caused the regression would help greatly.

-Kevin

FanDjango, 2020-02-28:
"Old klipper" seems at least prior to the "lift_speed" commit, as it does not have that value in the printout. But after the "average" commit. That narrows things down. Since I can confirm @rlesac 's report (see below), I will on the weekend do some _bisecting_

In my discussions with @Coffee0297 re. 2509, (I still have the discord transcript) the other day, we also saw this (to a degree). The increase **does not** happen on a "PROBE_ACCURACY" run with many samples, like 100. It happens from **one run to the next**, even if those only have 10 samples.

So I have tried this with a current klipper, 6 times PROBE ACCURACY directly one behind the other: (please look at the progression of the averages):
```
Recv: // PROBE_ACCURACY at X:69.500 Y:98.500 Z:4.996 (samples=10 retract=2.500 speed=5.0 lift_speed=30.0)
```
```
Recv: // probe accuracy results: maximum 2.462500, minimum 2.437500, range 0.025000, average 2.453875, median 2.455625, standard deviation 0.007190
Recv: // probe accuracy results: maximum 2.475000, minimum 2.446250, range 0.028750, average 2.461875, median 2.461250, standard deviation 0.008296
Recv: // probe accuracy results: maximum 2.485000, minimum 2.460000, range 0.025000, average 2.470125, median 2.468750, standard deviation 0.008412
Recv: // probe accuracy results: maximum 2.492500, minimum 2.465000, range 0.027500, average 2.480125, median 2.482500, standard deviation 0.007468
Recv: // probe accuracy results: maximum 2.506250, minimum 2.480000, range 0.026250, average 2.488875, median 2.488125, standard deviation 0.006992
Recv: // probe accuracy results: maximum 2.502500, minimum 2.490000, range 0.012500, average 2.496500, median 2.495625, standard deviation 0.004500
```
Performing a Z home resets this sequence of increasing averages.

Interesting: Setting "stow_on_each_sample" to "False" makes the sequence of averages increase **faster** by about a factor of 5. But to be compatible with the older klipper, I kept that parm at its default of True.

In the code, I can see that the probe is raised to a fluctuating height ( a constant added to the prior probe depth, which varies) instead of to the original starting height and that is where the probe is left after the sequence - and from that height, the next sequence is started. Could there be a cumulative rounding error coming in somewhere that propagates from one run to the next until a home is done to get back to the original Z pos?

FanDjango, 2020-02-28:
Preliminary result: 30 Sep 2019 db2ee2c01aaaea6207a520a43b1ebb8ec991ea83 fails like above.
Just a quick check with only klippy. So therefore, full bisect with firmware flash on each step, sigh.

KevinOConnor, 2020-02-28:
@FanDjango - do you have a log of the failure?  (Any log will do - I'd like to see the config and host/firmware types/versions.)

-Kevin

FanDjango, 2020-02-29:
@KevinOConnor  - certainly: here goes (part1) a log for the current klipper, fresh flash and klippy
[klippy.log](https://github.com/KevinOConnor/klipper/files/4270185/klippy.log)
[octoprint.txt](https://github.com/KevinOConnor/klipper/files/4270186/octoprint.txt)

FanDjango, 2020-02-29:
And here is one such pair for klipper at 29. October 2019 - commit db2ee2c01aaaea6207a520a43b1ebb8ec991ea83, that is a day or so after the "range" commit:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4270205/klippy.log)
[octoprint.txt](https://github.com/KevinOConnor/klipper/files/4270206/octoprint.txt)
Even with that old version, the averages are increasing.

FanDjango, 2020-02-29:
To be very sure, I radically reduced the values for top z speed and accelleration in `printer.cfg`:
```
[printer]
kinematics: cartesian
max_velocity: 200
max_accel: 1000
max_accel_to_decel: 1000
max_z_velocity: 10
max_z_accel: 10
square_corner_velocity: 5

```
I could see (and hear) the downward move of the probe accellerating smoothly after the pin-down, to an abrupt stop (of course) on trigger. After the pin-up, smooth accelleration upwards and decelleration to a stop. Then repeat.

One strage thing - occasionally at movement begin, there is a slight knocking noise from the steppers but not every time. The occurence of this knocking noise does not correllate at all with the z measurements being shown in the terminal.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4270219/klippy.log)
[octoprint.txt](https://github.com/KevinOConnor/klipper/files/4270220/octoprint.txt)

FanDjango, 2020-02-29:
> "Old klipper" seems at least prior to the "lift_speed" commit (early Feb 2020, c36fe76), as it does not have that value in the printout. But after the "range" commit (29 Sep 2019 0c247e5). That narrows things down. Since I can confirm @rlesac 's report (see below), I will on the weekend do some bisecting

This is incorrect. The "old klipper" output **has not** got "range". Need to go back further than 2019 Sep 29). So I would like to know how far back to go... could @rlesac say which version is his "old version"?

FanDjango, 2020-02-29:
I think I have found a possible explanation:

@KevinOConnor Please look at `homing.py` as coded in `def homing_move:` for `probe_pos=True` and note the ominous "0.48" in there:
```
        if probe_pos:
            for s, name, spos, epos in end_mcu_pos:
                md = (epos - 0.48 - spos) * s.get_step_dist()
                s.set_tag_position(s.get_tag_position() + md)
            self.set_homed_position(kin.calc_tag_position())
        else:
            self.toolhead.set_position(movepos)
```
Using a statistics approach I have found that (on average over many probe operations) there is a 0.48 chance that **one** step too few will be counted on probe endstop trigger, i.e. `epos` should be one step further to the minus-position direction about half the times.

Since we pick up the stop position by backward calculation from the step count, we slowly creep away from the real position. 

I have empirically determined this value **and it is only valid for my machine**. I think a lower bound for this value would be **zero**, but I know of no plausible upper bound. In my case this means a creep of 100 * 0.48 * 0.00125 on one hundred probes = ca. 0.06mm (which can be seen on the 100 probe sequence one post earlier).

Not a problem if you home once in a while and do not probe too numerously without homing in between.

I think the abrupt stop on endstop trigger is the problem, which of course does not matter if it is a homing operation, but on (repeated) probing, it does.

rlesac, 2020-02-29:
> 
> 
> Updates after some more testing:
> 
> > "Old klipper" seems at least prior to the "lift_speed" commit (early Feb 2020, [c36fe76](https://github.com/KevinOConnor/klipper/commit/c36fe76d692754d0244059a729e147c688c96062)), as it does not have that value in the printout. But after the "range" commit (29 Sep 2019 [0c247e5](https://github.com/KevinOConnor/klipper/commit/0c247e55e6ec62664d28f1274c553919ce870983)). That narrows things down. Since I can confirm @rlesac 's report (see below), I will on the weekend do some bisecting
> 
> **This is incorrect**. The "old klipper" output **has not** got "range". Need to go back further than 2019 Sep 29). So I would like to know how far back to go... could @rlesac say which version is his "old version"?
> ```
Last commit on my "old" version is https://github.com/KevinOConnor/klipper/commit/626cb71428f9160a697ebcad7b4e42682882d644 from Sep 18, 2019.

Sorry for the late reply, hope it helps!

KevinOConnor, 2020-02-29:
If you run GET_POSITION before and after each PROBE_ACCURACY command, what does it report?

-Kevin

FanDjango, 2020-02-29:
```
Send: GET_POSITION
Recv: // mcu: stepper_x:-14546 stepper_y:-16318 stepper_z:290 stepper_z1:290
Recv: // stepper: stepper_x:69.500000 stepper_y:98.500000 stepper_z:4.996250 stepper_z1:4.996250
Recv: // kinematic: X:69.500000 Y:98.500000 Z:4.996250
Recv: // toolhead: X:69.500000 Y:98.500000 Z:4.995625 E:0.000000
Recv: // gcode: X:69.500000 Y:98.500000 Z:5.000000 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok
Send: M48
Recv: // PROBE_ACCURACY at X:69.500 Y:98.500 Z:4.996 (samples=10 retract=2.500 speed=5.0 lift_speed=30.0)
Recv: // probe at 69.500,98.500 is z=2.447250
Recv: // probe at 69.500,98.500 is z=2.452000
Recv: // probe at 69.500,98.500 is z=2.440500
Recv: // probe at 69.500,98.500 is z=2.449000
Recv: // probe at 69.500,98.500 is z=2.443750
Recv: // probe at 69.500,98.500 is z=2.443500
Recv: // probe at 69.500,98.500 is z=2.453250
Recv: // probe at 69.500,98.500 is z=2.455500
Recv: // probe at 69.500,98.500 is z=2.456500
Recv: // probe at 69.500,98.500 is z=2.451250
Recv: // probe accuracy results: maximum 2.456500, minimum 2.440500, range 0.016000, average 2.449250, median 2.450125, standard deviation 0.005130
Recv: ok
Send: GET_POSITION
Recv: // mcu: stepper_x:-14546 stepper_y:-16318 stepper_z:256 stepper_z1:256
Recv: // stepper: stepper_x:69.500000 stepper_y:98.500000 stepper_z:4.951250 stepper_z1:4.951250
Recv: // kinematic: X:69.500000 Y:98.500000 Z:4.951250
Recv: // toolhead: X:69.500000 Y:98.500000 Z:4.951250 E:0.000000
Recv: // gcode: X:69.500000 Y:98.500000 Z:4.955575 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok
```

FanDjango, 2020-02-29:
This is without any "corrective", i.e. original code:
```
Send: G28
Recv: // Sending BLTOUCH_DEBUG COMMAND=reset
Recv: // Sending BLTOUCH_DEBUG COMMAND=pin_up
Recv: ok
Send: M105
Recv: ok B:23.4 /40.0 T0:43.5 /40.0
Send: M105
Recv: ok B:23.7 /40.0 T0:44.0 /40.0
Send: GET_POSITION
Recv: // mcu: stepper_x:-14546 stepper_y:-16313 stepper_z:277 stepper_z1:277
Recv: // stepper: stepper_x:69.500000 stepper_y:98.500000 stepper_z:4.996250 stepper_z1:4.996250
Recv: // kinematic: X:69.500000 Y:98.500000 Z:4.996250
Recv: // toolhead: X:69.500000 Y:98.500000 Z:4.995625 E:0.000000
Recv: // gcode: X:69.500000 Y:98.500000 Z:5.000000 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok
Send: M105
Recv: ok B:24.6 /40.0 T0:44.1 /40.0
Send: M48
Recv: // PROBE_ACCURACY at X:69.500 Y:98.500 Z:4.996 (samples=10 retract=2.500 speed=5.0 lift_speed=30.0)
Recv: // probe at 69.500,98.500 is z=2.445000
Recv: // probe at 69.500,98.500 is z=2.446250
Recv: // probe at 69.500,98.500 is z=2.443750
Recv: // probe at 69.500,98.500 is z=2.447500
Recv: // probe at 69.500,98.500 is z=2.447500
Recv: // probe at 69.500,98.500 is z=2.450000
Recv: // probe at 69.500,98.500 is z=2.458750
Recv: // probe at 69.500,98.500 is z=2.448750
Recv: // probe at 69.500,98.500 is z=2.453750
Recv: // probe at 69.500,98.500 is z=2.457500
Recv: // probe accuracy results: maximum 2.458750, minimum 2.443750, range 0.015000, average 2.449875, median 2.448125, standard deviation 0.004888
Recv: ok
Send: M105
Recv: ok B:26.1 /40.0 T0:43.1 /40.0
Send: M105
Recv: ok B:26.1 /40.0 T0:42.8 /40.0
Send: GET_POSITION
Recv: // mcu: stepper_x:-14546 stepper_y:-16313 stepper_z:246 stepper_z1:246
Recv: // stepper: stepper_x:69.500000 stepper_y:98.500000 stepper_z:4.957500 stepper_z1:4.957500
Recv: // kinematic: X:69.500000 Y:98.500000 Z:4.957500
Recv: // toolhead: X:69.500000 Y:98.500000 Z:4.957500 E:0.000000
Recv: // gcode: X:69.500000 Y:98.500000 Z:4.961832 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok
Send: M105
Recv: ok B:26.9 /40.0 T0:42.5 /40.0
```

FanDjango, 2020-02-29:
If you put debugging output into that loop, you can see that pos[2] varies depending on how far the stepper got before the probe triggers
[image]

KevinOConnor, 2020-02-29:
FWIW, the GET_POSITION results do not seem to indicate a software issue.  Looking at the probe start:
```
Recv: // mcu: stepper_x:-14546 stepper_y:-16313 stepper_z:277 stepper_z1:277
Recv: // stepper: stepper_x:69.500000 stepper_y:98.500000 stepper_z:4.996250 stepper_z1:4.996250
```

This indicates the micro-controller has taken 277 total steps in a positive direction (since it was last rebooted) and the printer thinks the Z is at position 4.996250.

Later we see:
```
Recv: // mcu: stepper_x:-14546 stepper_y:-16313 stepper_z:246 stepper_z1:246
Recv: // stepper: stepper_x:69.500000 stepper_y:98.500000 stepper_z:4.957500 stepper_z1:4.957500
```

At this point, the micro-controller has taken a total of 246 steps in a positive direction, which is 31 steps less.  The printer thinks the Z is at position 4.957500.  That makes sense, as 4.996250 - 4.957500 = 31 * .00125.

So, it seems to me, either the micro-controller software was unable to count how many steps it actually took, the host controller software got out of sync with the micro-controller step counter, or the issue is somewhere in the hardware.

You can add debugging to stepper.py:note_homing_end() if you want to verify the host code is in sync with the micro-controller counter.  If you can reproduce the issue on a printer with trinamic drivers, then you can also query the position from the driver itself using DUMP_TMC.  The software itself (at least in this example) seems to line up correctly, so I'm not sure where one would otherwise look.

-Kevin

FanDjango, 2020-03-01:
> At this point, the micro-controller has taken a total of 246 steps in a positive direction, which is 31 steps less. The printer thinks the Z is at position 4.957500. That makes sense, as 4.996250 - 4.957500 = 31 * .00125.

I concur. At this level, everything is working correctly

> the micro-controller software was unable to count how many steps it actually took

I wouldn't say "unable". **But: **

For some more high-level debugging, I did the following test:
`G28`, position is then 4.996... 
`PROBE_ACCURACY SAMPLES=2`, I just wanted the Z position on trigger: 2.45...
Then I performed 500 (yes 500) `G0 Z100 F3600` followed by `G0 Z110 F3600`. A nice long up-down sequence, followed by a `G0 Z5 F3600` to get back down to where we were before, which is 4.996...
And now a `PROBE_ACCURACY SAMPLES=2` to see how much drift might have accumulated due to the long up-down sequence. The Z position on trigger was: 2.45..., **no big drift.**

But if you perform 500 samples in the `PROBE_ACCURACY` command, we get a substantial drift on that repeated similar up-down movement.

The only difference is that the downward move is **terminated by an endstop instead of deceleration**. In `stepper.c`, the routine `stop_steppers` will _brutally_ terminate any running stepper sequence and reset `step_pin` after saving the current position. **But what if the stepper advances another step or more, maybe only once in a while, due to the abrupt stop?** Secondly, I have not managed to discern if this is locked or not IRQ-wise yet, **what if the `stop_steppers` routine comes in between the `step_pin` set and subseqent reset in `stepper_event` or its siblings?** (To keep things simple, I have compiled with `STEP_DELAY ` not set).

My conclusion:
1. This should be a problem for ANY software or printer that runs into endstops too fast.
2. Do a G28 after any probing and you are back to reality
3. Back to back probing will have a slight drift towards higher values - could hit people with big meshes or who do many samples. Needs to be quantified but it's not a huge value and probably depends on many things (stepper driver, speed, mass, backlash etc, also processor speeds on the controller etc.)

I will now test for drift when the downard probe motion is real slow. And if that still occurs, then more debugging is called for, because then the cause is not the abrupt mechanical stop.

KevinOConnor, 2020-03-01:
> But what if the stepper advances another step or more, maybe only once in a while, due to the abrupt stop? 

Stepper motors can not lose a microstep.  They can only permanently lose steps in increments of two full steps.  This is a property of the current phases that the stepper motor driver uses - there's some info on this in https://www.klipper3d.org/Endstop_Phase.html (as well as a blurb in the FAQ at https://www.klipper3d.org/FAQ.html#can-i-find-out-whether-the-printer-has-lost-steps ).

> what if the stop_steppers routine comes in between the step_pin set and subseqent reset in stepper_event or its siblings?

A software bug is certainly possible.  I don't know where to start looking for it though.

For the particular scenario you outline above - timers always run with irqs disabled, so endstop_event() can not run while stepper_event() is running.  There is a small chance that the step of stepper_event() runs and then endstop_event() cancels the step before the second unstep event of stepper_event() runs.  However, stepper_stop() should still clean that up (the timing for that window is very tiny as well).

> To keep things simple, I have compiled with STEP_DELAY not set

FYI, that's not valid on an lpc1768 unless all drivers are trinamic.  If STEP_DELAY=0 on other drivers it will lead to random lost steps pulses throughout a print.

-Kevin

KevinOConnor, 2020-03-01:
If you add the following, does it reveal anything useful?
```diff
--- a/klippy/stepper.py
+++ b/klippy/stepper.py
@@ -136,6 +136,8 @@ class MCU_stepper:
         if not did_trigger or self._mcu.is_fileoutput():
             return
         params = self._get_position_cmd.send([self._oid])
+        logging.info("Stepper stopped at position %d (cp=%.3f)",
+                     params['pos'], self.get_commanded_position())
         mcu_pos_dist = params['pos'] * self._step_dist
         if self._invert_dir:
             mcu_pos_dist = -mcu_pos_dist
```

-Kevin

FanDjango, 2020-03-01:
> FYI, that's not valid on an lpc1768 unless all drivers are trinamic. If STEP_DELAY=0 on other drivers

Yeah, reverting that after these tests. Normal value was 4

FanDjango, 2020-03-01:
>  There is a small chance that the step of stepper_event() runs and then endstop_event() cancels the step before the second unstep event of stepper_event() runs. However, stepper_stop() should still clean that up (the timing for that window is very tiny as well).

So that would be a small chance for a step pulse that is too short on account of the unstep event being a separate entity from its issuing step event?

KevinOConnor, 2020-03-01:
> Yeah, reverting that after these tests. Normal value was 4. Didn't want the step/unstep chiming in on this

Okay, but be aware that if a step is lost between mcu and driver (because the step pulse is not long enough) it'll look remarkably similar to the symptoms here.

> So that would be a small chance for a step pulse that is too short on account of the unstep event being a separate entity from its issuing step event?

Yes, that is theoretically possible.  I don't think it could occur in practice though.  A 2us time is at most 200 instructions on an lpc1768 (likely more than about 120 instructions) and I don't think the step event could complete, endstop detection complete, and stepper_stop() complete in that few instructions.

While we're on the topic of small possibilities, it's also possible that the stepper could be stopped between the step on the first stepper and the step on the second stepper on a multi-z axis.  Again, I don't see this occurring in practice though.  (I think the debugging at https://github.com/KevinOConnor/klipper/issues/2509#issuecomment-593117711 would reveal if this were happening too.)

-Kevin

KevinOConnor, 2020-03-01:
> Normal value was 4

What stepper driver are you using?  If you are using a less common driver and you think your drivers need longer time, then another test would be to try with the code at PR #2565 - which filters out rapid step pulses and direction changes.  (Though this wouldn't have been an issue for the original reporter who is using tmc2208.)

-Kevin

FanDjango, 2020-03-01:
8825 on MKS SBASE

FanDjango, 2020-03-02:
Before I do some more debugging, I installed "the other" firmware and had a look see how that one fares on this. Took a while, a lot of work configuring, but here is the result:
```
Recv: Mean: 0.004750 Min: 0.001 Max: 0.009 Range: 0.007 Standard Deviation: 0.002487
Recv: Mean: 0.006250 Min: 0.003 Max: 0.011 Range: 0.009 Standard Deviation: 0.002739
Recv: Mean: 0.007000 Min: 0.003 Max: 0.013 Range: 0.010 Standard Deviation: 0.002634
Recv: Mean: 0.008750 Min: 0.005 Max: 0.013 Range: 0.007 Standard Deviation: 0.002372
Recv: Mean: 0.006125 Min: 0.004 Max: 0.009 Range: 0.005 Standard Deviation: 0.001420
Recv: Mean: 0.007625 Min: 0.005 Max: 0.010 Range: 0.005 Standard Deviation: 0.001420
Recv: Mean: 0.008625 Min: 0.005 Max: 0.013 Range: 0.007 Standard Deviation: 0.002707
Recv: Mean: 0.009875 Min: 0.006 Max: 0.014 Range: 0.008 Standard Deviation: 0.002197
Recv: Mean: 0.007125 Min: 0.004 Max: 0.011 Range: 0.007 Standard Deviation: 0.002169
Recv: Mean: 0.008875 Min: 0.005 Max: 0.011 Range: 0.006 Standard Deviation: 0.001972
Recv: Mean: 0.009000 Min: 0.005 Max: 0.013 Range: 0.007 Standard Deviation: 0.002291
Recv: Mean: 0.011250 Min: 0.007 Max: 0.014 Range: 0.006 Standard Deviation: 0.001677
Recv: Mean: 0.009500 Min: 0.006 Max: 0.011 Range: 0.005 Standard Deviation: 0.001601
Recv: Mean: 0.009875 Min: 0.007 Max: 0.013 Range: 0.005 Standard Deviation: 0.001625
Recv: Mean: 0.010125 Min: 0.007 Max: 0.013 Range: 0.005 Standard Deviation: 0.001719
Recv: Mean: 0.009500 Min: 0.006 Max: 0.015 Range: 0.009 Standard Deviation: 0.002806
Recv: Mean: 0.009375 Min: 0.005 Max: 0.014 Range: 0.009 Standard Deviation: 0.002322
Recv: Mean: 0.009750 Min: 0.005 Max: 0.014 Range: 0.009 Standard Deviation: 0.002610
Recv: Mean: 0.011250 Min: 0.007 Max: 0.016 Range: 0.009 Standard Deviation: 0.002500
Recv: Mean: 0.011500 Min: 0.007 Max: 0.015 Range: 0.008 Standard Deviation: 0.002291
Recv: Mean: 0.010875 Min: 0.009 Max: 0.014 Range: 0.005 Standard Deviation: 0.001586
```
As you can see: **same problem**

IMHO, it is the abrubt stop at endstop-hit that makes all operations that involve **repeated** travels into the endstops cause a slight drift, however many steps or whatever a stepper can theoretically overshoot at that moment, in practice something is being overshot.

The "home" operations travel into the endstops, **but** use whatever position that then turns out to be as an absolute value and reference for the future.

Other operations like PROBE_ACCURACY (but also BED_MESH...) do this without using the position as a new reference. In the case of all of these operations, I suppose one could do a homing operation "in between" n probes (this is not a joke, ok), thereby resetting the reference point before every productive probe. On a 64 point bed mesh with (say) 5 probes per point (which is 320 probe operations), one could home at each of the mesh locations prior to the 5 probes.

I am not advocating this, just playful thinking.

KevinOConnor, 2020-03-03:
> IMHO, it is the abrubt stop at endstop-hit that makes all operations that involve repeated travels into an endstop cause a slight drift, however many steps or whatever a stepper can theoretically overshoot at that moment, in practice something is being overshot.

If you think the software is not at fault, and the error continues to increase endlessly (eg, it grows to milli-meters), then I'd look at the binding between the z axis leadscrew and the z stepper.  If the grub screw (or similar) is loose, the constant start/stop may be causing slippage in that binding.

If the error stops growing, I'd guess something like backlash - for example, the constant start/stop may be aligning the bearings in the z leadscrew so as to emphasize backlash.

-Kevin

KevinOConnor, 2020-03-21:
Any further updates on this?  Is the root cause now thought to be hardware or software?

-Kevin

GarthSnyder, 2020-03-21:
If you have a lead screw Z axis, this kind of slow upward creep on repeated probing can be characteristic of a Z axis coupler on which one of the cuffs isn't tightly enough secured. You might check that just to rule it out.

On my couplers, the clamp screws require far more torque than you would ever think reasonable. I'm not sure what you do if you have set screws; I don't think they can be tightened this much without stripping.

fredizzimo, 2020-03-28:
@FanDjango Have you tried setting the `sample_retract_dist` to a higher value?

According to the documentation https://www.antclabs.com/bltouch-v3, there's a 1.9 mm +- 0.5mm, difference between the trigger point, and the lowest point of the probe. Your `sample_retract_dist` is set to 2mm, in other words smaller than that.

That means that it could touch the bed when the pin lowered after the toolhead has moved up. That would mess with the accuracy. I recommend 5mm, to make sure that everything is completely stabilized when moving down.

The probe itself can actually detect this situation, but unfortunately it looks like klipper still doesn't verify that the pin is actually lowered without errors before probing. Note that klipper also puts the probe into touch mode, which disables this kind of error detection and that mode persists even after the probe is reset, at least on my version of BLTouch. In my case, retracting the probe and then calling reset is enough, but that doesn't work on all clones, so my recommendation is to never enter the touch mode.

FanDjango, 2020-03-29:
@fredizzimo 

> That means that it could touch the bed when the pin lowered after the toolhead has moved up. That would mess with the accuracy

If too close to the bed, it would not be able to deploy fully and go into error mode. I don't understand how that would mess with accuracy. **If you actually mean:** If it is just slightly deployed far enough to not go into error mode (i.e. **partly deployed**), it might be more inaccurate than if it extends fully to its detent, I understand what you are perhaps implying and I will test that when I have some time. I never imagined that might make a difference. I really wonder... and in that case...

> I recommend 5mm, to make sure that everything is completely stabilized when moving down.

...then that could maybe be true.

> The probe itself can actually detect this situation

True. If the deploy is not possible (for any reason, it need not be bed too close, it could be a clog, dirt or defective magnet in the probes pin, **or it could _already_ be** in an alarm state for any reason) the probe goes into alarm state after 650us. This should be checked by the software before attempting the probe operation, ideally.

> but unfortunately it looks like klipper still doesn't verify that the pin is actually lowered without errors before probing

True. I have a version of `bltouch.py` with many changes analog to the `bltouch.cpp` in Marlin which handles that. Makes no difference re. this issue though, I tested both versions.

> Note that klipper also puts the probe into touch mode, 

I have looked at the current code and this is not true. @KevinOConnor repeatedly also goes on record saying that touch mode is detrimental in early BLTouche probes and many clones.

>  In my case, retracting the probe and then calling reset is enough, but that doesn't work on all clones

You need to reset, which will retract as part of the reset operation, unless: If currently in touch mode, you need a retract command and then a reset command (which is strange and what I consider a bug in the BLTouch itself). Clones are also differ in how they want to be reset after an error.

> so my recommendation is to never enter the touch mode

IMHO this is a non-sequitur. I use touch mode often with very good results. If the underlying actual problems are taken care of, touch mode is a viabe option.

Ok, but even if the probe is giving inaccurate readings, this will not explain a continuous shift of the measurements in one direction. If you believe that your previous statements **do** actually explain such a drift you could please explain that better.

So far there have been the following observations and statements:

1. Each successive probe reports a value that is getting higher than the previous probe (offset by the probe repeatability which hides this increase once in a while).

2. This is also observed, albeit with less absolute rate of increase, in Marlin and they have an open issue about this as I have now seen, with the same sort of question: [see here](https://github.com/MarlinFirmware/Marlin/issues/16660)

3. Hardware explanations could be a drift due to a not-tight-enough Z-coupling, which is knocked in the +-direction on each probe stepper stop.

I have stopped worrying about this problem though, as the drift only accumulates when probing more than a few times, and the drift is small enough that a G28 and a G29 will be good enough.

Also, when I turn off the printer and turn it on again, this resets the values. And that kind of rules out a loose coupling somehow, but WTH.

KevinOConnor, 2020-03-29:
> > but unfortunately it looks like klipper still doesn't verify that the pin is actually lowered without errors before probing

> True.

If the BL-Touch is in an error state (sensor_pin reports triggered) at the start of a probe, then the head will not move and the bltouch.py code will report a "BLTouch failed to deploy" error.  So, I do not believe there is a problem here.

-Kevin

FanDjango, 2020-03-29:
>If the BL-Touch is in an error state (sensor_pin reports triggered) at the start of a probe, then the head will not move and the bltouch.py code will report a "BLTouch failed to deploy" error. So, I do not believe there is a problem here.

Yes. Sorry. My oversight in writing that text. To make things easier for readers of this thread I will edit my post above.

Rudgas, 2020-04-01:
Just to chime in here, I see a similar problem although I don't know what causes is. I don't want to discount temperature drift as I'm using an inductive probe which shows drift after ~50samples upwards. #2668

fredizzimo, 2020-04-03:
> If the BL-Touch is in an error state (sensor_pin reports triggered) at the start of a probe, then the head will not move and the bltouch.py code will report a "BLTouch failed to deploy" error. So, I [...]

Yes. I just retested it since it was one year since I last tested your branch. It does indeed report "BLTouch failed to raise probe" in some cases, I did never see it report "BLTouch failed to deploy". The error message is misleading though, it doesn't fail to raise the probe, it fails to deploy it.

I haven't verified it, but I think the cause for the error message, is the following. The probe is put down, and immediately put into error state. It still starts a normal homing operation, which detects that the endstop has triggered, because that's the same signal as error. Now the probe should be raised, but that fails due to the error state.

But we are still out on very dangerous waters here. The default value for `pin_move_time` is 0.675 seconds, which indeed seems to be enough on my BLTouch Smart v2. But lowering the value to 0.5, which is very close to 0.675 causes problems. If I do `G28` followed py `PROBE` I get the same error. But only after the printhead already has moved down a bit, which could cause it to crash into the bed. I suggest that you change the default value to something with a little bit bigger margin, considering that clones could have even smaller margin.

After that I tested `G28` followed by `G1 z2.33` (my z offset is 1.33mm, so this lifts the pin 1mm up). Then it successfully probes, but when the pin moves down it hits the bed, causing accuracy issues. Even if I set the `pin_move_time` high enough, my preferred value is 1 second, it doesn't detect it. After more testing with my own code, this is actually the case there too, so it appears like the probe is not able to detect being too close to the bed in all cases.

I also tested `G1 z3.33, which corresponds to a 2mm retract. For me it doesn't touch the bed, but it's very, very close.

@FanDjango 
>> Note that klipper also puts the probe into touch mode,
>
> I have looked at the current code and this is not true. @KevinOConnor repeatedly also goes on record saying that touch mode is detrimental in early BLTouche probes and many clones.

It's set here https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/bltouch.py#L154 and that function is called from `lower_probe` I don't see any sort of reset after that. It can be disabled by setting `pin_up_touch_mode_reports_triggered` to false, but that's not the default.

But if you do that you don't get any error checking at all. IMO, at least on a genuine BLTouch, klipper should test that the probe is error free after both lowering and raising the pin.

fredizzimo, 2020-04-04:
Anyway back to the actual issue. I tested it too and see the same behaviour 

```
Send: probe_accuracy probe_speed=10 samples=100
Recv: // PROBE_ACCURACY at X:97.000 Y:2.000 Z:3.315 (samples=100 retract=2.000 speed=10.0 lift_speed=10.0)
Recv: // probe at 97.000,2.000 is z=1.320000
Recv: // probe at 97.000,2.000 is z=1.317500
Recv: // probe at 97.000,2.000 is z=1.325000
Recv: // probe at 97.000,2.000 is z=1.320000
Recv: // probe at 97.000,2.000 is z=1.302500
Recv: // probe at 97.000,2.000 is z=1.315000
Recv: // probe at 97.000,2.000 is z=1.315000
Recv: // probe at 97.000,2.000 is z=1.322500
Recv: // probe at 97.000,2.000 is z=1.315000
Recv: // probe at 97.000,2.000 is z=1.312500
Recv: // probe at 97.000,2.000 is z=1.297500
Recv: // probe at 97.000,2.000 is z=1.305000
Recv: // probe at 97.000,2.000 is z=1.312500
Recv: // probe at 97.000,2.000 is z=1.312500
Recv: // probe at 97.000,2.000 is z=1.310000
Recv: // probe at 97.000,2.000 is z=1.322500
Recv: // probe at 97.000,2.000 is z=1.325000
Recv: // probe at 97.000,2.000 is z=1.320000
Recv: // probe at 97.000,2.000 is z=1.317500
Recv: // probe at 97.000,2.000 is z=1.320000
Recv: // probe at 97.000,2.000 is z=1.330000
Recv: // probe at 97.000,2.000 is z=1.320000
Recv: // probe at 97.000,2.000 is z=1.317500
Recv: // probe at 97.000,2.000 is z=1.327500
Recv: // probe at 97.000,2.000 is z=1.315000
Recv: // probe at 97.000,2.000 is z=1.330000
Recv: // probe at 97.000,2.000 is z=1.320000
Recv: // probe at 97.000,2.000 is z=1.325000
Recv: // probe at 97.000,2.000 is z=1.317500
Recv: // probe at 97.000,2.000 is z=1.322500
Recv: // probe at 97.000,2.000 is z=1.315000
Recv: // probe at 97.000,2.000 is z=1.330000
Recv: // probe at 97.000,2.000 is z=1.322500
Recv: // probe at 97.000,2.000 is z=1.322500
Recv: // probe at 97.000,2.000 is z=1.322500
Recv: // probe at 97.000,2.000 is z=1.320000
Recv: // probe at 97.000,2.000 is z=1.325000
Recv: // probe at 97.000,2.000 is z=1.335000
Recv: // probe at 97.000,2.000 is z=1.312500
Recv: // probe at 97.000,2.000 is z=1.330000
Recv: // probe at 97.000,2.000 is z=1.325000
Recv: // probe at 97.000,2.000 is z=1.312500
Recv: // probe at 97.000,2.000 is z=1.335000
Recv: // probe at 97.000,2.000 is z=1.335000
Recv: // probe at 97.000,2.000 is z=1.330000
Recv: // probe at 97.000,2.000 is z=1.322500
Recv: // probe at 97.000,2.000 is z=1.335000
Recv: // probe at 97.000,2.000 is z=1.312500
Recv: // probe at 97.000,2.000 is z=1.337500
Recv: // probe at 97.000,2.000 is z=1.337500
Recv: // probe at 97.000,2.000 is z=1.332500
Recv: // probe at 97.000,2.000 is z=1.312500
Recv: // probe at 97.000,2.000 is z=1.345000
Recv: // probe at 97.000,2.000 is z=1.337500
Recv: // probe at 97.000,2.000 is z=1.337500
Recv: // probe at 97.000,2.000 is z=1.335000
Recv: // probe at 97.000,2.000 is z=1.320000
Recv: // probe at 97.000,2.000 is z=1.330000
Recv: // probe at 97.000,2.000 is z=1.337500
Recv: // probe at 97.000,2.000 is z=1.335000
Recv: // probe at 97.000,2.000 is z=1.325000
Recv: // probe at 97.000,2.000 is z=1.340000
Recv: // probe at 97.000,2.000 is z=1.337500
Recv: // probe at 97.000,2.000 is z=1.340000
Recv: // probe at 97.000,2.000 is z=1.322500
[... 121 lines cut ...]
Recv: // probe at 97.000,2.000 is z=1.377500
Recv: // probe at 97.000,2.000 is z=1.405000
Recv: // probe at 97.000,2.000 is z=1.405000
Recv: // probe at 97.000,2.000 is z=1.380000
Recv: // probe at 97.000,2.000 is z=1.380000
Recv: // probe at 97.000,2.000 is z=1.382500
Recv: // probe at 97.000,2.000 is z=1.392500
Recv: // probe at 97.000,2.000 is z=1.390000
Recv: // probe at 97.000,2.000 is z=1.385000
Recv: // probe at 97.000,2.000 is z=1.372500
Recv: // probe at 97.000,2.000 is z=1.372500
Recv: // probe at 97.000,2.000 is z=1.407500
Recv: // probe at 97.000,2.000 is z=1.410000
Recv: // probe at 97.000,2.000 is z=1.405000
Recv: // probe at 97.000,2.000 is z=1.410000
Recv: // probe at 97.000,2.000 is z=1.392500
Recv: // probe at 97.000,2.000 is z=1.387500
Recv: // probe at 97.000,2.000 is z=1.377500
Recv: // probe at 97.000,2.000 is z=1.380000
Recv: // probe at 97.000,2.000 is z=1.412500
Recv: // probe accuracy results: maximum 1.412500, minimum 1.327500, range 0.085000, average 1.375325, median 1.378750, standard deviation 0.019741
```
I have done several other tests since, and it looks like it was just a coincidence that the first one is better, I get the same kind of bad accuracy with the higher speed as well in other tests. And because it also fails at lower speed, I don't think it's a mechanical issue.

It also doesn't look fully gradual, it looks more like it's skipping several substeps at once, and then settles near a higher value for a few probes, and then skips again. But that could also be my imagination.

FanDjango, 2020-04-04:
> it's set here https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/bltouch.py#L154 and that function is called from lower_probe I don't see any sort of reset after that. It can be disabled by setting pin_up_touch_mode_reports_triggered to false, but that's not the default.

`test_sensor` is called by 
```
    def lower_probe(self):
        self.test_sensor()
        self.sync_print_time()
        duration = max(MIN_CMD_TIME, self.pin_move_time - MIN_CMD_TIME)
        self.send_cmd('pin_down', duration=duration)
        self.send_cmd(None)
```

1. It does not get executed every time, rather only the first time and then maximum every 5*60s. If the problem you are seeing occurs more often, that is the first indication that what you write is **wrong**.

2. A BLTouch smart will come out of SW mode immediately when a pin-up or pin-down command is issued, as is the case in `lower_probe` after calling `test_sensor`. No need for "any sort of reset after that".
It looks like the pin-down in `lower_probe` is not checked, but it is. It's a bit tricky, but since due to the endstop getting invoked immediately, movement does not even begin, and then in `probe_finalize` you get the error message. @KevinOConnor can correct me if this is wrong.

Nuff said.

Regarding your pin hitting the bed on deploy:

I suggest you consider mounting the probe in such a fashion as to avoid this. You **do** have a certain influence on the characteristics. My V2.0 also has **much less** tolerance for this than my V3.0 and V3.1 probes. ANTClabs **does** state that the V3.x has more travel millimeters. Also adjust the raise height after probe and homing to account for this. There is no reason to program around problems of this kind.

> Yes. I just retested it since it was one year since I last tested your branch. It does indeed report "BLTouch failed to raise probe" in some cases, I did never see it report "BLTouch failed to [...]

Current code checks and detects and even resets a deploy error before moving. The previous code checked for a successful deploy and didn't move on detecting an error. Previous code much longer ago didn't wait long enough to detect the error. Much has changed, are you sure to be using the newest klippy (new stuff in the last weeks, even days), if not please do.

> But we are still out on very dangerous waters here. The default value for pin_move_time is 0.675 seconds, which indeed seems to be enough on my BLTouch Smart v2. But lowering the value to 0.5, which [...]

These are the values I coded into Marlin nearly a year ago:

```
#ifndef BLTOUCH_SET5V_DELAY
  #define BLTOUCH_SET5V_DELAY   150
#endif
#ifndef BLTOUCH_SETOD_DELAY
  #define BLTOUCH_SETOD_DELAY   150
#endif
#ifndef BLTOUCH_MODE_STORE_DELAY
  #define BLTOUCH_MODE_STORE_DELAY 150
#endif
#ifndef BLTOUCH_DEPLOY_DELAY
  #define BLTOUCH_DEPLOY_DELAY   750
#endif
#ifndef BLTOUCH_STOW_DELAY
  #define BLTOUCH_STOW_DELAY     750
#endif
#ifndef BLTOUCH_RESET_DELAY
  #define BLTOUCH_RESET_DELAY    500
#endif
```

So I agree, yes. Differentiating by command is done, by the way, to speed up operations in those places where it is sensible and to delay in those places where it is needed. Notice the 750ms for the wait to check error after a deploy.

My own `bltouch_fandjango.py` uses code like this and works very well. But we are putting changes into the current klippy bit by bit.

FanDjango, 2020-04-04:
> Anyway back to the actual issue. I tested it too and see the same behaviour

Finally!

> I have done several other tests since, and it looks like it was just a coincidence that the first one is better, I get the same kind of bad accuracy with the higher speed as well in other tests. And because it also fails at lower speed, I don't think it's a mechanical issue.

It also doesn't look fully gradual, it looks more like it's skipping several substeps at once, and then settles near a higher value for a few probes, and then skips again. But that could also be my imagination.

Same for me. But it happens in Marlin too, albeit slightly smaller rate of increase.

As soon as the printer is not so busy, I will do a 100.000+ sequence to see if the values grow and grow and grow -> thereby really ruling out a mechanical issue.

fredizzimo, 2020-04-04:
@FanDjango 

You are right, that the probe exits the touch mode when a pin down is issued on my probe. But I'm fairly sure I recall that not being the case on some clones from the lenghty discussions a year ago.

My update was from around one week ago when I tested those changes, and seeing that there are many new updates I retested things again. 

First the sequence `G28; PROBE`, It does indeed report that the probe failed to deploy, but only after reporting that it first failed to raise it
```
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: !! BLTouch failed to deploy
Changing monitoring state from "Operational" to "Error: BLTouch failed to deploy"
```
The first message must have confused me and I didn't see the second one, or then the second message is not happening all the time. I tried a few times with the newest updates, but only once with the older one and it seems to be consistent now, so that's good.

But then I lowered the `pin_move_time` to 0.675 from my default of 1 and this happened. I got the same result with a `pin_move_time` of 0.7, but 0.8 finally made it detect the problem in time
```
Send: G28
Recv: ok
[...]
Send: probe
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // probe at 97.000,2.000 is z=1.262500
Recv: // Result is z=1.262500
Recv: ok
[...]
Send: probe
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // probe at 97.000,2.000 is z=1.195000
Recv: // Result is z=1.195000
Recv: ok
[...]
Send: probe
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // probe at 97.000,2.000 is z=1.122500
Recv: // Result is z=1.122500
Recv: ok
[...]
Send: probe
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // probe at 97.000,2.000 is z=1.050000
Recv: // Result is z=1.050000
Recv: ok
[...]
```
Each time I can hear the steppers move, and the lowering z value also indicates that. The probe also enters the error state, but this code 
```
    def probe_finalize(self):
        if self.multi == 'OFF':
            self.raise_probe()
        self.sync_print_time()
        # Verify the probe actually deployed during the attempt
        for s, mcu_pos in self.start_mcu_pos:
            if s.get_mcu_position() == mcu_pos:
                raise homing.EndstopError("BLTouch failed to deploy")
```
Does not really care about the error state, since `self.raise_probe` resets giving only that `Failed to verify BLTouch probe is raised; retrying.` warning

This was what I thought was your initial issue, and that's why I reported it here. However, it's clearly not since our accuracy test numbers goes the other direction.

My point about the probe hitting the bed was not about mounting. I issued 
`G28; G1 Z2.33; PROBE`. This simulates probing with a lift distance of only 1mm, and it does indeed hit the bed immediately in this case. It's just a matter of how much pre-travel the probe needs before triggering, you need to lift at least that amount before lowering the probe.

Also no probe mounting will prevent the head from hitting the bed if it moves down even if it detects and error. In practice it's probably won't happen if your coniguration is correct, and you always have a lift command before probing. With manual probes or a misconfigured configuration it can definitely happen and because the probe itself is able to detect the error, so should the code.

FanDjango, 2020-04-04:
> But then I lowered the pin_move_time to 0.675 from my default of 1 and this happened. I got the same result with a pin_move_time of 0.7, but 0.8 finally made it detect the problem in time

Ok, I see what you mean (in part).

Actually the head shouldn't move until the deploy is finished. Let's recheck the deploy code. On my 32-bit controller and my Tinkerboard, maybe some omission to use `sync_print_tim()` in the right places is simply not noticable because my system is fast. For the sake of curiosity, what is your system comprised of?

>  I issued
G28; G1 Z2.33; PROBE. This simulates probing with a lift distance of only 1mm, and it does indeed hit the bed immediately in this case.

I will try that, even on my fast system maybe I will thus see a move happening "to early" while the pin is still moving. Tomorrow, ok? I need to re-setup that printer and refit a V2.0, I currently use a V3.1.

> First the sequence G28; PROBE, It does indeed report that the probe failed to deploy, but only after reporting that it first failed to raise it

I will also look at that, because it looks strange.

What I would really like, is consistent "DEPLOY invoke", "succeeded" and "STOW invoked", "succeeded", and fails of course, and all other commands. in the log and I will attempt a PR in that direction coming week.

I have that in my code version and its a huge help. I know @KevinOConnor can see things in the log that others cannot, but I am just the BLTouch person and then I (and you) could see at a glance some of the sequence errors that might be happening.

Give it a few more days and we will be getting there. Ok?

FanDjango, 2020-04-04:
> Probe accuracy: always higher z?

Can we sum up the problem again?

1. The probes Z-value slowly but surely increases as more and more probe operations are completed

2. Of course the probe inaccuracy means that the values sometimes decrease but over time, there is a definitve slow trend upwards

**3.If you do a home, you are back to where you were in the beginning.**

Is number 3 true?

FanDjango, 2020-04-04:
@fredizzimo Why don't you copy out some of this discussion and make a separate issue for the **G28; G1 Z2.33; PROBE** discussion?

fredizzimo, 2020-04-04:
I hooked up a logic analyzer, and although I have only been able to take a very quick look the results are quite interesting.

You should be able to open the capture with this https://ideas.saleae.com/f/changelog/ after unzipping the zip file.

Channel 0 is the BLTouch command
Channel 1 was supposed to be the direction pin, but it looks like garbage, it looks like using PB6 on my Ender 3 did not work for that
Channel 2 is the stepper commands
Chnnel 3 empty
Channel 4 is the endstop signal

There are two things I can immediately see, as illustrated in the following screen capture 

[image]

You can see that it for some reason takes three extra steps after the probe is triggered.

The other thing you can see is that there's quite a bit of noise around the trigger, with one very early spike. Fortunately it seems to stabilize within one step, so it shouldn't be an issue, but I haven't verified it against the sample count and sample time in the code, to check if the timings there makes sense.

[probe_accuracy.txt](https://github.com/KevinOConnor/klipper/files/4431867/probe_accuracy.txt)
[proce_accuracy.zip](https://github.com/KevinOConnor/klipper/files/4431868/proce_accuracy.zip)

KevinOConnor, 2020-04-04:
FWIW, if this is being seen on a machine with TMC drivers on the Z axis, there's an easy test that would confirm if Klipper is getting out of sync or not.  First make sure you have the latest version of Klipper (see the FAQ for update instructions), make sure you are running a pristine version of Klipper, and then run: `M400`, `GET_POSITION`, `DUMP_TMC STEPPER=stepper_z`, `PROBE_ACCURACY SAMPLES=20`, `M400`, `GET_POSITION`, `DUMP_TMC STEPPER=stepper_z`, `M112` .  Then attach the resulting Klipper log file.  Without a log file, there's little I can do.

-Kevin

fredizzimo, 2020-04-04:
I analysed the capture file a bit more and wrote a python script to count the steps. I started my probe at a hight of 10mm, then counted the steps up and down (which was slightly tricky due to the missing direction signal, but the pauses made it quite easy to recognize still). The final distance ended up being 3.4025, which is exactly the same as the final probe result 1.4025 + 2mm lift. 

This confirms that the firmware is counting the steps correctly and that there's indeed is a physical loss of steps somewhere. 

Since @FanDjango already tested movement without probing, and didn't see any loss of steps there, the only reasonable explanation I can think of is the abrupt stop without deceleration at the end of the move. But the strange thing is that it doesn't look like it's loosing full steps, so maybe it's actually slipping on the threaded rod?

I guess I should test with a really slow speed, if 2mm/s was still too much. 

If that proves to be the case, I think some kind of deceleration could be implemented purely on the controller side. It would then be followed by a very short move up again to get back to the position it was supposed to be in. This operation should be quite safe to do with BLTouch probe, since if mounted correctly, there's still some distance left to move down before the nozzle actually hits the bed. 

Unfortunately I don't have a TMC to test with, well I have a bigtreetech skr mini E3 with TMC2208 drivers waiting to be installed. But as far as I know, the 2208 drivers don't support missing step detection, so it wouldn't help.

FanDjango, 2020-04-04:
> This operation should be quite safe to do with BLTouch probe, since if mounted correctly, there's still some distance left to move down before the nozzle actually hits the bed.

I was on the track of a deceleration idea also (see my posts far above where I worried about missing steps on the end stop trigger point). 

I got sidetracked by the statement that a stepper cannot lose less than 2 full steps and so on.

But first: This is not really as big a problem as it seems, as apart from probing a huge mesh, an endstop approach is done once for X, once for Y, and maybe 2 + NxN*2 (2 samples per) for Y. So, so what? Next home will reset this.

To compensate the slight drift during a mesh probe operation I just took the correction factor I statistically determing for **my** machine and factored that in, as described above.

Works for me and thus I have removed myself from the discussion of slipping on the rod (isn't the case) or lost steps.

Still, I do believe the steps are being lost. Do the probe downward move at speed 30 and listen to the bang when the axis stops with no deceleration. But at speed 2, kinetic energy is lower and it still happens, so it must be systematic somehow.

FanDjango, 2020-04-04:
Here is what I did (see the 0.48 in that code):

        if probe_pos:
            for s, name, spos, epos in end_mcu_pos:
                md = (epos - 0.48 - spos) * s.get_step_dist()
                s.set_tag_position(s.get_tag_position() + md)
            self.set_homed_position(kin.calc_tag_position())
        else:
            self.toolhead.set_position(movepos)

Using a statistics approach I have found that (on average over many probe operations) there is a 0.48 chance that one step too few will be counted on probe endstop trigger, i.e. epos should be one step further to the minus-position direction about half the times.

Since we pick up the stop position by backward calculation from the step count, we slowly creep away from the real position.

I have empirically determined this value and it is only valid for my machine. I think a lower bound for this value would be zero, but I know of no plausible upper bound. In my case this means a creep of 100 * 0.48 * 0.00125 on one hundred probes = ca. 0.06mm (which can be seen on the 100 probe sequence many posts earlier).

KevinOConnor, 2020-04-04:
> But as far as I know, the 2208 drivers don't support missing step detection, so it wouldn't help.

Any tmc driver (as long as it is configured via uart or spi) is sufficient to run this test.  All the tmc drivers can report their current microstep phase, and we can use `mcu_total_steps % (microsteps*4) == stepper_phase`.  Thus having the stepper phase info is enough to determine if the mcu and stepper ever get out of sync.

-Kevin

fredizzimo, 2020-04-04:
Hm... I just did a test with a downward speed of 0.1, and it still drifts
```
Recv: // PROBE_ACCURACY at X:97.000 Y:2.000 Z:10.000 (samples=100 retract=2.000 speed=0.1 lift_speed=10.0)
Recv: // probe at 97.000,2.000 is z=1.307500
Recv: // probe at 97.000,2.000 is z=1.310000
Recv: // probe at 97.000,2.000 is z=1.312500
Recv: // probe at 97.000,2.000 is z=1.335000
Recv: // probe at 97.000,2.000 is z=1.310000
Recv: // probe at 97.000,2.000 is z=1.315000
Recv: // probe at 97.000,2.000 is z=1.315000
Recv: // probe at 97.000,2.000 is z=1.342500
Recv: // probe at 97.000,2.000 is z=1.342500
Recv: // probe at 97.000,2.000 is z=1.315000
Recv: // probe at 97.000,2.000 is z=1.340000
Recv: // probe at 97.000,2.000 is z=1.320000
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.327500
Recv: // probe at 97.000,2.000 is z=1.320000
Recv: // probe at 97.000,2.000 is z=1.322500
Recv: // probe at 97.000,2.000 is z=1.350000
Recv: // probe at 97.000,2.000 is z=1.350000
Recv: // probe at 97.000,2.000 is z=1.357500
Recv: // probe at 97.000,2.000 is z=1.352500
Recv: // probe at 97.000,2.000 is z=1.352500
Recv: // probe at 97.000,2.000 is z=1.335000
Recv: // probe at 97.000,2.000 is z=1.335000
Recv: // probe at 97.000,2.000 is z=1.352500
Recv: // probe at 97.000,2.000 is z=1.335000
Recv: // probe at 97.000,2.000 is z=1.355000
Recv: // probe at 97.000,2.000 is z=1.335000
Recv: // probe at 97.000,2.000 is z=1.355000
Recv: // probe at 97.000,2.000 is z=1.357500
Recv: // probe at 97.000,2.000 is z=1.335000
Recv: // probe at 97.000,2.000 is z=1.357500
Recv: // probe at 97.000,2.000 is z=1.360000
Recv: // probe at 97.000,2.000 is z=1.355000
Recv: // probe at 97.000,2.000 is z=1.340000
Recv: // probe at 97.000,2.000 is z=1.335000
Recv: // probe at 97.000,2.000 is z=1.365000
Recv: // probe at 97.000,2.000 is z=1.360000
Recv: // probe at 97.000,2.000 is z=1.335000
Recv: // probe at 97.000,2.000 is z=1.362500
Recv: // probe at 97.000,2.000 is z=1.362500
Recv: // probe at 97.000,2.000 is z=1.362500
Recv: // probe at 97.000,2.000 is z=1.340000
Recv: // probe at 97.000,2.000 is z=1.362500
Recv: // probe at 97.000,2.000 is z=1.340000
Recv: // probe at 97.000,2.000 is z=1.365000
Recv: // probe at 97.000,2.000 is z=1.370000
Recv: // probe at 97.000,2.000 is z=1.367500
Recv: // probe at 97.000,2.000 is z=1.345000
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.345000
Recv: // probe at 97.000,2.000 is z=1.345000
Recv: // probe at 97.000,2.000 is z=1.342500
Recv: // probe at 97.000,2.000 is z=1.345000
Recv: // probe at 97.000,2.000 is z=1.372500
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.342500
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.345000
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.350000
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.377500
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.377500
Recv: // probe at 97.000,2.000 is z=1.377500
Recv: // probe at 97.000,2.000 is z=1.377500
Recv: // probe at 97.000,2.000 is z=1.382500
Recv: // probe at 97.000,2.000 is z=1.380000
Recv: // probe at 97.000,2.000 is z=1.357500
Recv: // probe at 97.000,2.000 is z=1.355000
Recv: // probe at 97.000,2.000 is z=1.382500
Recv: // probe at 97.000,2.000 is z=1.355000
Recv: // probe at 97.000,2.000 is z=1.385000
Recv: // probe at 97.000,2.000 is z=1.382500
Recv: // probe at 97.000,2.000 is z=1.382500
Recv: // probe at 97.000,2.000 is z=1.385000
Recv: // probe at 97.000,2.000 is z=1.382500
Recv: // probe at 97.000,2.000 is z=1.385000
Recv: // probe at 97.000,2.000 is z=1.387500
Recv: // probe at 97.000,2.000 is z=1.385000
Recv: // probe at 97.000,2.000 is z=1.387500
Recv: // probe at 97.000,2.000 is z=1.390000
Recv: // probe at 97.000,2.000 is z=1.362500
Recv: // probe at 97.000,2.000 is z=1.370000
Recv: // probe at 97.000,2.000 is z=1.362500
Recv: // probe at 97.000,2.000 is z=1.365000
Recv: // probe at 97.000,2.000 is z=1.362500
Recv: // probe at 97.000,2.000 is z=1.372500
Recv: // probe at 97.000,2.000 is z=1.367500
Recv: // probe at 97.000,2.000 is z=1.387500
Recv: // probe at 97.000,2.000 is z=1.390000
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.365000
Recv: // probe at 97.000,2.000 is z=1.365000
Recv: // probe accuracy results: maximum 1.390000, minimum 1.307500, range 0.082500, average 1.356500, median 1.357500, standard deviation 0.021092
```
This generates a step time of 25ms, which is much longer than the last step of a deceleration 
With my values step distance 0.0025 and 100 mm/s^2 acceleration would be sqrt(step_distance / (0.5*a)) = sqrt(0.0025 / 50) = 7ms

One possibly interesting thing in the above log is the oscillation between 1.37 and 1.34, it almost goes a full step back and forward. 

So, now I have no idea what's going on. And the problem has to be on the downward movement, because lost steps when moving up would result in the distance getting smaller and smaller. A bug in the stepper driver is also a possibility but I'm using an original Ender 3 with A4988 stepper drivers and the original post has TMC2208. I didn't find what drivers you use @FanDjango.

Anyway, i guess I will install the other board tomorrow, and run the TMC microstepping tests.

FanDjango, 2020-04-04:
The printer I am using for this issue uses DRV8825 in an MKS SBASE (32bit LPC 1768)

FanDjango, 2020-04-04:
> One possibly interesting thing in the above log is the oscillation between 1.37 and 1.34, it almost goes a full step back and forward.

Yes, I agree

fredizzimo, 2020-04-04:
I realized that I should have used the second last step time in the calculation above, which is around 3ms, which corresponds to a speed of around 0.8 mm/s, so I ran a test
```
Recv: // PROBE_ACCURACY at X:97.000 Y:2.000 Z:10.000 (samples=100 retract=2.000 speed=0.8 lift_speed=10.0)
Recv: // probe at 97.000,2.000 is z=1.342500
Recv: // probe at 97.000,2.000 is z=1.342500
Recv: // probe at 97.000,2.000 is z=1.342500
Recv: // probe at 97.000,2.000 is z=1.342500
Recv: // probe at 97.000,2.000 is z=1.342500
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.360000
Recv: // probe at 97.000,2.000 is z=1.345000
Recv: // probe at 97.000,2.000 is z=1.342500
Recv: // probe at 97.000,2.000 is z=1.350000
Recv: // probe at 97.000,2.000 is z=1.370000
Recv: // probe at 97.000,2.000 is z=1.345000
Recv: // probe at 97.000,2.000 is z=1.340000
Recv: // probe at 97.000,2.000 is z=1.345000
Recv: // probe at 97.000,2.000 is z=1.350000
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.350000
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.345000
Recv: // probe at 97.000,2.000 is z=1.355000
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.352500
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.345000
Recv: // probe at 97.000,2.000 is z=1.352500
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.362500
Recv: // probe at 97.000,2.000 is z=1.355000
Recv: // probe at 97.000,2.000 is z=1.355000
Recv: // probe at 97.000,2.000 is z=1.357500
Recv: // probe at 97.000,2.000 is z=1.355000
Recv: // probe at 97.000,2.000 is z=1.355000
Recv: // probe at 97.000,2.000 is z=1.352500
Recv: // probe at 97.000,2.000 is z=1.347500
Recv: // probe at 97.000,2.000 is z=1.355000
Recv: // probe at 97.000,2.000 is z=1.357500
Recv: // probe at 97.000,2.000 is z=1.357500
Recv: // probe at 97.000,2.000 is z=1.357500
Recv: // probe at 97.000,2.000 is z=1.385000
Recv: // probe at 97.000,2.000 is z=1.360000
Recv: // probe at 97.000,2.000 is z=1.362500
Recv: // probe at 97.000,2.000 is z=1.355000
Recv: // probe at 97.000,2.000 is z=1.355000
Recv: // probe at 97.000,2.000 is z=1.367500
Recv: // probe at 97.000,2.000 is z=1.360000
Recv: // probe at 97.000,2.000 is z=1.357500
Recv: // probe at 97.000,2.000 is z=1.357500
Recv: // probe at 97.000,2.000 is z=1.362500
Recv: // probe at 97.000,2.000 is z=1.360000
Recv: // probe at 97.000,2.000 is z=1.355000
Recv: // probe at 97.000,2.000 is z=1.365000
Recv: // probe at 97.000,2.000 is z=1.360000
Recv: // probe at 97.000,2.000 is z=1.362500
Recv: // probe at 97.000,2.000 is z=1.362500
Recv: // probe at 97.000,2.000 is z=1.362500
Recv: // probe at 97.000,2.000 is z=1.372500
Recv: // probe at 97.000,2.000 is z=1.360000
Recv: // probe at 97.000,2.000 is z=1.370000
Recv: // probe at 97.000,2.000 is z=1.362500
Recv: // probe at 97.000,2.000 is z=1.365000
Recv: // probe at 97.000,2.000 is z=1.367500
Recv: // probe at 97.000,2.000 is z=1.365000
Recv: // probe at 97.000,2.000 is z=1.390000
Recv: // probe at 97.000,2.000 is z=1.370000
Recv: // probe at 97.000,2.000 is z=1.372500
Recv: // probe at 97.000,2.000 is z=1.367500
Recv: // probe at 97.000,2.000 is z=1.385000
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.370000
Recv: // probe at 97.000,2.000 is z=1.367500
Recv: // probe at 97.000,2.000 is z=1.370000
Recv: // probe at 97.000,2.000 is z=1.367500
Recv: // probe at 97.000,2.000 is z=1.367500
Recv: // probe at 97.000,2.000 is z=1.370000
Recv: // probe at 97.000,2.000 is z=1.370000
Recv: // probe at 97.000,2.000 is z=1.380000
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.372500
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.372500
Recv: // probe at 97.000,2.000 is z=1.377500
Recv: // probe at 97.000,2.000 is z=1.370000
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.380000
Recv: // probe at 97.000,2.000 is z=1.372500
Recv: // probe at 97.000,2.000 is z=1.380000
Recv: // probe at 97.000,2.000 is z=1.372500
Recv: // probe at 97.000,2.000 is z=1.382500
Recv: // probe at 97.000,2.000 is z=1.375000
Recv: // probe at 97.000,2.000 is z=1.382500
Recv: // probe at 97.000,2.000 is z=1.382500
Recv: // probe at 97.000,2.000 is z=1.380000
Recv: // probe at 97.000,2.000 is z=1.380000
Recv: // probe at 97.000,2.000 is z=1.382500
Recv: // probe at 97.000,2.000 is z=1.370000
Recv: // probe at 97.000,2.000 is z=1.380000
Recv: // probe accuracy results: maximum 1.390000, minimum 1.340000, range 0.050000, average 1.361750, median 1.361250, standard deviation 0.012572
```

That's very interesting, it's very consistent with that speed, and very gradually increasing.

KevinOConnor, 2020-04-08:
FYI, I created a new work-bltouch-20200408 branch with additional debugging. It may be useful to retest with that branch and then attach the resulting Klipper log file here.

`cd ~/klipper ; git fetch ; git checkout origin/work-bltouch-20200408 ; make ; make flash ; sudo service klipper restart`

-Kevin

bitbyt3r, 2020-04-13:
> FWIW, if this is being seen on a machine with TMC drivers on the Z axis, there's an easy test that would confirm if Klipper is getting out of sync or not. First make sure you have the latest version [...]

[klippy.log](https://github.com/KevinOConnor/klipper/files/4468979/klippy.log)

I'm not sure how to interpret the attached log, but I've been seeing similar drift issues on my printer as well. I'm running the latest version of klipper as of last night, and using TMC drivers. I'm happy to run any tests or try other klipper versions as needed.

KevinOConnor, 2020-04-19:
Thanks, but the provided log is not complete and it is not using the pristine code from this repo.  Unfortunately, this type of detailed analysis is too time consuming for me to justify with such additional unknowns.

-Kevin

KevinOConnor, 2020-05-08:
Any further updates on this?

-Kevin

closed 2020-06-01 by KevinOConnor

LeviTheOtaku, 2021-01-05:
I'm still having this issue. (latest klipper build)
Was it ever solved?

KevinOConnor, 2021-01-05:
As far as I'm aware, the software works as intended.  My best guess is that an increasing Z during probing is due to slippage of the hardware.

I think the only way to make progress on this issue would be for someone with the problematic hardware to do the research and work necessary to provide evidence that the software is doing something incorrect.

-Kevin

mattswest, 2021-01-09:
I am having this issue as well.

Firmware version: v0.9.1-140-ge68cf08d
Ender 3 Pro
BIGTREETCH SKR Mini E3 V2.0
Generic 3D Touch from AliExpress

The sensor itself is super cheap, like however cheap you're thinking it was probably cheaper, but given everything I've read in this thread, as well as my experience with the same sensor on four other machines.

Here's an example of what I'm seeing.
```
Send: G28
Recv: ok
Send: GET_POSITION
Recv: // mcu: stepper_x:3625 stepper_y:-801 stepper_z:284
Recv: // stepper: stepper_x:135.000000 stepper_y:120.000000 stepper_z:-0.199000
Recv: // kinematic: X:135.000000 Y:120.000000 Z:-0.199000
Recv: // toolhead: X:135.000000 Y:120.000000 Z:10.000000 E:0.000000
Recv: // gcode: X:135.000000 Y:120.000000 Z:9.226708 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok
Send: probe_accuracy probe_speed=2 samples=12
Recv: // PROBE_ACCURACY at X:135.000 Y:120.000 Z:10.000 (samples=12 retract=4.000 speed=2.0 lift_speed=2.0)
Recv: // probe at 135.000,120.000 is z=-0.146500
Recv: // probe at 135.000,120.000 is z=-0.124000
Recv: // probe at 135.000,120.000 is z=-0.106500
Recv: // probe at 135.000,120.000 is z=-0.074000
Recv: // probe at 135.000,120.000 is z=-0.054000
Recv: // probe at 135.000,120.000 is z=-0.031500
Recv: // probe at 135.000,120.000 is z=-0.031500
Recv: // probe at 135.000,120.000 is z=-0.004000
Recv: // probe at 135.000,120.000 is z=0.023500
Recv: // probe at 135.000,120.000 is z=0.048500
Recv: // probe at 135.000,120.000 is z=0.073500
Recv: // probe at 135.000,120.000 is z=0.083500
Recv: // probe accuracy results: maximum 0.083500, minimum -0.146500, range 0.230000, average -0.028583, median -0.031500, standard deviation 0.073057
Recv: ok
Send: GET_POSITION
Recv: // mcu: stepper_x:3625 stepper_y:-801 stepper_z:397
Recv: // stepper: stepper_x:135.000000 stepper_y:120.000000 stepper_z:0.083500
Recv: // kinematic: X:135.000000 Y:120.000000 Z:0.083500
Recv: // toolhead: X:135.000000 Y:120.000000 Z:4.083500 E:0.000000
Recv: // gcode: X:135.000000 Y:120.000000 Z:3.310208 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok
```

Not sure if my issue is the same, but it seems to be much worse on my setup.

I ran the diagnostic commands you mentioned above. As far as I know, my code is pristine from the repo, but if it's not I can pull a fresh copy and run it again. The logs should be otherwise empty.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5790214/klippy.log)

[1 more reply]