Klipper3d/klipper #2510: MIN/MAX offsets

Coffee0297, 2020-02-15:
hello kevin, 

Why cant we agree on 1 type of measurements to do the min/max on? ie. the bed_mesh min/max refers to __probe__ position. 
__But__ the screws_tilt_ajust screw pos is the __nozzle__. 
Its kinda annoying to all the time have to switch between measurements just bc there is not 1 standard. 
The z tilt is also in the nozzle coordinates.  It seems to me some just wanted to redo the bed_mesh and for some reason chose to use probe pos instead of nozzle as we are used to. Either everything are gonna change to probe pos or all are gonna change to nozzle.

 __Not both__.

My 2 cents.

-Tonny

Fake log to make bot happy. 
[klippy (2).log](https://github.com/KevinOConnor/klipper/files/4208457/klippy.2.log)

GarthSnyder, 2020-02-17:
I find this a bit irksome as well and agree with your general point, although I’m not sure a universal default is the right solution. Some things are more naturally dealt with in nozzle space and others in probe space.

I would suggest instead:

- Where feasible and meaningful, commands should allow the use of either coordinate system.

- There should be a standard way to specify which coordinate system you are using, e.g., through “coordinates: nozzle“ versus “coordinates: probe“, or an optional decorator such as “screw1: P50,50”.

- The default in each context should be whichever coordinate system does not require the user to do their own offset arithmetic in the average case. (So, SCREWS_TILT_CALCULATE should be using probe coordinates by default, even though it’s a good example of a context where nozzle coordinates are sometimes useful.)

The implementation burden for this should be pretty low since most contexts where you might want a choice of coordinate system are already using ProbePointsHelper, which supports both.

I have a PR coming to change the SCREWS_TILT_CALCULATE default, but if there’s interest I could do a more general sweep.

Coffee0297, 2020-02-18:
I think @KevinOConnor need to chime in with his thoughts. 

i just think its a little weird we have to keep track of 2 coordinates in the same config.. 
so a option to change all coordinates used to either nozzle or probe would be nice.
ofc its not a breaking thing at all so its not in a rush. i just think it has to be taken in to consideration 

-Tonny

jakep82, 2020-02-18:
This discussion already occurred [here](https://github.com/KevinOConnor/klipper/pull/2295) when the decision was made to change the default coordinate system for probes, and this change was made in response to numerous complaints that probe offsets were not automatically accounted for when configuring bed_mesh.

Coffee0297, 2020-02-18:
so do that mean all coordinates are gonna change to the probe pos? i can't seem to see if it was global they wanted to change it?

-Tonny

jakep82, 2020-02-18:
The other modules you referenced were written by other people, and I'm not sure if they intend to update them.  It would probably be best to standardize on probe coordinates for any operation that uses the probe, but it's unclear to me when or if those other extras will be updated.

GarthSnyder, 2020-02-18:
This looks like an issue @Arksine might have a perspective on as well.

Arksine, 2020-02-19:
I can explain why the change was made to bed_mesh, and provide my opinion as to why it isn't quite as straight forward for the other modules that probe the bed.  At this point I'm not sure if there are any plans to add probe relative points to the other modules.

With regards to bed_mesh, @jakep82 is correct that the change was made due to confusion over the expected mesh location.  Users transitioning from Marlin and unfamiliar with Klipper had difficulty understanding how to configure bed_mesh to get desired results.  Consider that many users home Z toward the center of the bed, and they like to probe a mesh with a midpoint that is also centered.  This is easier to calculate now, as mesh_min and mesh_max refer to the actual mesh location.

From a configuration perspective, making this change to bed_mesh was relatively simple.   Because bed_mesh generates its points I needed break the previous configuration by renaming min_point and max_point to mesh_min and mesh_max.  It wasn't a huge ask for current users to recalculate two coordinates.    This may not be the same for the other modules, the only other one I have used extensively is Quad Gantry Level which would require four adjustments.   I'm not sure if its desirable, but you could probe a significant number of points for bed_tilt and z_tilt.

Also, I'm not sure that it makes sense to rename the 'points' option, however if we make this change we would need to either break the current configuration or make it optional.

KevinOConnor, 2020-02-24:
FWIW, I generally print on glass and so I don't currently use bed mesh.  As a result, I don't have any particularly strong feelings on this.

I don't think of this as "nozzle coordinates" vs "probe coordinates".  Instead, I think of it as "nozzle coordinates during probing" vs "nozzle coordinates during printing".  That is, if you think of the config file coordinates as the locations to adjust the nozzle during a print (and not as the location of the nozzle during the probing routine) then it makes more sense to specify the valid coordinate range of the bed/print.

I discussed this with @Arksine a few months back - we both observed a lot of confusion from users with the previous configuration of mesh coordinates.  At that time, we both agreed that a change would be a good idea.  YMMV.

It's not clear to me that changing the other probing routines makes sense.  I'm not aware of anyone currently working on changing these other probing routines.

-Kevin

Coffee0297, 2020-02-24:
i can just see that in the last week we have had 5 users on the discord,that cant figure out why the mesh is trying to go out of bed. which seems to me that they either don't know(docs?) or think its non intuitive bc all other probes are in nozzle ordinates.
i would personally like an option where you could set __ALL__ offsets to 1 or the other "offsets"
just my 2 cents... 

-Tonny

AmenoBars, 2020-02-24:
> 
> 
> cant figure out why the mesh is trying to go out of bed. which seems to me that they either don't know(docs?) 

This was the iffy bit for me. I couldn't find anything in the docs that told me that the bedmesh max size was +/- the probe offset. Makes sense now that I know, but before that I was going down the route of the bedsize not being correct

Arksine, 2020-02-24:
Its documented, with illustrations, in [Bed_mesh.md](https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Mesh.md).  It was also documented in Config_Changes.md for current users.

JohnEdwa, 2020-02-26:
A rather simple, but maybe not the most elegant solution, would be to simply have both - add new config items to the modules that specify the direct probe coordinates without removing the old nozzle based ones. Nothing changes for current users, and future ones can use whichever they prefer.

For example, keep the `[bed_screws] screw1:` as the nozzle coordinate, and add `screw1_probe:` that, obviously, is the position of the probe. Nothing breaks, and users get more flexibility.

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like the conversation has concluded.

-Kevin

closed 2020-03-21 by KevinOConnor