Klipper3d/klipper #2511: Pressure advance tuning error; 'ADVANCE must have minimum of 0.0'

3nigmatics, 2020-02-15:
I've updated to the latest Klipper and was running the tuning tower print for PA calibration.  

The printer shuts down and outputs the following error:
13:26:02 Starting tuning test (start=0.000000 factor=0.020000)
13:26:11 Error on 'SET_PRESSURE_ADVANCE ADVANCE=-0.005400000': ADVANCE must have minimum of 0.0

What am I missing here?

My gcode start is as follows:
;M201 X16000.00 Y16000.00 Z100.00 E5000.00 ;Setup machine max acceleration
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate
SET_PRESSURE_ADVANCE ADVANCE=0
SET_GCODE_OFFSET Z=-.55 MOVE=1
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=1 ACCEL=500
TUNING_TOWER COMMAND=SET_PRESSURE_ADVANCE PARAMETER=ADVANCE START=0 FACTOR=.020
G28 ;Home

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up

[klippy.log](https://github.com/KevinOConnor/klipper/files/4208517/klippy.log)

Hywelmartin, 2020-02-15:
issue the command during the first layer...

TUNING_TOWER COMMAND=SET_PRESSURE_ADVANCE PARAMETER=ADVANCE START=0 FACTOR=.020

3nigmatics, 2020-02-16:
> issue the command during the first layer...
> 
> TUNING_TOWER COMMAND=SET_PRESSURE_ADVANCE PARAMETER=ADVANCE START=0 FACTOR=.020

I issued the command during the first layer and the same error occurred:

TUNING_TOWER COMMAND=SET_PRESSURE_ADVANCE PARAMETER=ADVANCE START=0 FACTOR=.020
Warning:  Starting tuning test (start=0.000000 factor=0.020000)
Error:  Error on 'SET_PRESSURE_ADVANCE ADVANCE=-0.001000000': ADVANCE must have minimum of 0.0
Error:  Error on 'SET_PRESSURE_ADVANCE ADVANCE=-0.001000000': ADVANCE must have minimum of 0.0
Warning:  Klipper state: Shutdown

I will try it on layer 2 in case its an issue with my Z offset

3nigmatics, 2020-02-16:
> I issued the command during the first layer and the same error occurred:
>
> TUNING_TOWER COMMAND=SET_PRESSURE_ADVANCE PARAMETER=ADVANCE START=0 FACTOR=.020
> Warning: Starting tuning test [...]

The command works so long as Z was above .1mm or so.  This seems like a bug to me..

KevinOConnor, 2020-02-18:
This github issue tracker is only for use with code from this repo.  You'll either need to reproduce the issue with the pristine code (no dwc2) and provide the corresponding log, or you'll need to report the issue to the dwc2 developers.

-Kevin

dmbutyugin, 2020-02-20:
I think the problematic part of gcode is this one:
```
SET_GCODE_OFFSET Z=-.55 MOVE=1
```
This adds an offset to each command that follows, effectively pushing all commands from the first layer below Z==0 point.

It seems like it is an unfortunate coincidence that `TUNING_TOWER` takes GCode offset into consideration when calculating the values for the parameters, however it is unclear how `TUNING_TOWER` could be improved in that regard - `SET_GCODE_OFFSET` could be used in different scenarios. @3nigmatics you can adjust `START=0.0054` to mitigate that effect.

KevinOConnor, 2020-02-20:
Yes, the TUNING_TOWER command does not take into account any Z offset added by SET_GCODE_OFFSET.  I suppose the tuning tower code could query the Z offset from the gcode class during its `calc_value()` operation.

-Kevin

PR #2881 (2020-05-24): extras: Tuning tower compensate z-offset
Tuning tower helper now takes care of printer z-offset.
This pull request fixes #2511
closed 2020-05-24 by KevinOConnor