Klipper3d/klipper #2512: BlTouch intermittent issues

Coffee0297, 2020-02-15:
so here is another log from my intermittent bltouch issues. When can we expect to get a fix?
im kinda tired to try 5 times just to get a bed mesh to complete. 
all hardware is okay. its not a hardware issue.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4208519/klippy.log)

-Tonny

Coffee0297, 2020-02-15:
heres another one 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4208534/klippy.log)

ld3300, 2020-02-16:
Can you give more information about what you have tried? Can you tell us what tests confirm hardware is not related?  I wouldn't expect anyone to be able to answer a timeline question for a fix for an issue that is intermittent.  It may take some time and more examples to narrow down the issue.

Coffee0297, 2020-02-17:
Kevin allready knows, from issue #1938

ld3300, 2020-02-17:
Looks like you understand the problem, can you find the issue in the source, I'll help fix it.

KevinOConnor, 2020-02-18:
Unfortunately, your first log is not a complete log, so there isn't much that can be done with it.

The second log indicates that the bltouch signal was successfully received and a corresponding pin_up command was implemented within 105ms of the event:
```
mcu 'Z': Receive: 73 6249.796985 6249.774356 9: seq: 17(8070), endstop_state oid=4 homing=0 pin_value=1
...
mcu 'Z': Sent 41 6249.801739 6249.801739 15: seq: 1b(8075), schedule_soft_pwm_out oid=3 clock=2770201386(6249.902043) on_ticks=147500
```
It is not known why the bltouch transitioned into an error state in this situation.  This is not related to the timing issues reported in #1938 (the timing race did not occur here so fixing them would not help).

Perhaps @FanDjango has some insight, but I don't know what would have been needed to prevent the bltouch from entering an error state.  Maybe this particular bltouch is not working properly.

-Kevin

FanDjango, 2020-02-18:
@Coffee0297 
Let me say first, I have read through #1938, ok?
I saw you saying that Marlin works, but I myself do not think holding the PWM is the solution, as there are other things that differ as well. While I was coding the new BLTouch stuff in Marlin I experimented with **not holding** the PWM and noticed no difference.

Here are some things I would ask you to do in order to (try to) isolate the problem a little bit, if I may:

Please use the newest **master** from here, so that in the following sequence of tests, you can at one point use the new option `stow_on_each_sample`, as this will help isolate the problem. On each step, save the klippy log.

**PROBE_ACCURACY** Test - just probing, no movement in between, better than the mesh operation for testing initially:

Step 1: Use `stow_on_each_sample=True` (which is the default) and do a `PROBE_ACCURACY SAMPLES=100` and report the result - or if it fails and by when, and the klippy log

Step 2: Do the same but use `stow_on_each_sample=False`, same report please

The difference in behaviour between those two should help pinpointing **where** a possible timing issue might be found - since there are no more commands being sent in the Step 2 scenario.

Step 3: Add a parameter `lift_speed: 20` to your `[bltouch]` section. 20 is the value of your max Z speed in the kinematics section., then **repeat the two tests above** with higher lift speed and report.

And from that information, we can progress onwards to testing some simple timing changes. That would entail modifictions to `bltouch.py` and we would first need to discuss how we would organize that, since I don't know how much effort you are prepared to put into this. I am sure we can progress on this somehow.

Coffee0297, 2020-02-18:
Hmm that's weird... i thought most of bl issues where from timing. 
I'm running a 100 probe test on old system now to be sure it still occurs, and it seems like it don't. 
Is klipper really sensitive over emi? 

Im more than happy to help :D i just wanna print like we all do 👍 

-Tonny

FanDjango, 2020-02-18:
They are very probably due to some timing issue, or a command sequence issue, but **not** due to continuous or not continuous PWM.

FanDjango, 2020-02-18:
If the "100 probe test" works, then that does not mean the bed mesh **will** work, it just tells me where the problem **isn't**. Let's wait and see your results as per the steps described above

Coffee0297, 2020-02-18:
yea the 100 probe test works on "old" klipper (version im running now) ill upgrade to the newest and try the provided steps, thanks.

and i dont think that its a directly pwm issue, i jsut found it weird that marlin keeps it on and that antslabs recomends that :D 

-Tonny

Coffee0297, 2020-02-18:
test 1 faild with new software
heres a log: 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4219576/klippy.log)

-Tonny

FanDjango, 2020-02-18:
Ok, 30 or 40 probes and then "boom". Now step 2...?

Coffee0297, 2020-02-18:
i assume klipper flushes and start a new log on m112?

Coffee0297, 2020-02-18:
step 2 completed no issue

-Tonny

FanDjango, 2020-02-18:
Ok, **very** good. And now like step1 again (which failed), but with higher lift speed. I suppose you could also maybe use your mobile and make a small video of the probe going up and down, maybe, but I don't know if that is really needed right now. I am already looking at the code to check a few things

Coffee0297, 2020-02-18:
https://photos.app.goo.gl/uEJNeAbwzHpRcV3E6

This is step 3.1 it completed fine 
Gooing 3.2 now
-Tonny

FanDjango, 2020-02-18:
Fine, I see the problem in the video, I think. Step 4, I think, will also look ok. So in my opinion, until I find a real solution, you might do the following: Use the configuration of Step 4 and now try a **bed mesh** sequence.

Coffee0297, 2020-02-18:
3.2/4 went fine 👍 so what is the issue? 
imma make a mesh now 

-Tonny

FanDjango, 2020-02-18:
In a nutshell, timing is the issue.
Not all BLTouch versions are the same, and there are also clones, ok, **but** this one, if you deploy in **touch mode**, the pin goes down. If you then push up the pin (i.e. trigger it) and do nothing (keep it triggered for too long), the probe will go into error mode. You need to retract the pin **or** move up (= untrigger) as quickly as possible or it will go into an error mode. 
If you set `stow_on_each_probe` to false, the time-consuming stow will be missing, and it goes up earlier, also if the lift speed is higher, it will also benefit the timing how fast it will be untriggered.
I am now searching and testing in the code for a way to make this better. I will need some time for that, though.

Coffee0297, 2020-02-18:
mesh went fine so ill make a "real" mesh with 12,12,3 
that makes good sense, but the retrying somtimes saves it and sometimes it dont. wich is a little weird.
Thanks so far

-Tonny

FanDjango, 2020-02-18:
The retrying is a totally separate piece of logic that also has it's own problems which need to be fixed. For now, it is a good idea to try to get to a point where you do not need the retry too often, but I am also working on that part. 
Since your changes are just **reducing** the number of DEPLOY commands which then can fail on the trigger, any of your mesh or other sequences might fail on the **first** deploy and the subsequent trigger **any time by chance** but no longer in the middle of a sequence

Coffee0297, 2020-02-18:
okay thats nice to know. so if i do a big mesh it dont die in the middle. great

-Tonny

FanDjango, 2020-02-18:
> so if i do a big mesh it dont die in the middle
Well, it would be nice if you could raise even faster than 20. Mine can handle 30, for example. But apart from that, until I find the best way to actually improve the code, there is a slight risk left over, of course, that it might happen in the middle of a print. Using SW mode should help, also, but that's in the pipeline somewhere, because older probes or clones **should not** use SW mode. SO it needs to be made into an option, maybe, but like I said, I need to go step by step.

Is your big mesh running?

Coffee0297, 2020-02-18:
how wuld it happen in a middle of a print? i don't use a probe there? 
and my bed is heating so it will take a wile before i set it gooing

-Tonny

FanDjango, 2020-02-18:
I am going to quote myself, so sorry, but the point needs to be reiterated:

> Not all BLTouch versions are the same, and there are also clones, ok, but this one, if you deploy in touch mode, the pin goes down. If you then push up the pin (i.e. trigger it) and do nothing (keep it triggered for too long), the probe will go into error mode, because its currently active command (even if the PWM is stopped) is to be deployed. You need to retract the pin or move up (= untrigger, which is equal to being deployed) as quickly as possible or it will go into an error mode.
If you set stow_on_each_probe to false, the time-consuming stow will be missing, and it goes up earlier, also if the lift speed is higher, it will also benefit the timing how fast it will be untriggered.

The other issue contains exactly this observation by some of the others contributing there. It is not something new that I discovered **just now**.

FanDjango, 2020-02-18:
> how wuld it happen in a middle of a print?

Type. **middle of a mesh probe**

Coffee0297, 2020-02-18:
ahh makes sense. yea i know that anstlabs are kinda an ass with the documentation of changes of older hardware, so a option is differently needed... 

-Tonny

FanDjango, 2020-02-18:
I am going to need some time to work some more on this. I estimate 2 days. That ok?

Coffee0297, 2020-02-18:
yea that no issue at all for me, take your time. i only have 1 printer atm running, so no rush. you can just ping me if you need any more logs or testing of beta builds

-Tonny

Coffee0297, 2020-02-18:
i just got the error again with 3.2/4 combiation in the bed mesh

-Tonny

Coffee0297, 2020-02-18:
heres a log, i dont know if its complete bc klipper frose totaly and i had to restart service
[klippy.log](https://github.com/KevinOConnor/klipper/files/4219869/klippy.log)

FanDjango, 2020-02-18:
What's your hardware?

Coffee0297, 2020-02-18:
for klipper? a dual core amd pc. i think it frose as in klipper was thinking it still was dooing mesh, and then octo closed the connection bc no recived data

-Tonny

FanDjango, 2020-02-18:
Klipper on PC together with Octoprint via USB to...
Controller board?

FanDjango, 2020-02-18:
I've got a first very small code change for you, can you do one manually in `btlouch.py`?

Coffee0297, 2020-02-18:
1x skr v1.1 for Z and E including bltouch
1x mks base for X,Y and heating
-Tonny

Coffee0297, 2020-02-18:
yea no problem 
-Tonny

FanDjango, 2020-02-18:
Sorry, it will take longer. I will now start some testing here.

Coffee0297, 2020-02-18:
again, take your time. no rush

-Tonny

FanDjango, 2020-02-18:
Please test `[bltouch]` `pin_up_reports_not_triggered = False` instead of `true` and test the 100 accuracy 3.2(4) thing again, tell me what has changed. This turns **off** probe raise verify, might speed things up...

FanDjango, 2020-02-18:
Second thing to please test on **your setup**:
Find this place in `bltouch.py`:
[image]
and test this change:
`duration=0.160)`
but don't change `pin_move_time` in your config, just in that place shown above

Coffee0297, 2020-02-18:
im testing now
-Tonny

Coffee0297, 2020-02-18:
it faild again. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4220349/klippy.log)

FanDjango, 2020-02-18:
Interesting. Different error on mesh probes:
```
probe at 109.038,56.800 is z=4.058750
Stats 275155.8: gcodein=95 mcu: mcu_awake=0.008 mcu_task_avg=0.000087 mcu_task_stddev=0.000095 bytes_write=10329 bytes_read=31738 bytes_retransmit=9 bytes_invalid=6 send_seq=527 receive_seq=527 retransmit_seq=2 srtt=0.009 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15878863 Z: mcu_awake=0.005 mcu_task_avg=0.000013 mcu_task_stddev=0.000016 bytes_write=137431 bytes_read=32527 bytes_retransmit=9 bytes_invalid=0 send_seq=4613 receive_seq=4613 retransmit_seq=2 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001703 adj=100765029 heater_bed: target=0 temp=26.5 pwm=0.000 print_time=248.926 buffer_time=0.000 print_stall=0 extruder: target=0 temp=27.4 pwm=0.000
Stats 275156.8: gcodein=95 mcu: mcu_awake=0.008 mcu_task_avg=0.000087 mcu_task_stddev=0.000095 bytes_write=10335 bytes_read=31852 bytes_retransmit=9 bytes_invalid=6 send_seq=528 receive_seq=528 retransmit_seq=2 srtt=0.008 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15878862 Z: mcu_awake=0.005 mcu_task_avg=0.000014 mcu_task_stddev=0.000017 bytes_write=137877 bytes_read=32638 bytes_retransmit=9 bytes_invalid=0 send_seq=4630 receive_seq=4630 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001700 adj=100765036 heater_bed: target=0 temp=26.5 pwm=0.000 print_time=249.676 buffer_time=0.000 print_stall=0 extruder: target=0 temp=27.4 pwm=0.000
Stats 275157.8: gcodein=95 mcu: mcu_awake=0.008 mcu_task_avg=0.000087 mcu_task_stddev=0.000095 bytes_write=10341 bytes_read=31966 bytes_retransmit=9 bytes_invalid=6 send_seq=529 receive_seq=529 retransmit_seq=2 srtt=0.009 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15878860 Z: mcu_awake=0.005 mcu_task_avg=0.000014 mcu_task_stddev=0.000017 bytes_write=138261 bytes_read=32776 bytes_retransmit=9 bytes_invalid=0 send_seq=4650 receive_seq=4650 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001699 adj=100765047 heater_bed: target=0 temp=26.5 pwm=0.000 print_time=250.603 buffer_time=0.105 print_stall=0 extruder: target=0 temp=27.3 pwm=0.000
probe at 109.038,56.800 is z=4.061250
Stats 275158.8: gcodein=95 mcu: mcu_awake=0.008 mcu_task_avg=0.000087 mcu_task_stddev=0.000095 bytes_write=10347 bytes_read=32066 bytes_retransmit=9 bytes_invalid=6 send_seq=530 receive_seq=530 retransmit_seq=2 srtt=0.008 rttvar=0.005 rto=0.026 ready_bytes=0 stalled_bytes=0 freq=15878859 Z: mcu_awake=0.005 mcu_task_avg=0.000014 mcu_task_stddev=0.000017 bytes_write=138956 bytes_read=32875 bytes_retransmit=9 bytes_invalid=0 send_seq=4664 receive_seq=4664 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001704 adj=100765078 heater_bed: target=0 temp=26.4 pwm=0.000 print_time=251.697 buffer_time=0.000 print_stall=0 extruder: target=0 temp=27.4 pwm=0.000
BLTouch failed to deploy
Stats 275159.8: gcodein=100 mcu: mcu_awake=0.007 mcu_task_avg=0.000074 mcu_task_stddev=0.000061 bytes_write=10353 bytes_read=32193 bytes_retransmit=9 bytes_invalid=6 send_seq=531 receive_seq=531 retransmit_seq=2 srtt=0.008 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15878858 Z: mcu_awake=0.005 mcu_task_avg=0.000014 mcu_task_stddev=0.000017 bytes_write=139251 bytes_read=32978 bytes_retransmit=9 bytes_invalid=0 send_seq=4677 receive_seq=4677 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001702 adj=100765145 heater_bed: target=0 temp=26.0 pwm=0.000 print_time=251.951 buffer_time=0.000 print_stall=0 extruder: target=0 temp=27.3 pwm=0.000
Stats 275160.8: gcodein=100 mcu: mcu_awake=0.007 mcu_task_avg=0.000074 mcu_task_stddev=0.000061 bytes_write=10359 bytes_read=32307 bytes_retransmit=9 bytes_invalid=6 send_seq=532 receive_seq=532 retransmit_seq=2 srtt=0.009 rttvar=0.005 rto=0.028 ready_bytes=0 stalled_bytes=0 freq=15878857 Z: mcu_awake=0.005 mcu_task_avg=0.000014 mcu_task_stddev=0.000017 bytes_write=139257 bytes_read=32994 bytes_retransmit=9 bytes_invalid=0 send_seq=4678 receive_seq=4678 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001701 adj=100765123 heater_bed: target=0 temp=26.3 pwm=0.000 print_time=251.951 buffer_time=0.000 print_stall=0 extruder: target=0 temp=27.3 pwm=0.000
Stats 275161.8: gcodein=100 mcu: mcu_awake=0.007 mcu_task_avg=0.000074 mcu_task_stddev=0.000061 bytes_write=10365 bytes_read=32407 bytes_retransmit=9 bytes_invalid=6 send_seq=533 receive_seq=533 retransmit_seq=2 srtt=0.009 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15878856 Z: mcu_awake=0.003 mcu_task_avg=0.000012 mcu_task_stddev=0.000015 bytes_write=139263 bytes_read=33024 bytes_retransmit=9 bytes_invalid=0 send_seq=4679 receive_seq=4679 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001699 adj=100765085 heater_bed: target=0 temp=26.3 pwm=0.000 print_time=251.951 buffer_time=0.000 print_stall=0 extruder: target=0 temp=27.4 pwm=0.000
Stats 275162.8: gcodein=100 mcu: mcu_awake=0.007 mcu_task_avg=0.000074 mcu_task_stddev=0.000061 bytes_write=10371 bytes_read=32521 bytes_retransmit=9 bytes_invalid=6 send_seq=534 receive_seq=534 retransmit_seq=2 srtt=0.009 rttvar=0.004 rto=0.027 ready_bytes=0 stalled_bytes=0 freq=15878855 Z: mcu_awake=0.003 mcu_task_avg=0.000012 mcu_task_stddev=0.000015 bytes_write=139269 bytes_read=33039 bytes_retransmit=9 bytes_invalid=0 send_seq=4680 receive_seq=4680 ret [... cut ...] mcu_task_avg=0.000012 mcu_task_stddev=0.000015 bytes_write=139281 bytes_read=33070 bytes_retransmit=9 bytes_invalid=0 send_seq=4682 receive_seq=4682 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001694 adj=100765054 heater_bed: target=0 temp=26.3 pwm=0.000 print_time=251.951 buffer_time=0.000 print_stall=0 extruder: target=0 temp=27.4 pwm=0.000
```
I see that was also the case earlier. I will research some more. Did you notice a change in the behaviour?

FanDjango, 2020-02-18:
Find this place:
[image]
and put `duration=0.750` instead. We will see... and get to the bottom of this.
And then I will continue tomorrow...

KevinOConnor, 2020-02-18:
FYI, in the log at https://github.com/KevinOConnor/klipper/issues/2512#issuecomment-587471973 you hit the command transmission race condition problem:
```
mcu 'Z': Receive: 83 265908.360487 265908.341159 9: seq: 16(2501), endstop_state oid=4 homing=0 pin_value=1
...
mcu 'Z': Sent 78 265908.866698 265908.866698 15: seq: 1b(2507), schedule_soft_pwm_out oid=3 clock=1396499027(265908.967363) on_ticks=147500
```
That is, the Klipper micro-controller code recognized the event and stopped the stepper motors, but it took the Klipper host code over 600ms to send the corresponding pin_up command.

FWIW, the log for this event looks notably different than the event at https://github.com/KevinOConnor/klipper/issues/2512#issuecomment-587244720 .

-Kevin

FanDjango, 2020-02-18:
Also try removing these lines:
[image]
Note: THIS WILL NOT BE A SOLUTION, THIS IS ONLY DEBUGGING on your setup to find the problem.

FanDjango, 2020-02-18:
@Coffee0297 @KevinOConnor I noticed too that probing the mesh raises a different problem than the simple repeated accuracy probing. If it is the race condition in this case, then I suppose @Coffee0297 you don't need to do the stuff I just suggested. In that case, I should concentrate on making the recovery succeed / finding out why it does not.

Coffee0297, 2020-02-18:
tried the 0.750 and i get error again
[klippy.log](https://github.com/KevinOConnor/klipper/files/4220624/klippy.log)
im trying the deletion of those lines now

-Tonny

Coffee0297, 2020-02-18:
the delete of that made the probe stay down after a reset of the bltouch? 

-Tonny

Coffee0297, 2020-02-18:
i got it to mesh still gave an error

Coffee0297, 2020-02-18:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4220673/klippy.log)
another error than last tho

-Tonny

FanDjango, 2020-02-18:
Ok, we need to slow down again - I see that @KevinOConnor has put out a PR #2532 which would be a logical step to check.

Go back to your 3.2(4) stage and wait a moment. I can make a branch and put the #2532 in and tell you where to clone it from, if you want. I will also install that here on my side, and then we can start again from the same status tomorrow.

That ok?

Coffee0297, 2020-02-18:
yea that will be fine, the bltouch.py will be revertet if i pull new code right?

-Tonny

FanDjango, 2020-02-18:
Yup 👍

Coffee0297, 2020-02-18:
great :D you can just ping me wen you are ready for me to test. are you on discord? i might be faster to reach there

-Tonny

FanDjango, 2020-02-19:
Ok, discord: https://discord.gg/7zHMJP

Coffee0297, 2020-02-19:
@KevinOConnor @FanDjango 
i have done some testing with #2532 and it seems like that fixed the bed_mesh issue.

the probe accuracy issue was mitigated by

```
stow_on_each_sample: False
lift_speed: 20
```

in [bltouch] 

ill test some more today and report back..
On a side note it seems like this also helps with accuracy of the probe, 
im down to a SDEV = 0.0015 at 10 probes where before this it was SDEV = 0.01-2

-Tonny

Coffee0297, 2020-02-19:
I have done 5 12x12x3 probes (2160 probes) in bed mesh mode, and i have not encounterd the issue after the fix.

so i wuld say #2532 is a go
-Tonny

KevinOConnor, 2020-02-24:
Okay, thanks.  I committed #2532 to the master branch.

-Kevin

[1 more reply]

closed 2020-03-05 by KevinOConnor