Klipper3d/klipper #2513: Extruder stops moving

smark-, 2020-02-15:
I just installed a G5 flex extruder which has a 40:1 gear ration, thus a very small value for step_distance. I think that's a bug that makes the extruder stop moving during the print. I've been able to reproduce the issue myself by issuing a set of G1 commands to the extruder on Octoprint. If I increase the step size then the issue seem to disappear
[klippy.log](https://github.com/KevinOConnor/klipper/files/4208938/klippy.log)

Arksine, 2020-02-15:
I would hazard a guess that its most likely the driver.  Trinamic drivers have auto-cutoff mechanisms when they detect an error condition.  If you can reproduce you could execute the DUMP_TMC gcode to see if GSTAT is has logged an error:

```
DUMP_TMC STEPPER=extruder
```

smark-, 2020-02-17:
Problem solved, it was indeed the Trinamic auto-cutoff mechanisms kicking in. In particular I experienced this issue, even if I'm using TMC2130 drivers: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#my-tmc-motor-driver-turns-off-in-the-middle-of-a-print
The solution was to configure the drivers in spreadcycle mode.

closed 2020-02-17 by smark-