Klipper3d/klipper #2514: Replicape MCU Crashes when using Servo for BLtouch

vvvteddybearvvv, 2020-02-15:
I am attempting to setup  a BLtouch on a Replicape board.

I used the following config Settings
`[bltouch]
sensor_pin: ^!P9_18
control_pin: replicape:servo1
pin_move_time: 0.675
pin_up_reports_not_triggered: True
pin_up_touch_mode_reports_triggered: false
z_offset:.03`

I am getting the following error MCU 'host' shutdown: Unable to config pwm device
****
[klippy.log](https://github.com/KevinOConnor/klipper/files/4209044/klippy.log)

BlackStump, 2020-02-16:
comment out your [probe] either use probe or bltouch not both

vvvteddybearvvv, 2020-02-16:
New Log file with [Probe] Commented out   Same error   MCU 'host' shutdown: Unable to config pwm device
[klippy.log](https://github.com/KevinOConnor/klipper/files/4209275/klippy.log)

BlackStump, 2020-02-16:
have a read of this

https://github.com/KevinOConnor/klipper/pull/1443

vvvteddybearvvv, 2020-02-16:
The discussion is a bit over myhead. looks like there were some linux commands to run. Ive attached the output here. 

```
root@kamikaze:~#` ls -l /sys/class/pwm/
total 0
lrwxrwxrwx 1 root root 0 Feb 15 22:02 pwm-0:0 -> ../../devices/platform/ocp/4830         0000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0
lrwxrwxrwx 1 root root 0 Feb 15 22:02 pwm-1:0 -> ../../devices/platform/ocp/4830         0000.epwmss/48300200.pwm/pwm/pwmchip1/pwm-1:0
lrwxrwxrwx 1 root root 0 Feb 15 22:02 pwm-1:1 -> ../../devices/platform/ocp/4830         0000.epwmss/48300200.pwm/pwm/pwmchip1/pwm-1:1
lrwxrwxrwx 1 root root 0 Feb 15 22:02 pwm-3:0 -> ../../devices/platform/ocp/4830         2000.epwmss/48302100.ecap/pwm/pwmchip3/pwm-3:0
lrwxrwxrwx 1 root root 0 Feb 15 22:02 pwm-4:0 -> ../../devices/platform/ocp/4830         2000.epwmss/48302200.pwm/pwm/pwmchip4/pwm-4:0
lrwxrwxrwx 1 root root 0 Feb 15 22:02 pwm-4:1 -> ../../devices/platform/ocp/4830         2000.epwmss/48302200.pwm/pwm/pwmchip4/pwm-4:1
lrwxrwxrwx 1 root root 0 Feb 15 22:02 pwm-6:0 -> ../../devices/platform/ocp/4830         4000.epwmss/48304100.ecap/pwm/pwmchip6/pwm-6:0
lrwxrwxrwx 1 root root 0 Feb 15 22:02 pwm-7:0 -> ../../devices/platform/ocp/4830         4000.epwmss/48304200.pwm/pwm/pwmchip7/pwm-7:0
lrwxrwxrwx 1 root root 0 Feb 15 22:02 pwm-7:1 -> ../../devices/platform/ocp/4830         4000.epwmss/48304200.pwm/pwm/pwmchip7/pwm-7:1
lrwxrwxrwx 1 root root 0 Feb 15 22:01 pwmchip0 -> ../../devices/platform/ocp/483         00000.epwmss/48300100.ecap/pwm/pwmchip0
lrwxrwxrwx 1 root root 0 Feb 15 22:01 pwmchip1 -> ../../devices/platform/ocp/483         00000.epwmss/48300200.pwm/pwm/pwmchip1
lrwxrwxrwx 1 root root 0 Feb 15 22:01 pwmchip3 -> ../../devices/platform/ocp/483         02000.epwmss/48302100.ecap/pwm/pwmchip3
lrwxrwxrwx 1 root root 0 Feb 15 22:01 pwmchip4 -> ../../devices/platform/ocp/483         02000.epwmss/48302200.pwm/pwm/pwmchip4
lrwxrwxrwx 1 root root 0 Feb 15 22:01 pwmchip6 -> ../../devices/platform/ocp/483         04000.epwmss/48304100.ecap/pwm/pwmchip6
lrwxrwxrwx 1 root root 0 Feb 15 22:01 pwmchip7 -> ../../devices/platform/ocp/483         04000.epwmss/48304200.pwm/pwm/pwmchip7
```

```
root@kamikaze:~# ls -lH /sys/class/pwm/*
'/sys/class/pwm/pwm-0:0':
total 0
-r--r--r-- 1 root gpio 4096 Feb 15 22:01 capture
lrwxrwxrwx 1 root gpio    0 Feb 15 22:02 device -> ../../pwmchip0
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 duty_cycle
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 enable
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 period
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 polarity
drwxrwxr-x 2 root gpio    0 Feb 15 22:01 power
lrwxrwxrwx 1 root gpio    0 Feb 15 22:02 subsystem -> ../../../../../../../../class/pwm
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 uevent

'/sys/class/pwm/pwm-1:0':
total 0
-r--r--r-- 1 root gpio 4096 Feb 15 22:01 capture
lrwxrwxrwx 1 root gpio    0 Feb 15 22:02 device -> ../../pwmchip1
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 duty_cycle
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 enable
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 period
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 polarity
drwxrwxr-x 2 root gpio    0 Feb 15 22:01 power
lrwxrwxrwx 1 root gpio    0 Feb 15 22:02 subsystem -> ../../../../../../../../class/pwm
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 uevent

'/sys/class/pwm/pwm-1:1':
total 0
-r--r--r-- 1 root gpio 4096 Feb 15 22:01 capture
lrwxrwxrwx 1 root gpio    0 Feb 15 22:02 device -> ../../pwmchip1
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 duty_cycle
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 enable
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 period
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 polarity
drwxrwxr-x 2 root gpio    0 Feb 15 22:01 power
lrwxrwxrwx 1 root gpio    0 Feb 15 22:02 subsystem -> ../../../../../../../../class/pwm
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 uevent

'/sys/class/pwm/pwm-3:0':
total 0
-r--r--r-- 1 root gpio 4096 Feb 15 22:01 capture
lrwxrwxrwx 1 root gpio    0 Feb 15 22:02 device -> ../../pwmchip3
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 duty_cycle
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 enable
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 period
-rw-rw-r-- 1 root gpio 4096 Feb 15 22:01 polarity
drwxrwxr-x 2 root gpio    0 Feb 15 22:01 power
lrwxrwxrwx 1 root gpio    0 Feb 15 22:02 subsystem -> ../../../../../../../../class/pwm
[... 149 lines cut ...]
/sys/kernel/debug/pinctrl/pinctrl-handles:    type: MUX_GROUP controller pinctrl-single group: pinmux_P9_17_pwm_pin (41) function: pinmux_P9_17_pwm_pin (41)
/sys/kernel/debug/pinctrl/pinctrl-handles:  state: pwm
/sys/kernel/debug/pinctrl/pinctrl-handles:    type: MUX_GROUP controller pinctrl-single group: pinmux_P9_21_pwm_pin (51) function: pinmux_P9_21_pwm_pin (51)
/sys/kernel/debug/pinctrl/pinctrl-maps:device 48300200.pwm
/sys/kernel/debug/pinctrl/pinctrl-maps:device 48302200.pwm
/sys/kernel/debug/pinctrl/pinctrl-maps:device 48304200.pwm
/sys/kernel/debug/pinctrl/pinctrl-maps:device ocp:pwm_enable
/sys/kernel/debug/pinctrl/pinctrl-maps:state pwm
/sys/kernel/debug/pinctrl/pinctrl-maps:group pinmux_P9_15_pwm_pin
/sys/kernel/debug/pinctrl/pinctrl-maps:function pinmux_P9_15_pwm_pin
/sys/kernel/debug/pinctrl/pinctrl-maps:state pwm
/sys/kernel/debug/pinctrl/pinctrl-maps:group pinmux_P9_17_pwm_pin
/sys/kernel/debug/pinctrl/pinctrl-maps:function pinmux_P9_17_pwm_pin
/sys/kernel/debug/pinctrl/pinctrl-maps:state pwm
/sys/kernel/debug/pinctrl/pinctrl-maps:group pinmux_P9_21_pwm_pin
/sys/kernel/debug/pinctrl/pinctrl-maps:function pinmux_P9_21_pwm_pin
```
```
root@kamikaze:~# echo -n 0 > /sys/class/pwm/pwmchip0/export
-bash: echo: write error: Device or resource busy
root@kamikaze:~# echo -n 1 > /sys/class/pwm/pwmchip0/export
-bash: echo: write error: No such device
```

`

BlackStump, 2020-02-16:
maybe @jannau  can shed some light

KevinOConnor, 2020-02-18:
What OS image was installed on this Beaglebone?

-Kevin

jannau, 2020-02-20:
I guess newer kernel than the 4.4 kernel I develop the the servo support against. I'm currently away from the hardware and busy with moving so I probably won't have time to look into it soon.

BlackStump, 2020-02-20:
Debian 9.9 uses kernel 4.14

vvvteddybearvvv, 2020-02-21:
Sorry. ive been away from the hardware.. If memory serves me correctly it is \Ubuntu 18.04.3 LTS 4.14.108-ti-r124

KevinOConnor, 2020-03-21:
Any further updates on this?

-Kevin

vvvteddybearvvv, 2020-03-22:
Not from me. I have no idea what to do next.

KevinOConnor, 2020-04-15:
Okay, lets close this then.  The beaglebone linux distributions are too unstable to support them in general - only the distribution versions listed in https://www.klipper3d.org/beaglebone.html will work.

-Kevin

closed 2020-04-15 by KevinOConnor

jannau, 2020-04-18:
That image (Debian 9.9 2019-08-03 4GB SD IoT) come with kernel 4.14 .108-ti and is unfortunatel broken. There are two problems:
- the beaglebord kernel 4.14 and 4.19 have a patch which was never applied upstream which the sysfs names (https://github.com/beagleboard/linux/commit/8b395d68b2712512b8a027539c0c7acd969deafb). It is not in the 5.4 beagleboard kernel.
- more pwmchips are exported in the sysfs and correct one is now `pwmchip4` (follow `/sys/bus/platform/devices/48302200.pwm` for future reference). this unfortunately also requires a code change in extras/replicape.py

I will submit patches which fixes the problems for this kernel but both problems are annoying.

PR #2746 (2020-05-03): replicape: adapt servo0 channels for beagleboard lernel 4.14.108-ti
More pwmchips are exported in the sysfs and the correct one is now
pwmchip4. The corect device for the servo pwm channels is
/sys/bus/platform/devices/48302200.pwm
The beagleboard kernel 4.14 and 4.19 contain in addition a commit
which chnages the pwm control directories from 'pwmX' to 'pwm-Y:X'.
Since the commit is not in upstream Linux a config option for the
sysfs path is added. Since beaglebone+replicape is the main user
of the linux PWM support the config defaults to the non-standard
path. Fixes #2514.