Klipper3d/klipper #2515: STM32F103 Timeout on serial connect

cooked, 2020-02-15:
Hi,
I've installed Klipper cloning the master branch, and configured for the STM32F103.
I could then flash the microcontroller but when starting klipper I got a Timeout on serial connect.
Everything was working fine in release 0.8.0

Stef

[klippy.log](https://github.com/KevinOConnor/klipper/files/4209087/klippy.log)

BlackStump, 2020-02-16:
https://www.klipper3d.org/FAQ.html#wheres-my-serial-port

KevinOConnor, 2020-02-18:
Did you recompile the micro-controller code as part of the update?  If so, what Kconfig options did you set?

-Kevin

cooked, 2020-02-18:
Hi Kevin,
yes the uC code has been recompiled. Config file is attached

[config.txt](https://github.com/KevinOConnor/klipper/files/4220977/config.txt)

Stef

KevinOConnor, 2020-02-24:
Alas, I don't know why that would be.  I'm not aware of any issues in this area.  Make sure you are using the first serial port device on the stm32 (or update the config accordingly).

-Kevin

Fwirt, 2020-02-26:
I'm experiencing the same issue. FYI I'm running klippy on my laptop. Klipper is running on an SKR mini E3 (STM32F103) configured for serial port 2 (PA2/3) at 115200 baud and I'm using a PL2303 cable to connect. My config file has:

`[mcu]
serial: /dev/ttyUSB0
baud: 115200`

When trying to connect I get:

`Starting serial connect`
`Starting stk500v2 leave programmer sequence`
`Got '' from stk500v2`
`Timeout on serial connect`
`Starting stk500v2 leave programmer sequence`
`Wait for identify_response`
`Traceback (most recent call last):`
`  File "/home/fwirt/klipper/klippy/serialhdl.py", line 67, in _get_identify_data`
`    params = self.send_with_response(msg, 'identify_response')`
`  File "/home/fwirt/klipper/klippy/serialhdl.py", line 182, in send_with_response`
`    return src.get_response(cmd, self.default_cmd_queue)`
`  File "/home/fwirt/klipper/klippy/serialhdl.py", line 238, in get_response`
`    self.serial.raw_send_wait_ack(cmd, minclock, minclock, cmd_queue)`
`  File "/home/fwirt/klipper/klippy/serialhdl.py", line 174, in raw_send_wait_ack`
`    raise error("Serial connection closed")`
`error: Serial connection closed`

I know that the cable is connected correctly and that the drivers are working because in Marlin, using the same cable, pins, and baud rate I can connect and send commands via serial terminal (GNU screen). I have both firmwares built and have been flashing back and forth without making hardware changes.

Connecting to /dev/ttyUSB0 via GNU screen shows 4-5 bytes coming from the printer every few seconds. But for some reason klippy isn't detecting the printer or the printer isn't responding to its commands. Is there anything else I can try to help debug this?

I tried commenting out the stk500v2 code since I'm not using an stk500v2 but that had no effect.

FWIW Klipper works using USB serial, but it disconnects during the first couple layers of the print. I've tried 3 different mini usb cables (one as short as 1ft) and two different machines and have the same problem every time. From staring at dmesg output and hours of googling my suspicion is that I'm using a USB 3.0 port and the xhci_hcd driver doesn't like it, but I don't have any machines with USB 2.0 ports or a USB 2.0 hub sitting around to try it on. I thought that using a dedicated serial port would be more stable, but I can't verify that if I can't get it working...

Fwirt, 2020-02-28:
Ok good news: I'm new to klipper and was staying current on the master branch, but as @cooked mentioned if I downgrade to the 0.8.0 release archive then everything appears to work properly, so one of the commits between the last release and the present must have broken this. When I get some time, I'll replay some of the commits since then and see which one broke this.

KevinOConnor, 2020-02-29:
Unfortunately, there was an error in commit bd6c25c9.  Commit 31b2c3ea should improve serial support.

-Kevin

Fwirt, 2020-03-02:
Thanks! It's able to connect now, but it dies about 30 sec into the print with
```
got {'#receive_time': 116062.97084694267, '#name': u'starting', '#sent_time': 0.0}
MCU 'mcu' shutdown: Invalid oid type
clocksync state: mcu_freq=72000000 last_clock=25283099625 clock_est=(116033.822 23227235492 71998587.532) min_half_rtt=0.000173 min_rtt_time=115828.028 time_avg=116033.822(843.515) clock_avg=23227235492.821(60731886300.408) pred_variance=3937617.318
```
Attaching the log. Looks like right before it dies, there's a glut of stalled bytes. And to be clear, the exact same setup is working consistently on version 0.8.
[log.txt](https://github.com/KevinOConnor/klipper/files/4273895/log.txt)

KevinOConnor, 2020-03-04:
@Fwirt - one of my boards was showing a crash when spi was used on the stm32f103xb chip.  It was an unrelated issue to the serial issue.  I was able to track it down and revert the problematic commit (commit 121249c9).  You should retry with the latest code.

-Kevin

Fwirt, 2020-03-05:
That makes sense, I'm using an SPI temperature sensor. I've done 3 successful prints at 250000 baud so far, so that seems to have fixed it, at least for the hardware serial port. I haven't tried the USB connection. Thanks Kevin!

KevinOConnor, 2020-03-21:
I'm going to close this as it looks like the underlying issue is now fixed.

-Kevin

closed 2020-03-21 by KevinOConnor

spolti, 2021-07-11:
Was this issue fixed?

Getting this

`Extruder max_extrude_ratio=0.266081
mcu 'mcu': Starting serial connect
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 316, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/pi/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
error: mcu 'mcu': Serial connection closed
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
`

csharp2a, 2021-10-17:
I am still having this exact issue.  I have validated multiple times that the serial port is correct as well as rebuilt the firmware multiple times.

What is the status of this issue?  Thanks

mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 316, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/pi/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
error: mcu 'mcu': Serial connection closed

ofek256, 2021-11-11:
Having the same issue, using a pi zero 2 and a SKR mini e3 v1.2:

Extruder max_extrude_ratio=0.266081
mcu 'mcu': Starting serial connect
webhooks client 1970540632: New connection
webhooks client 1970540632: Client info {'program': 'Moonraker', 'version': 'v0.7.1-115-g7ff126a'}
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 316, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/pi/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
error: mcu 'mcu': Serial connection closed
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 316, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/pi/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
error: mcu 'mcu': Serial connection closed
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 316, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/pi/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
error: mcu 'mcu': Serial connection closed
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 316, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/pi/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
error: mcu 'mcu': Serial connection closed
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 316, in get_response
[... 257 lines cut ...]
CONFIG_HAVE_GPIO_HARD_PWM=y
CONFIG_HAVE_GPIO_BITBANGING=y
CONFIG_HAVE_STRICT_TIMING=y
CONFIG_HAVE_CHIPID=y
CONFIG_HAVE_STEPPER_BOTH_EDGE=y
CONFIG_INLINE_STEPPER_HACK=y
=======================
Build file /home/pi/klipper/klippy/../out/klipper.dict(6860): Fri Nov 12 00:44:16 2021
Last MCU build version: v0.10.0-122-gc179db3d
Last MCU build tools: gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-11+rpi1+11) 2.31.1
Last MCU build config: BUS_PINS_i2c1=PB6,PB7 RECEIVE_WINDOW=192 RESERVE_PINS_serial=PA3,PA2 BUS_PINS_i2c2=PB10,PB11 BUS_PINS_i2c1a=PB8,PB9 INITIAL_PINS=!PC13 STEPPER_BOTH_EDGE=1 SERIAL_BAUD=250000 ADC_MAX=4095 BUS_PINS_spi3=PB4,PB5,PB3 BUS_PINS_spi2=PB14,PB15,PB13 BUS_PINS_spi1=PA6,PA7,PA5 PWM_MAX=255 MCU=stm32f103xe STATS_SUMSQ_BASE=256 BUS_PINS_spi1a=PB4,PB5,PB3 CLOCK_FREQ=72000000
Build file /home/pi/klipper/klippy/../out/klipper.elf(252132): Fri Nov 12 00:44:26 2021
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 316, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/pi/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
error: mcu 'mcu': Serial connection closed

jordiblasco, 2021-11-23:
Hi,

I'm facing the same issue with Raspberry Pi Zero 2.

- Printer: Ender 3 v2
- Board: 4.2.2
- klipper: v0.10.0-164
- Moonracker: v0.7.1-167
- mainsail: v2.0.1
- CR-Touch installed

```
Build file /home/pi/klipper/klippy/../.config(2050): Tue Nov 23 20:03:01 2021
========= Last MCU build config =========
# CONFIG_LOW_LEVEL_OPTIONS is not set
# CONFIG_MACH_AVR is not set
# CONFIG_MACH_ATSAM is not set
# CONFIG_MACH_ATSAMD is not set
# CONFIG_MACH_LPC176X is not set
CONFIG_MACH_STM32=y
# CONFIG_MACH_RP2040 is not set
# CONFIG_MACH_PRU is not set
# CONFIG_MACH_LINUX is not set
# CONFIG_MACH_SIMU is not set
CONFIG_BOARD_DIRECTORY="stm32"
CONFIG_MCU="stm32f103xe"
CONFIG_CLOCK_FREQ=72000000
CONFIG_SERIAL=y
CONFIG_FLASH_START=0x8007000
CONFIG_FLASH_SIZE=0x10000
CONFIG_RAM_START=0x20000000
CONFIG_RAM_SIZE=0x5000
CONFIG_STACK_SIZE=512
CONFIG_STM32_SELECT=y
CONFIG_MACH_STM32F103=y
# CONFIG_MACH_STM32F207 is not set
# CONFIG_MACH_STM32F401 is not set
# CONFIG_MACH_STM32F405 is not set
# CONFIG_MACH_STM32F407 is not set
# CONFIG_MACH_STM32F429 is not set
# CONFIG_MACH_STM32F446 is not set
# CONFIG_MACH_STM32F031 is not set
# CONFIG_MACH_STM32F042 is not set
# CONFIG_MACH_STM32F070 is not set
# CONFIG_MACH_STM32F072 is not set
# CONFIG_MACH_STM32H743 is not set
# CONFIG_MACH_STM32H750 is not set
CONFIG_MACH_STM32F1=y
CONFIG_HAVE_STM32_USBFS=y
CONFIG_HAVE_STM32_CANBUS=y
# CONFIG_STM32_FLASH_START_2000 is not set
# CONFIG_STM32_FLASH_START_5000 is not set
CONFIG_STM32_FLASH_START_7000=y
# CONFIG_STM32_FLASH_START_8000 is not set
# CONFIG_STM32_FLASH_START_8800 is not set
# CONFIG_STM32_FLASH_START_10000 is not set
# CONFIG_STM32_FLASH_START_800 is not set
# CONFIG_STM32_FLASH_START_4000 is not set
# CONFIG_STM32_FLASH_START_0000 is not set
CONFIG_CLOCK_REF_FREQ=8000000
CONFIG_STM32F0_TRIM=16
# CONFIG_STM32_USB_PA11_PA12 is not set
CONFIG_STM32_SERIAL_USART1=y
# CONFIG_STM32_CANBUS_PA11_PA12 is not set
CONFIG_CANBUS_FREQUENCY=500000
CONFIG_SERIAL_BAUD=250000
CONFIG_USB_VENDOR_ID=0x1d50
CONFIG_USB_DEVICE_ID=0x614e
CONFIG_USB_SERIAL_NUMBER_CHIPID=y
CONFIG_USB_SERIAL_NUMBER="12345"
CONFIG_HAVE_GPIO=y
CONFIG_HAVE_GPIO_ADC=y
CONFIG_HAVE_GPIO_SPI=y
CONFIG_HAVE_GPIO_I2C=y
CONFIG_HAVE_GPIO_HARD_PWM=y
CONFIG_HAVE_GPIO_BITBANGING=y
CONFIG_HAVE_STRICT_TIMING=y
CONFIG_HAVE_CHIPID=y
CONFIG_HAVE_STEPPER_BOTH_EDGE=y
CONFIG_INLINE_STEPPER_HACK=y
=======================
Build file /home/pi/klipper/klippy/../out/klipper.dict(6838): Tue Nov 23 20:03:31 2021
Last MCU build version: v0.10.0-164-ga1af78cb
Last MCU build tools: gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-11+rpi1+11) 2.31.1
Last MCU build config: BUS_PINS_i2c1=PB6,PB7 RECEIVE_WINDOW=192 RESERVE_PINS_serial=PA10,PA9 BUS_PINS_i2c2=PB10,PB11 BUS_PINS_i2c1a=PB8,PB9 STEPPER_BOTH_EDGE=1 SERIAL_BAUD=250000 ADC_MAX=4095 BUS_PINS_spi3=PB4,PB5,PB3 BUS_PINS_spi2=PB14,PB15,PB13 BUS_PINS_spi1=PA6,PA7,PA5 PWM_MAX=255 MCU=stm32f103xe STATS_SUMSQ_BASE=256 BUS_PINS_spi1a=PB4,PB5,PB3 CLOCK_FREQ=72000000
Build file /home/pi/klipper/klippy/../out/klipper.elf(252028): Tue Nov 23 20:03:41 2021
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 316, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/pi/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
error: mcu 'mcu': Serial connection closed
```

csharp2a, 2021-11-25:
Hi Jordi,

 

Sorry to hear about your issue.  I never was able to find a solution.  Unless I missed something, I have not had any response from the devs.

Yesterday I went back to Marlin and there I will stay until Klipper is fixed.  I really like Klipper but it just does not work anymore.

 

Craig

Arksine, 2021-11-25:
For those of you with a Pi Zero 2, if you have a spare USB hub try putting it between the Pi and the Board.  Some boards had an issue connecting via OTG with the original Pi Zero.