Klipper3d/klipper #2517: Feature: Extruder option: same_extruder

gcurtis79, 2020-02-16:
I have a setup on my printer that allows 2 filaments with a single motor and single hot-end. A servo pushes one or the other against the drive gear, and runs in the opposite direction for each filament.

[Single nozzle single drive](https://www.thingiverse.com/thing:1643906) 

Klipper can't support this currently as far as I know, because it won't allow me to use the same motor pins for both extruders.

`
[extruder]
dir_pin: !ar28
...

[extruder1]
dir_pin: ar28
`

That's the only different option on the two extruders I need. I have the gcode_macros all set and ready for tool changes, but need to have the ability to reverse the extruder direction when using the second tool.

gcurtis79, 2020-02-16:
@klipper-gitissuebot There's no log file to add since this is a feature request.

Yay, I'm talking to a bot again. 👍

jalanjarosz, 2020-02-16:
Search the cases in the repo

Someone just did this successfully a few weeks ago using a “virtual_pin”

James

jalanjarosz, 2020-02-16:
Duo of: https://github.com/KevinOConnor/klipper/issues/30#issuecomment-581197304

jalanjarosz, 2020-02-16:
That was your issue in #30. 
Did the virtual pin switch the direction of the stepper?

What about releasing the servo so you can remove the filament?

Put a klippy.log into this ticket or it will auto-close.

gcurtis79, 2020-02-16:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4209825/klippy.log)

There's a log. I fired up the printer to get a clean one. The most recent *actual* log is over 12MB.

The printer works awesome with Klipper, I just want to get the multi-color thing working. It worked well enough in [The FW who's name I dare not speak], but Klipper performs SO much better over-all. I've moved all of my printers to Klipper.

Side note: I just saw this issue that looks similar to what I'm seeking, but it's closed:

[Shared Heater different Moving Direction #1631](https://github.com/KevinOConnor/klipper/issues/1631)

gcurtis79, 2020-02-16:
I'm playing around, and I think I realized that I'm way out of my depth in this python code.

I tried to sort-of duplicate the shared heater thing with the stepper, but there's some underlying functions for the shared heater that I can't figure out how to effectively duplicate.

I tried to make it ignore pins declared multiple times, but when I switch to T1, the pins for [extruder1] are not activated, and the stepper rotates the same directions as when T0 is active. It seems like it sets the pin values on init, and won't change them during run-time.

In a perfect world, I think something like this would be nice:
```
[extruder]
step_pin: ar26
dir_pin: !ar28
enable_pin: !ar24
....

[extruder1]
shared_stepper: extruder
# Enable (true) stepper_reverse if filaments are on opposite sides of the drive gear
# Disable (false) stepper_reverse if filaments are on same side of the drive gear
stepper_reverse: true
shared_heater: extruder
# Everything else is the same
```

ld3300, 2020-02-16:
Now, this is a stupid possible workaround that may be a temporary bandaid.  I can test it when my current print is done, but you might be able to use the new rename_existing function of gcode_macro to intercept G1 moves and if extruder1 is active multiply any e moves by -1.

gcurtis79, 2020-02-16:
"If it works, it wasn't a stupid idea..." - Some Body.

I wasn't aware you could do math in the macros like that, but if it works, I'll take it.

ld3300, 2020-02-16:
Sorry, @gcurtis79 i deleted my suggestion because I think it might not work with things like pressure advance.

KevinOConnor, 2020-02-18:
This looks like a dup of #2171 .

-Kevin

KevinOConnor, 2020-02-24:
Lets close this in favor of the existing #2171.  

-Kevin

closed 2020-02-24 by KevinOConnor