Klipper3d/klipper #2519: Prints fail randomly with "Lost communication with MCU"

mejobloggs, 2020-02-16:
My prints seem to randomly fail at different points. I come back to find it frozen and the hotend will have melted a little hole in the print

The same gcode (without reslicing) can fail early or later, or sometimes if I'm lucky it will complete the print with no errors (very rare, usually errors).

I've tried a new sdcard, new usb cable, checked my wiring...

Kevin, I've read the FAQ and other forum posts and from what I gather it sounds like a hardware issue? And not particularly anything you can solve?

Just looking for some final ideas before I go back to Marlin. Thanks very much for your work on Klipper it's an amazing project but maybe i've just got some rubbish wire somewhere causing issues

[klippylog.zip](https://github.com/KevinOConnor/klipper/files/4209407/klippylog.zip)

ld3300, 2020-02-16:
I had a similar problem that turned out to be a failing power supply on my raspberry pi.  It would have voltage drops randomly that would cause instability in the klipper connection to the cpu.

KevinOConnor, 2020-02-18:
Alas, the best information I have on this is in the FAQ: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors

It is normal for USB connections to last months/years without a disconnect, so it should be possible to fix this and not have the issue occur ever again.  The issue wont be related to the sd card - all previous reports of this were eventually traced to voltage or wiring problems.

-Kevin

mejobloggs, 2020-02-19:
Yesterday I printed 4 hours straight no MCU errors.

Today it errors as soon as I click print and it doesnt even start heating nozzle/bed.

I'm very confused

mejobloggs, 2020-02-28:
Fixed by replacing Raspberry Pi with Orange Pi

closed 2020-02-28 by mejobloggs