Klipper3d/klipper #2521: [SUPPORT REQUEST] Klipper host running on separate computer as Beaglebone PRU

sapell, 2020-02-16:
Preface: My printer setup has been evolving over time quite much and currently it is run by a BeagleBone Black with Replicape and a SKR v1.3. The SKR is the main MCU, but I still need to utilize some Replicape hardware resources. The BBB is rather underpowered to be running Octoprint, so I put Octoprint on a separate RasPi 3 and bridged the Klipper serial communication with socat to be passed through the BBB USB-Serial Gizmo to the RasPi 3. This works ok, but there is the issue that the Klipper remains on the BBB and it's tricky to get the virtual SD and Klipper config accessible through Octoprint.

Therefore I'd like to have Klipper host running on the RasPi 3 with Octoprint while the Klipper PRU MCU code would remain running on the BeagleBone Black. There are two PRUs running on the BBB and there are two pseudo serial devices used for communication. Would it be possible to use only one serial port for host <> PRU communication (/tmp/klipper_host_mcu)? If both are needed, then it could be possible to use socat to pass the serial ports over TCP, that will just be a bit more involved.

KevinOConnor, 2020-02-18:
It should be possible to route the /tmp/klipper_host_mcu messages over a socat connection.  It would also be necessary to route the messages to /dev/rpmsg_pru30 and I do not know if that is possible to do with socat.

Separately, there's been a lot of interest in building an http interface directly into Klipper - that might be a possible long term solution for this issue.

-Kevin

vladbabii, 2020-02-22:
You can run this with socat and ser2net. I have docker images running klipper and octoprint on separate containers. You can use a pi zero connected to printer serial and via network run klipper remotely and then via network run octoprint on a different host. Let me know if you need some config examples.

sapell, 2020-02-22:
That's a good idea, that's more-or-less what I had in mind with 'pass serial over TCP'. However, I have come to a conclusion to keep the Klippy running on the BBB with the Replicape. In this configuration Klipper is totally separate of Octoprint with the benefit of having no issues to reboot Octoprint/Webcam RasPi even while printing (from virtual SD). To have Klipper files accessible for Octoprint, I just use sshfs to mount Klipper virtual SD and configuration directories on OctoPi. I still need to set up internet connection sharing on the OctoPi to enable easier updating of Klipper. For clarification, I use the BBB host/OTG USB-connection to communicate between the BBB and RasPi, it has both serial and Ethernet gadgets on the same physical interface. I'm closing this issue as it is no longer relevant (for me) and potential solutions have been suggested.

- Simo

closed 2020-02-22 by sapell

rizwansarwar, 2020-10-13:
> You can run this with socat and ser2net. I have docker images running klipper and octoprint on separate containers. You can use a pi zero connected to printer serial and via network run klipper [...]

Do you mind sharing your setup? I am trying something like this but can't get ser2net/socat to work. Thanks

vladbabii, 2020-10-13:
Docker stack example here: https://github.com/klipperondocker/stack/blob/master/standard.yml

Ser2net params here: https://github.com/klipperondocker/stack/blob/master/standard.yml#L61

Socat params: https://github.com/klipperondocker/stack/blob/master/standard.yml#L40

Also have a homeassistant image using socat for connecting to remove zwave serial 

docker file: https://github.com/vladbabii/homeassistant-socat

socat check/run bash file: https://github.com/vladbabii/homeassistant-socat/blob/master/runwatch/100.socat-zwave.enabled.sh