Klipper3d/klipper #2522: Bypassing pin ar63 used multiple times in config

bl4ckm4mba97, 2020-02-16:
I have a Ramps 1.6 Plus from bigtreetech with the integrated SPI jumpers in the PCB. They use for XCS: D63 and for YCS: D40. The problem is that my LCD2004 (5 Buttons) is connected to AUX2 and is using the same Pins. In Marlin this multiple using of the same pin was no problem and worked just fine. Why is there a limitation in klipper for this and can i bypass it? I could somehow use the spi_software_pins but I dont know how to reroute miso,mosi,sclk.

jalanjarosz, 2020-02-16:
@ bl4ckm4mba97, I have a Re-ARM and a beta of a RAMPS-32/v2 on my bench.  I’ve setup a MKS Mini v2.1, which also uses the SPI pins.  I did not define SPI pins in the display section and the display works with all the features.

Try to not define SPI and see the 5button 2004 display works.  I’d be interested to know since I have one of these on my Anet A8.

-James

bl4ckm4mba97, 2020-02-16:
[klippylog.txt](https://github.com/KevinOConnor/klipper/files/4209756/klippylog.txt)
 My Log File

bl4ckm4mba97, 2020-02-16:
> @ bl4ckm4mba97, I have a Re-ARM and a beta of a RAMPS-32/v2 on my bench. I’ve setup a MKS Mini v2.1, which also uses the SPI pins. I did not define SPI pins in the display section and the display [...]

Hm how exactly do I "not define the SPI Pins" ?  
My Display section is this: 
[display]
lcd_type = hd44780
rs_pin = ar64
e_pin = ar44
d4_pin = ar63
d5_pin = ar40
d6_pin = ar42
d7_pin = ar65
up_pin = analog12
analog_range_up_pin = 9000, 13000
down_pin = analog12
analog_range_down_pin = 800, 1300
click_pin = analog12
analog_range_click_pin = 2000, 2500
back_pin = analog12
analog_range_back_pin = 4500, 5000

And my TMC´s 
[tmc2130 stepper_x]
cs_pin = ar63
microsteps = 16
interpolate = True
run_current = 0.800
hold_current = 0.500
stealthchop_threshold = 100

[tmc2130 stepper_y]
cs_pin = ar40
microsteps = 16
interpolate = True
run_current = 0.800
hold_current = 0.500
driver_sgt = -10
stealthchop_threshold = 100
diag1_pin = ^!ar14

jalanjarosz, 2020-02-16:
I see what you are doing, you trying to use EXP2, not EXP1 on the RRD adapter.   That is the problem for the Cs pins.  

Let's talk about the display.   EXP1 is your best bet for the "display", but not the navigation buttons.  You need an analog pin, which EXP1 does not have.

Best solution is to re-route the single wire off the ribbon cable and move it to an analog pin on EXP2.   
More info can be found here on how to do that.: https://forum.repetier.com/discussion/4165/configuring-lcd2004-5-button-from-anet-melzi-zonestar-etc-with-ramps-1-4

Basically, you are cutting the wire (looks to be the 1st wire, off the black connector and extending it with to a single Dupont connector and plugging it into a free analog pin on AUX1/2.   The following Analog pins should be free, unless you use them for something else:
AUX1-analog3
AUX1-analog4
AUX2-analog5
AUX2-analog10

jalanjarosz, 2020-02-16:
Your display .cfg would look like:

`[display]
lcd_type: hd44780
rs_pin: d16
e_pin: d17
d4_pin: d23
d5_pin: d25
d6_pin: d27
d7_pin: d29
up_pin = analogX
analog_range_up_pin = 9000, 13000
down_pin = analogX
analog_range_down_pin = 800, 1300
click_pin = analogX
analog_range_click_pin = 2000, 2500
back_pin = analogX
analog_range_back_pin = 4500, 5000`

closed 2020-02-16 by bl4ckm4mba97