Klipper3d/klipper #2523: Last Endstop Status in macros not working as documented

vladbabii, 2020-02-16:
Document:
https://github.com/KevinOConnor/klipper/blob/master/docs/Command_Templates.md

```
printer.query_endstops.last_query["<endstop>"]: Returns True if the given endstop was reported as "triggered" during the last QUERY_ENDSTOP command. Note, due to the order of template expansion (see above), the QUERY_STATUS command must be run prior to the macro containing this reference.
```

gcode:
```
[gcode_macro CHECK_ENDSTOPS_PULLBACK]
gcode:
  {% if printer.query_endstops.last_query["x"] %}
    RESPOND PREFIX="info" MSG="Moving away from endstop X"
  {% endif %}
  {% if printer.query_endstops.last_query["y"] %}
    RESPOND PREFIX="info" MSG="Moving away from endstop Y"
  {% endif %}
```

ran gcode with:
```
QUERY_ENDSTOPS
CHECK_ENDSTOPS_PULLBACK
```

error:
```
Error evaluating 'gcode_macro CHECK_ENDSTOPS_PULLBACK:gcode': UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'query_endstops'
```

Also
* gcode command is QUERY_ENDSTOPS
* there is no QUERY_STATUS command
* tried both query_endstops and query_endstop in gcode macro

Thank you

Arksine, 2020-02-16:
Template evaluation of query_endstops works for me.   Here are the results I get from your macro:
```
Send: QUERY_ENDSTOPS
Recv: x:TRIGGERED y:open z:open
Recv: ok
Send: M105
Recv: ok B:22.8 /0.0 C:19.8 /0.0 T0:23.5 /0.0
Send: CHECK_ENDSTOPS_PULLBACK
Recv: info Moving away from endstop X
```

If you could reproduce the issue and provide a complete log we can further assist.

vladbabii, 2020-02-16:
@Arksine i'm on 
```
Firmware version: v0.8.0-310-g086c131b
```

What version are you on ?

[printer_cfg.zip](https://github.com/KevinOConnor/klipper/files/4210909/printer_cfg.zip)

ld3300, 2020-02-16:
hi @vladbabii can you please attach your log file. That was just your config file. [https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md](https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md)

vladbabii, 2020-02-16:
Updating to latest klipper version and also updated the board firmware and now it works. This was initially installed about 2 weeks ago.

Sorry for wasting everyone's time. Is there a better way to keep up to date with changes other than reading commit logs? I don't update very often on "production" printers due to the possible extended downtime from some change.

jakep82, 2020-02-17:
Changes that will break or alter configs are listed [here.](https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md)

vladbabii, 2020-02-17:
But not new features that are non breaking....

vladbabii, 2020-02-17:
Thank you all for your help!

closed 2020-02-17 by vladbabii