Klipper3d/klipper #2527: Homing -> Changing X - Y

TheRunnerPi, 2020-02-17:
Hi all,

if my Heatbed is pulled out and i do a G28 Homing, the X Axis drives into the Kabelchain snd cant home.

Is it possible to switch the G28 code, that it homes Y Axis first and then the X Axis?

[klippy.log](https://github.com/KevinOConnor/klipper/files/4215870/klippy.log)

Regards, 
Daniel

theopensourcerer, 2020-02-17:
I think this is what you are after:

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L335

TheRunnerPi, 2020-02-18:
Thanks @theopensourcerer 

but still not fixed.

After typing this code into the printer.cfg 
```
[homing_override]
axes: yxz
```
i get the Error: 

**homing_override and safe_z_homing cannot be used simultaneously**

After typing this code into the printer.cfg 
```
[safe_z_home]
home_xy_position: 85, 73
speed: 150
z_hop: 10.0
z_hop_speed: 20.0
move_to_previous: False
axes: yxz
```
i get the Error: 

**Option 'axes' is not valid in section 'safe_z_home'**

What can i else try?

Arksine, 2020-02-18:
If you want to use homing_override you can't use safe_z_home.  They essentially are capable of the same thing, the difference being that safe_z_home is specifically targeted toward homing Z and homing override can be used for all axes.

Given your configuration above, it would need to look something like this:
```
[homing_override]
axes: xyz
set_position_z: 0
gcode:
  G1 Z10 F600
  G28 Y
  G28 X
  G1 X85 Y73 F9000
  G28 Z
```

The above homing_override would move z up 10mm, Home Y, Home Z, Move to X85 Y73, then home Z.  The caveat here is that you can no longer home each axis individually, if you issue a `G28 X` gcode then it will still home all 3 axes.

TheRunnerPi, 2020-02-18:
Thank you much @Arksine 
seems like it works :) 👍

closed 2020-02-18 by TheRunnerPi

leorockford, 2021-03-28:
Good Morning.
It does not work for me.
I need to do the homing on the Y axis first and then on the Z axis.
With this sequence, keep making me home first on the X axis, then on the Y axis and finally on the Z axis.
What am I doing wrong?

# [safe_z_home]
#home_xy_position: 0.360

[homing_override]
axes: yxz
set_position_x: 0
set_position_z: 0
gcode:
   G1 Z10 F600
   G1 X10 F600
   G28 AND
   G28 X
# G1 X85 Y73 F9000
   G28 Z

theopensourcerer, 2021-03-28:
Is your gcode indented? i.e.

gcode:
    G1 Z10 F6000
    G28 Y
    G28 X
    G28 Z

Also I think your axes statement should be xyz. The docs do not say the order makes any difference, only that each axis specified will have it's normal homing sequence overridden by the gcode:

https://www.klipper3d.org/Config_Reference.html#homing_override

leorockford, 2021-03-28:
Update as you told me:

[homing_override]
axes: xyz
gcode:
G1 Z10 F600
G28 AND
G28 X
G28 Z

He answered this:

Send: G28
Recv: // File contains parsing errors: /home/pi/printer.cfg
Recv: // [line 27]: 'G1 Z10 F600 \ n'
Recv: // [line 28]: 'G28 Y \ n'
Recv: // [line 29]: 'G28 X \ n'
Recv: // [line 30]: 'G28 Z \ n'
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: !! File contains parsing errors: /home/pi/printer.cfg