Klipper3d/klipper #2528: Homing Endstop still triggered after retract

aliaksei135, 2020-02-17:
In my start G-Code for all my prints, I have a `G28` command. When I execute this in isolation from the Octoprint terminal, this works as expected, however when I start a print and `G28` is executed from the start G-Code, this is thrown: `Endstop z still triggered after retract`. I am using a capacitive probe, which is wired and configured correctly (`M119` returns the expected results). I am sure the probe is okay, because an LED on top indicates it is not triggered when this occurs.

I have tried increasing `homing_retract_dist` very high (upto 20mm), with no effect.

Printer is a Modded Anet A8.

Log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4215892/klippy.log)

KevinOConnor, 2020-02-24:
FWIW, the log indicates that the endstop was reporting as still being in a triggered state after the retraction completed.  As far as I know, the code works as intended.  You can disable the second homing attempt if you wish (see config/example.cfg for the details).

-Kevin

aliaksei135, 2020-03-03:
Turns out the bed heater was causing electrical interference of some kind. Fixed by adding `TURN_OFF_HEATERS` before `G28` then settings temps back using `M109` and `M140` after homing in the start G-Code

closed 2020-03-03 by aliaksei135