Klipper3d/klipper #2530: Homing failing

kurtisane, 2020-02-18:
Hi, 
I just flashed Klipper on my SKR PRO . 

When trying to use the G28 command I'm facing the following issue.
G28 or G28 X in X direction works perfectly.
G28 or G28 Y leads to "!! Internal error on command:"G28""
G28 Z  leads to "!! Failed to home z: Timeout during endstop homing"

I'm not sure what is going on here. 
The log file is created after a regular G28. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4216739/klippy.log)

It says deviding by 0. Never a good thing.

Regards !

jakep82, 2020-02-18:
If your endstop for Y is located at the max, then your position_endstop should not be 0.  Try using the below settings if you're trying to home to max.

```[stepper_y]
step_pin = PE11
dir_pin = !PE8
enable_pin = !PD7
step_distance = .001566
endstop_pin = PE10
position_endstop = 267
position_max = 267
homing_speed = 50
homing_retract_dist = 5.0

For the Z, you probably need to slow it down, and you definitely need to drop your max Z speed.  With your current settings, you're max Z velocity would require almost 1 million steps per second.  I would personally start with 16 microsteps on each axis, verify everything is working, and make changes if you feel the need after that.  Running really high microstep settings often leads to issues even if the MCU is technically capable of producing the required number of stepper pulses.

KevinOConnor, 2020-03-05:
I'm going to close this ticket as it looks like the original question was answered (don't specify an incorrect homing_positive_dir).

-Kevin

closed 2020-03-05 by KevinOConnor