Klipper3d/klipper #2531: Shutdown klipper host via printer controls

chromoxdor, 2020-02-18:
I use klipper on my Raspberry Pi 3A+ with an Ender 3Pro.
First of all it works very well! (apart from octoprint to klipper communication) Thank you for your work!!!!!!
I was wondering if it is possible to send a shutdown command to the raspberry via the controller on the Ender 3. Or commands apart from gcode in general.
Couldn´t find any infos about it.

timmit99, 2020-02-18:
It would be possible to make a menu item on the display that just sends the `M112` "Shutdown" gcode command.
Another option is to have a button connected to a spare pin on the neder control board that is configured as a gcode-button in Klipper and have THAT send the `M112` command

chromoxdor, 2020-02-18:
For shutting down the pi? How is that supposed to work?

JohnEdwa, 2020-02-18:
Octoprint can do it for you with the ["Action Commands" plugin.](https://plugins.octoprint.org/plugins/actioncommands/)    
I use it to start the print from the LCD menu by sending a custom `action:startprint` that the plugin then changes to a system command of `curl -X POST -H "Content-Type: application/json" -H "X-Api-Key:>APIKEY>" -d '{"command":"start"}' http://127.0.0.1/api/job`

You could use that to create a `action:shutdown` command that runs `shutdown now` on the pi - though you might need to figure out how to give it the correct permission to run that command without using sudo.

chromoxdor, 2020-02-19:
Thanks a lot!

Added this to menu.conf:
`[menu __octoprint __poweroff]
type: command
enable: !toolhead.is_printing
name: Turn off Octopi
action: respond action:poweroff`

In Action Commands i did a:`sudo shutdown now`
An last but not least did: `sudo visudo`
and added this line: `user_name ALL=(ALL) NOPASSWD: /sbin/poweroff, /sbin/reboot, /sbin/shutdown`  [(source:)](https://askubuntu.com/questions/168879/shutdown-from-terminal-without-entering-password)

BTW:

I added this to the menu.cfg too since i start my jobs from the ender control panel:
`[menu __sdcard __start]
type: command
enable: !toolhead.is_printing
name: Start printing
gcode: M24
**action: exit**` 

It brings you directly to the main screen...much more convenient than doing it manually!

chromoxdor, 2020-02-19:
I have to reopen this.
Another question regarding the menue.cfg:
Is it possible to send multiple actions.
eg:
`[menu __octoprint __poweroff]`
`type: command`
`enable: !toolhead.is_printing`
`name: Turn off Octopi`
`gcode: M117 Pi is shutting down...`
**`action: respond action:poweroff`**
**`action2: exit`**
Would give a nice feedback. (returns to the mainscreen and shows the message )
Just the first action would also do the trick if i could overlay the M117 right after
Its always good to have some sort of feedback.

markwinger, 2020-02-24:
I am trying to implement the OctoPi Shutdown command described above.  I have the menus setup but I don't understand how to add the action with the action command plugin.   Can someone point me to a description of how this is done?

markwinger, 2020-02-25:
I was confused by "Action Command Prompt" vs "Action Commands".  I installed Action Commands and now all makes sense.   Just ignore my silly question. ;)

ld3300, 2020-03-01:
@chromoxdor you could use the G-code command RESPOND TYPE=command MSG="<message>"  which also works with action command.

galosre, 2020-03-04:
> I have to reopen this.
> Another question regarding the menu.cfg:
> Is it possible to send multiple actions.
> eg:
> `[menu __octoprint __poweroff]`
> `type: command`
> `enable: !toolhead.is_printing`
> `name: Turn off Octopi`
> `gcode: M117 Pi is shutting down...`
> **`action: respond action:poweroff`**
> **`action2: exit`**
> Would give a nice feedback. (returns to the mainscreen and shows the message )
> Just the first action would also do the trick if i could overlay the M117 right after
> Its always good to have some sort of feedback.

 I am unable to configure action Command plugin, in order to achieve this! what should I input under Action; Type; and commands?

ld3300, 2020-03-05:
That's more a documentation question about the plugin.  There is some documentation there: [Action Commands](https://plugins.octoprint.org/plugins/actioncommands/)

galosre, 2020-03-05:
> 
> 
> That's more a documentation question about the plugin. There is some documentation there: [Action Commands](https://plugins.octoprint.org/plugins/actioncommands/)

I would have  hoped a little more help from here since the documentation is rather succinct.

ld3300, 2020-03-05:
so the action field is any word you want.  The gcode command is: `RESPOND TYPE=command MSG=""` 

When you hover over "type" it gives a pretty good description"

@chromoxdor  gives some really good directions further above: [here](https://github.com/KevinOConnor/klipper/issues/2531#issuecomment-588229093)

galosre, 2020-03-06:
> Is poweroff what you are looking for?
Not only poweroff but I truly appreciate your help, I am trying to get this running [https://www.reddit.com/r/klippers/comments/csjzid/sample_config_for_ender_3_bltouch_v30_skr_v13/](url)

ld3300, 2020-03-06:
Ahh, after the power off most of those will be a breeze

galosre, 2020-03-06:
> 
> 
> Ahh, after the power off most of those will be a breeze

Good for you, so for example how to to configure this to make if functional with action command?
`[gcode_macro M80]
gcode:
    RESPOND TYPE=command MSG=action:poweron
    G4 P1500
    FIRMWARE_RESTART

[gcode_macro M81]
gcode:
    RESPOND TYPE=command MSG=action:poweroff
`

ld3300, 2020-03-06:
> so the action field is any word you want. The gcode command is: `RESPOND TYPE=command MSG=<phrase>` the menu command is `respond action:<phrase>`
> 
> When you hover over "type" it gives a pretty good description"
> 
> @chromoxdor gives some really good directions further above: [here](https://github.com/KevinOConnor/klipper/issues/2531#issuecomment-588229093)
> The first line is what he put in action commands. The other 2 lines need to be done in a terminal just once to set permissions.

galosre, 2020-03-06:
No matter what I put in the field Command I get Unknown command:"RESPOND"
for [menu __command_power_off]
type: command
name: Power off
gcode: RESPOND TYPE=command MSG=action:poweroff

ld3300, 2020-03-06:
[RESPOND](https://github.com/KevinOConnor/klipper/blob/a074af7c871633f2fad5dd207f9945a020a0cde5/config/example-extras.cfg#L514)

galosre, 2020-03-07:
Thanks! now I getRecv: // action:poweroff in terminal, but still not being pickup by action commands, in relation  with my example above what should be the exact Command in the plugin?

ld3300, 2020-03-07:
> No matter what I put in the field Command I get Unknown command:"RESPOND"
> for [menu __command_power_off]
> type: command
> name: Power off
> gcode: RESPOND TYPE=command MSG=action:poweroff

remove the "action:" from the last line and try it.  so it just says 
`gcode: RESPOND TYPE=command MSG=poweroff`

ld3300, 2020-03-07:
Sorry, I was wrong, the action: does need to be there. 
Try looking at these steps and see if they help: [Actions](https://github.com/KevinOConnor/klipper/issues/643#issuecomment-596059810)

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like the original question was answered.

-Kevin

closed 2020-03-21 by KevinOConnor

ribbles, 2020-07-13:
In the end, what did you do @ld3300? Did you need to enable the RESPOND command too?

ld3300, 2020-07-13:
This is my current menu command.  I am a ways out of date from being ill, so the menu implementation might be different.

```
[menu __octoprint __shutdown]
type: command
enable: !toolhead.is_printing
name: Turn off Octopi
action: respond action:shutdown
```

then in octopi in the Action Commands plugin I have a command that is Action Shutdown, Type - System, Command - sudo shutdown now

Cirromulus, 2020-09-09:
Sorry for reopening, but the formerly working solution of ld3300

> ```
> [menu __octoprint __shutdown]
> type: command
> enable: !toolhead.is_printing
> name: Turn off Octopi
> action: respond action:shutdown
> ```

Does not work with the newer version of Klipper (currently at 9197975625daa421334807f6ce6c2b077f615023):

```
ERROR:root:Config error
Traceback (most recent call last):
  File "klipper/klippy/klippy.py", line 146, in _connect
    self._read_config()
  File "klipper/klippy/klippy.py", line 143, in _read_config
    pconfig.check_unused_options(config)
  File "/home/dwd/klipper/klippy/configfile.py", line 248, in check_unused_options
    option, section))
Error: Option 'action' is not valid in section 'menu __octoprint __poweroff'
```

tempestaz, 2021-08-06:
Using Klipper v0.9.1-525-g8e74e6aa

Klippy.log reports an issue with toolhead.is_printing. From the current docs it appears that status is no longer available.

Revised working menu item looks like below:

```
[menu __main __octoprint __poweroff]
type: command
name: Turn off Octopi
gcode:
    {action_respond_info('action:poweroff')}

```

sotima, 2021-11-22:
You now can use 

`enable: {not printer.idle_timeout.state == "Printing"}` 

in order to show the item only if the printer is not printing...
So in whole:

```
[menu __main __octoprint __poweroff]
type: command
enable: {not printer.idle_timeout.state == "Printing"}
name: Turn off Octopi
gcode:
    {action_respond_info('action:poweroff')}
```