Klipper3d/klipper #2536: How to escape @ sign in klipper configuration for BEDLEVELVISUALIZER

MoJo2600, 2020-02-18:
I have a short question on how to configure BEDLEVELVISUALIZER in klipper config.

Under `bed visualizer settings` i configured the following command which reads the latest bed mesh and shows it in the visualizer.

```
@BEDLEVELVISUALIZER
BED_MESH_OUTPUT
``` 
I'd like to have the visualizer updated everytime i do a `BED_MESH_CALIBRATE` so i created a gcode macro G29 with the following content:

```
[gcode_macro G29]
gcode:
        G28
        G1 Z10 F600
        BED_MESH_CALIBRATE
        @BEDLEVELVISUALIZER
        BED_MESH_OUTPUT
```

The issue is, that klipper always strips the ***@*** sign and tells me that the command `BEDLEVELVISUALIZER` is not found. I tried to add a backslash and double backslash, but it had no effect. How can I convince klipper to keep the ***@*** sign and use the command as it is stated in my macro?

Thanks, and thanks for this great piece of software!

Arksine, 2020-02-18:
Unfortunately that wont work.  Even if you escaped the @, Octoprint would never see it as `@BEDLEVELVISUALIZER` would be executed by Klipper internally.  What you are seeing in the terminal is Klipper's response at attempting (and failing) to execute the command.

What you need is an action command you can send back to octoprint that will execute @BEDLEVELVISUALIZER.

https://docs.octoprint.org/en/master/features/action_commands.html

I believe that there may be plugins that allow you to define your own action commands, or you could perhaps create a feature request on the plugin's repo for an action command.

ld3300, 2020-02-19:
The action command plugin for octoprint will respond to command responses

austinodega, 2020-02-19:
Please stop sending me emails, I tried to unsubscribe but yet emails are still coming Tuesday,

MoJo2600, 2020-02-19:
Thanks @Arksine and @ld3300 with your explanation it makes absolute sense! I thought that octoprint would parse the output of the log. I will try the action command and i am confident that this will work.

closed 2020-02-19 by MoJo2600