Klipper3d/klipper #2537: When homing overriding x and y g28 only homes those and skips z

vladbabii, 2020-02-19:
Config:
```
[gcode_macro CHECK_ENDSTOPS_PULLBACK]
gcode:
  {% if printer.query_endstops.last_query["x"] or printer.toolhead.position.x < 5 %}
    RESPOND PREFIX="info" MSG="Moving away from endstop X"
    SET_KINEMATIC_POSITION X=50 Y=50 Z=50
    G91
    G0 X5
    G90
  {% endif %}
  {% if printer.query_endstops.last_query["y"] %}
    RESPOND PREFIX="info" MSG="Moving away from endstop Y"
    SET_KINEMATIC_POSITION X=50 Y=50 Z=50
    G91
    G0 Y-5
    G90
  {% endif %}

[homing_override]
axes: xy
gcode:

  QUERY_ENDSTOPS
  CHECK_ENDSTOPS_PULLBACK

  RESPOND PREFIX="info" MSG="Homing > Y"
  G90
  G28 Y0
  G91
  G0 Y-5 F2000

  RESPOND PREFIX="info" MSG="Homing > X"
  G90
  G28 X0
  G91
  G0 X5 F2000

  G90
```

Running G28 homes X and Y then stops. I have to run both G28 and G28 Z to home all of them. I can probably fix this by overring G28 with a macro but this seems like a bug .

Thank you for your time

ld3300, 2020-02-19:
#2500

vladbabii, 2020-02-19:
Thank you @ld3300

vladbabii, 2020-02-19:
Temporarely fixed with

```

[gcode_macro G28]
rename_existing: G990028
gcode:
  {% set do_x = 0 %}
  {% set do_y = 0 %}
  {% set do_z = 0 %}

  {% if params.Y is defined %}
    ## for y home just y
    {% set do_y = 1 %} 
  {% endif %}

  {% if params.X is defined %}
    ## for x home y first then X
    {% set do_x = 1 %} 
    {% set do_y = 1 %} 
  {% endif %}

  {% if params.Z is defined %}
    ## for z home just z
    {% set do_z = 1 %} 
  {% endif %}

  {% if do_y == 1 %}
    RESPOND PREFIX="info" MSG="Home > Todo: Y"
  {% endif %}
  {% if do_x == 1 %}
    RESPOND PREFIX="info" MSG="Home > Todo: X"
  {% endif %}
  {% if do_z == 1 %}
    RESPOND PREFIX="info" MSG="Home > Todo: Z"
  {% endif %}

  {% if do_x == 0 and do_y == 0 and do_z == 0 %}
    {% set do_x = 1 %} 
    {% set do_y = 1 %} 
    {% set do_z = 1 %} 
    RESPOND PREFIX="info" MSG="Home > Doing all XYZ"
  {% endif %}

  {% if do_x == 1 or do_y == 1 %}
    QUERY_ENDSTOPS
    CHECK_ENDSTOPS_PULLBACK
  {% endif %}

  {% if do_y == 1 %}
    RESPOND PREFIX="info" MSG="Home > Y"
    G90
    G990028 Y0
    G91
    G0 Y-5 F2000
    G90
  {% endif %}

  {% if do_x == 1 %}
    RESPOND PREFIX="info" MSG="Homing > X"
    G90
    G990028 X0
    G91
    G0 X5 F2000
    G90
  {% endif %}

  {% if do_z == 1 %}
    RESPOND PREFIX="info" MSG="Homing > Z"
    G90
    G990028 Z0
    G91
    G0 Z20
    G90
  {% endif %}

  RESPOND PREFIX="info" MSG="Homing > Done"
```

Posted here - may be useful for someone else in the future.

@KevinOConnor Since i can override G28 to do anything i want, is homing_override needed anymore ?

closed 2020-02-26 by klipper-gitissuebot