Klipper3d/klipper #2539: Klipper doesn't like speed transition at all

mseriukov, 2020-02-19:
@megablue Here you wrote https://github.com/KevinOConnor/klipper/issues/1423
> I should say, you shouldn't use kisslicer on klipper at all. the reason being kisslicer will generate code that slow down the print speed during overhang and Klipper doesn't like speed transition at all, it will result in a micro pause during the transition and again when it transits from slow to fast.

Could you describe this `Klipper doesn't like speed transition at all` in more detail, please? 
Doesn't every slicer slice with speed transition between perimeter, loops, infill, etc?
Can you explain why micro pause will happen(and maybe related Klipper code)? 
Thanks!

Hywelmartin, 2020-02-19:
that fact is old... it should work nicely now days...

before klipper would go to a complete stop in between the different speed settings, as it treated them as different extrusions..

mseriukov, 2020-02-19:
@Hywelmartin Is it safe to assume that KISSlicer with preloadVE turned off is now should work fine with Klipper?
Is it also safe to assume that KISSlicer preload should work fine (as it will just generate several segments with different flowrate and as you say it should be ok) until it is not used simultaneously with Klipper PA?
So one should choose one or another.

Hywelmartin, 2020-02-19:
.... try it.. I have never used the slicer..

FHeilmann, 2020-02-19:
can confirm that KISSlicer with preloadVE turned off works with klipper.

KevinOConnor, 2020-03-05:
I'm going to close this ticket as it looks like the original question was answered (the latest Klipper code should no longer have the issues described in the original citation).  I don't know if preloadVE would work well with Klipper;  I suspect Klipper's pressure_advance would be better though.

-Kevin

closed 2020-03-05 by KevinOConnor