Klipper3d/klipper #2541: [FEATURE REQUEST] Smooth time everywhere where a thermistor is defined

hevilp, 2020-02-19:
Hi,

we do not have the option to add a smooth time in the fan chamber section. Would be nice to smooth the results there, too.

kind regards

hevilp, 2020-02-19:
No log needed here

KevinOConnor, 2020-02-24:
That's fine if an interested developer wishes to step forward and implement it.

-Kevin

hevilp, 2020-02-25:
For PID there is something like that already there:
#pid_deriv_time: 2.0
#   A time value (in seconds) over which the derivative in the pid
#   will be smoothed to reduce the impact of measurement noise. The
#   default is 2 seconds.

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like the conversation has concluded.  If there is a desire to add smoothing to temperature_fan then a developer will need to implement it.

-Kevin

closed 2020-03-21 by KevinOConnor