Klipper3d/klipper #2542: gcode_maco: Error while using printer.configfile.config

mkuf, 2020-02-19:
Hello, 

I'm getting an Error while trying to use `printer.configfile.config["<section>"]["<option>"]` within an gcode_macro. <br>

```
[gcode_macro TEST]
gcode:
  RESPOND MSG={printer.configfile.config['printer']['max_velocity']}
```
This simple Macro results in an undefined Error when called.
`Error evaluating 'gcode_macro TEST:gcode': UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'configfile'`

I'm currently on 7d76067ff9efaf75808b05f768dfef4295251ede, further Info can be found in the [klippy.log](https://github.com/KevinOConnor/klipper/files/4227267/klippy.log)

-Markus

KevinOConnor, 2020-02-20:
That `configfile` support was only added a few days ago, so you'll need to upgrade.

Separately, a truncated log doesn't help much - best to always attach the full log.

-Kevin

mkuf, 2020-02-20:
Oh, I did not realize that this was such a recent Change. 
Pulling the current master fixed this issue. Thanks!

-Markus

closed 2020-02-20 by mkuf