Klipper3d/klipper #2543: Bed_mesh_output gives weird numbers

Coffee0297, 2020-02-19:
from https://github.com/KevinOConnor/klipper/pull/2295#issuecomment-588406670
@Arksine 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4227357/klippy.log)
heres a log, i wuld assume that bed_mesh_output shuld give the real value from the printer.cfg
like here=https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/92#issuecomment-430741520
the octoklipper plugin will display some weird mesh that dont fit the mesh klipper follows

[image]

Coffee0297, 2020-02-19:
heres output from bed_mesh_output
```
Recv: Mesh X,Y: 34,34
Recv: Search Height: 7
Recv: Mesh Average: -0.02
Recv: Mesh Range: min=-0.0834 max=0.0774
Recv: Interpolation Algorithm: bicubic
Recv: Measured points:
Recv:   -0.019697  -0.037626  -0.067184  -0.085113  -0.087607  -0.087193  -0.085113  -0.075366  -0.058952  -0.047613  -0.036558  -0.020798  -0.011780  -0.012187  -0.015669  -0.020113  -0.035662  -0.060619  -0.075113  -0.073085  -0.065873  -0.059697  -0.049567  -0.033308  -0.021363  -0.004860  0.019939  0.034887  0.037652  0.038510  0.038220  0.034893  0.029195  0.025720
Recv:   -0.015561  -0.034159  -0.064830  -0.083477  -0.086362  -0.086428  -0.084607  -0.074722  -0.058038  -0.046644  -0.036357  -0.021920  -0.013808  -0.015045  -0.019691  -0.024780  -0.040022  -0.064347  -0.078610  -0.077754  -0.072535  -0.067530  -0.057273  -0.040482  -0.028011  -0.010195  0.016690  0.032878  0.035647  0.036218  0.035735  0.032453  0.026858  0.023448
Recv:   -0.008759  -0.028422  -0.060867  -0.080657  -0.084142  -0.084939  -0.083508  -0.073395  -0.056270  -0.044786  -0.035741  -0.023442  -0.016799  -0.019396  -0.025961  -0.032113  -0.046876  -0.070202  -0.084107  -0.085164  -0.083199  -0.080104  -0.069638  -0.051982  -0.038659  -0.018752  0.011450  0.029609  0.032381  0.032490  0.031695  0.028468  0.023015  0.019696
Recv:   -0.004697  -0.024861  -0.058135  -0.078447  -0.082126  -0.083120  -0.081780  -0.071521  -0.054150  -0.042613  -0.034221  -0.023052  -0.017197  -0.020595  -0.028308  -0.035113  -0.049690  -0.072582  -0.086363  -0.088511  -0.088382  -0.086363  -0.075777  -0.057635  -0.043863  -0.022987  0.008741  0.027803  0.030562  0.030433  0.029470  0.026208  0.020723  0.017387
Recv:   -0.004851  -0.024065  -0.055734  -0.074919  -0.077397  -0.076646  -0.074274  -0.063964  -0.046739  -0.035215  -0.026128  -0.013804  -0.007246  -0.010509  -0.018186  -0.025088  -0.040419  -0.064541  -0.079002  -0.080752  -0.079803  -0.077298  -0.066746  -0.048839  -0.035499  -0.016629  0.011838  0.028936  0.031587  0.031720  0.030883  0.027092  0.020716  0.016837
Recv:   -0.005968  -0.023460  -0.052227  -0.069391  -0.069827  -0.066135  -0.062027  -0.051658  -0.034719  -0.023224  -0.012861  0.001589  0.009464  0.006586  -0.000802  -0.007730  -0.024339  -0.050575  -0.066178  -0.067021  -0.064302  -0.060752  -0.050284  -0.032875  -0.020357  -0.005067  0.017592  0.031199  0.033673  0.034305  0.033723  0.029080  0.021265  0.016510
Recv:   -0.006363  -0.022746  -0.049647  -0.065530  -0.064774  -0.059388  -0.054280  -0.043845  -0.027021  -0.015530  -0.004536  0.010958  0.019470  0.016726  0.009390  0.002387  -0.014903  -0.042249  -0.058447  -0.058780  -0.055077  -0.050947  -0.040524  -0.023397  -0.011363  0.001757  0.020896  0.032387  0.034760  0.035708  0.035303  0.030251  0.021735  0.016553
Recv:   -0.003681  -0.019607  -0.045767  -0.061246  -0.060778  -0.055962  -0.051178  -0.040500  -0.023241  -0.011641  -0.001730  0.011902  0.019174  0.015506  0.006940  -0.000672  -0.017216  -0.043175  -0.058586  -0.059337  -0.056507  -0.052916  -0.042493  -0.025249  -0.013135  -0.000022  0.019094  0.030578  0.032965  0.033966  0.033704  0.029445  0.022223  0.017825
Recv:   0.000779  -0.014621  -0.039941  -0.055017  -0.055282  -0.051774  -0.047743  -0.036675  -0.018666  -0.006882  0.001098  0.011419  0.016507  0.011349  0.000850  -0.007693  -0.022912  -0.046440  -0.060484  -0.062031  -0.060832  -0.058237  -0.047769  -0.030217  -0.017833  -0.004254  0.015593  0.027528  0.029974  0.031014  0.030957  0.028062  0.023070  0.020022
Recv:   0.003220  -0.011826  -0.036576  -0.051363  -0.051999  -0.049172  -0.045530  -0.034221  -0.015764  -0.003863  0.003007  0.011433  0.015303  0.009544  -0.001660  -0.010530  -0.025234  -0.047826  -0.061363  -0.063366  -0.063036  -0.060947  -0.050314  -0.032367  -0.019697  -0.005854  0.014377  0.026553  0.029081  0.030220  0.030303  0.028016  0.024007  0.021553
Recv:   0.001532  -0.013039  -0.036982  -0.051187  -0.051117  -0.047241  -0.042962  -0.031524  -0.012991  -0.000993  0.006231  0.015292  0.019831  0.016269  0.008501  0.001384  -0.015414  -0.041912  -0.057792  -0.059599  -0.058382  -0.055580  -0.043964  -0.024530  -0.011052  0.002288  0.021492  0.033072  0.035946  0.037737  0.038004  0.034161  0.027510  0.023448
Recv:   -0.001841  -0.015735  -0.038521  -0.051863  -0.050567  -0.044842  -0.039450  -0.027898  -0.009407  0.002693  0.010811  0.021454  0.027433  0.027609  0.025618  0.021427  0.001016  -0.032199  -0.052091  -0.053457  -0.050550  -0.046435  -0.033218  -0.011399  0.003371  0.015882  0.033401  0.043998  0.047416  0.050228  0.050742  0.044163  0.032888  0.026010
Recv:   -0.003863  -0.017440  -0.039684  -0.052613  -0.050678  -0.043984  -0.038030  -0.026592  -0.008394  0.003637  0.012516  0.024507  0.031553  0.033710  0.034701  0.031970  0.009368  -0.027845  -0.050113  -0.051159  -0.047151  -0.042197  -0.028040  -0.004826  0.010720  0.022911  0.039714  0.049887  0.053560  0.056825  0.057387  0.049158  0.035115  0.026553
Recv:   -0.003625  -0.017763  -0.040955  -0.054564  -0.053368  -0.047792  -0.042663  -0.032509  -0.016205  -0.005091  0.005310  0.019995  0.028399  0.028586  0.025976  0.021035  -0.001764  -0.038793  -0.060857  -0.061553  -0.056980  -0.051688  -0.037578  -0.014479  0.001171  0.014668  0.033605  0.044986  0.048172  0.050343  0.050223  0.042002  0.028115  0.019661
[... cut ...] 4896  -0.045864  -0.033089  -0.022230  -0.007130  0.001742  0.002258  0.001081  0.000927  0.006316  0.015254  0.020680
Recv:   -0.010113  -0.025153  -0.049897  -0.064697  -0.064953  -0.061820  -0.059280  -0.056897  -0.053274  -0.050947  -0.049391  -0.047465  -0.046780  -0.049067  -0.053539  -0.058030  -0.071064  -0.091672  -0.103447  -0.100789  -0.093076  -0.086363  -0.073851  -0.053950  -0.040530  -0.028888  -0.012616  -0.003030  -0.002533  -0.003777  -0.003447  0.005791  0.021047  0.030303
```

Coffee0297, 2020-02-19:
and this is the data that the plugin gets and shows
[image]

Arksine, 2020-02-19:
Everything here looks correct.  The value for "points" in printer.cfg are the probed points.  Klipper uses these points internally to create the mesh. 

BED_MESH_OUTPUT outputs the entire interpolated mesh, which is the data visualizers expect.  Looking at the spreadsheet it seems that the plugin correctly fetched the mesh data.

FWIW, you can disable mesh interpolation by setting the following value in the [bed_mesh] section:
```
mesh_pps: 0
```

Keep in mind that the mesh data in BED_MESH_OUTPUT is printed with the y axis inverted.  When looking at mesh data it is as if  you are looking down at the bed, with the origin in the bottom left corner.  You may need to check the settings for your visualizer to make sure it knows the y axis is inverted.

Coffee0297, 2020-02-19:
it just seems difrent from the "old" post i found and seems too much up and down, but might just be me... 

will the ``mesh_pps: 0`` be a benefit at 12,12,3? or is it just for debug

-Tonny

Arksine, 2020-02-19:
Given that you are probing roughly 200x200mm with a dense 12x12 grid you likely don't benefit from mesh interpolation.  That said, you won't likely see any difference from turning it off.  Cubic interpolation with a low tension value at that density generally behaves like linear interpolation.

Coffee0297, 2020-02-19:
yea true that, i just think it looks too lumpy, but thanks so much for the clarification. it makes sense
and sorry to bother you :D

-Tonny

closed 2020-02-19 by Coffee0297