Klipper3d/klipper #2544: TUNING_TOWER not stepping properly on BAND.

ProfEngr, 2020-02-19:
I'm trying to tune some properties using the TUNING_TOWER feature and getting what I consider to be abnormal behavior. It worked fine yesterday when I just used START and FACTOR.
I tried: TUNING_TOWER COMMAND=SET_VELOCITY_LIMIT PARAMETER=ACCEL START=100 FACTOR=100 BAND=5 started with a response of max_acce: =350 then increments by 500 for every step in the print.
I tried: TUNING_TOWER COMMAND=SET_VELOCITY_LIMIT PARAMETER=SQUARE_CORNER_VELOCITY START=1 FACTOR=1 BAND=5 started with square_corner_velocity: 3.5 then goes to 8.5 and I cancelled the print at that point.
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4227349/klippy.zip)
Timestamp for the first instance is around noon on 2/19 and the 2nd was around 15:30 so you can match up the mess in the log file.

KevinOConnor, 2020-02-22:
It's not clear what you are reporting.  What did you expect to see?  What did you actually see?

-Kevin

ProfEngr, 2020-02-23:
Sorry, what I expected to see was the START value change by STEP every BAND mm. What I saw when testing ACCEL values was the value increasing by 500. Pasting a screen grab from my post on the Facebook Klipper group I joined.
[image]
When I ran the same command but for square corner: START=1, STEP=1, BAND=5 (only a 30mm tall print), the value hit 8.5 before mid-print and I cancelled it. Here are the serial log snips...

2020-02-19 15:26:29,118 - Send: TUNING_TOWER COMMAND=SET_VELOCITY_LIMIT PARAMETER=SQUARE_CORNER_VELOCITY START=1 FACTOR=1 BAND=5
2020-02-19 15:26:29,124 - Recv: // Starting tuning test (start=1.000000 factor=1.000000)

Layer 1 response:
2020-02-19 15:26:35,631 - Recv: // max_velocity: 200.000000
2020-02-19 15:26:35,635 - Recv: // max_accel: 200.000000
2020-02-19 15:26:35,639 - Recv: // max_accel_to_decel: 1500.000000
2020-02-19 15:26:35,641 - Recv: // square_corner_velocity: 3.500000

Layer 25 (Z=5mm) response:
2020-02-19 15:35:30,465 - Recv: // max_velocity: 200.000000
2020-02-19 15:35:30,469 - Recv: // max_accel: 200.000000
2020-02-19 15:35:30,474 - Recv: // max_accel_to_decel: 1500.000000
2020-02-19 15:35:30,487 - Recv: // square_corner_velocity: 8.500000

KevinOConnor, 2020-02-24:
I'm still not really sure what you expected, but your observed behaviour is the intended and documented behaviour.  From https://www.klipper3d.org/G-Codes.html#extended-g-code-commands : 

> If BAND is provided then the adjustment will only be made every BAND millimeters of z height - in that case the formula used is `value = start + factor * ((floor(z_height / band) + .5) * band`).

-Kevin

ProfEngr, 2020-02-24:
OK, I expected a start where I specified then an increment of STEP every BAND mm. Since that's not the coded behavior then I'll close the issue. Thanks.

closed 2020-02-24 by ProfEngr

KevinOConnor, 2020-02-24:
Okay, I understand now.  FWIW, there are two reasons the code was written to use the "average Z" for each band: 1) a number of tools don't react well to zero and so starting with a slightly larger number can simplify some tests, and 2) it makes the final measurement more accurate if the bands themselves aren't fully clear on the final print.

-Kevin

GarthSnyder, 2020-02-26:
I was going to file this same issue. Not that it's broken or doesn't work as documented, just that the conventions seem strange from a user's perspective.

In particular, it's counterintuitive that START isn't the actual starting value. For example, if you want an acceleration tower that starts at 200mm/s^2 and increases by 300mm/s^2 for every 5mm of height, you need a START value of 50. Wha?

If you're using bands, you probably have a per-band increment in mind as well. Why force the user to do the arithmetic to figure out what the corresponding increment per mm would be? 

In short, current syntax:

```
TUNING_TOWER COMMAND=M204 PARAMETER=S START=50 FACTOR=60 BAND=5
```

Really seems like it should be:

```
TUNING_TOWER COMMAND=M204 PARAMETER=S START=200 FACTOR=300 BAND=5
```

@KevinOConnor, in response to your comments above: 1) This point seems to envision that the normal use case is an implicit START=0. Maybe, but it seems unlikely to me - all the things people usually print towers for (speed, acceleration, corner speed, temperature) will have some nonzero value that the user wants to start at, no? 2) Why's that?

jordan-woyak, 2021-05-23:
I also think the current math the `BAND` parameter is counterintuitive and illogical.

`value = start + factor * ((floor(z_height / band) + .5) * band)`

The `+ .5` essentially adds `factor * band / 2` to the start value which is used for the first band.
This is silly. I want to "start" at the "start". This is the name of the parameter.

A sensible person would expect the first band to have the start value.

The `+ .5` should be removed.
`value = start + factor * (floor(z_height / band) * band)`

`START=10 FACTOR=1 BAND=10` would then produce bands of 10,20,30...

If I wanted the first band to be 15 I would have asked for it. :P

@KevinOConnor How would you feel about a PR changing this behavior?

buikpijn, 2021-10-11:
Glad i found this. I am struggling with the same lack of logic in these parameters.

buikpijn, 2021-10-11:
This confuses me even more; if I use the vallues you've posted it doesn't give the results you've  posted. (TUNING_TOWER COMMAND=SET_VELOCITY_LIMIT PARAMETER=ACCEL START=100 FACTOR=100 BAND=5)  I don't think layer height has anything to do with it, it's just Z height in mm (band) that is affecting the buildup in acceleration. 

I can't understand how the formula `value = start + factor * ((floor(z_height / band) + .5) * band)` (documentation found [url=https://www.klipper3d.org/Resonance_Compensation.html]here[/url]) TUNING_TOWER COMMAND=SET_VELOCITY_LIMIT PARAMETER=ACCEL START=1250 FACTOR=100 BAND=5 

results in start = 1500, increase 500 every 5mm. The formula has changed? So it's now value = ((start+(factor*band)/2)) (this also aplies to your outcome in the first post)