Klipper3d/klipper #2545: 'mcu' shutdown: Missed scheduling of next pwm event

IndicaHaze, 2020-02-19:
Hey guys! Been using klipper for a week now and love it! I started my first long print in it this morning(12 hours) and about half way through I noticed the printer had froze and Octoprint said that the MCU shutdown "'mcu' shutdown: Missed scheduling of next pwm event," Printer was unresponsive until I restarted both the printer and Octoprint. 

If this isn't the proper log please say so!

[klippy.log](https://github.com/KevinOConnor/klipper/files/4227784/klippy.log)

KevinOConnor, 2020-02-20:
You've found the right log file, but it doesn't contain information on the event you mentioned.  You can look in the /tmp/ directory to see if you can still find the logs from previous days.  (If you do, feel free to attach it here.)

In general, that type of error is due to a loss of communications between micro-controller and host machine - but without the log from the event there isn't much more info we can provide.

-Kevin

IndicaHaze, 2020-02-20:
Unfortunately that is the only log in the folder. Thanks for the response!

closed 2020-02-20 by IndicaHaze

FeepingCreature, 2020-10-02:
Just got this error in the middle of a print.

This is after I switched to master and enabled the MZV shaper.

Relevant log snippet:

```
Stats 24991.8: gcodein=18255438 mcu: mcu_awake=0.157 mcu_task_avg=0.000265 mcu_task_stddev=0.000232 bytes_write=37072871 bytes_read=5032920 bytes_retransmit=34 bytes_invalid=0 send_seq=664006 receive_seq=664003 retransmit_seq=27263 srtt=0.013 rttvar=0.001 rto=0.025 ready_bytes=1112 stalled_bytes=0 freq=16000058 heater_bed: target=60 temp=60.0 pwm=0.262 print_time=15960.612 buffer_time=3.130 print_stall=33 extruder: target=205 temp=205.0 pwm=0.415 sysload=4.55 cputime=937.882 memavail=666628
MCU 'mcu' shutdown: Missed scheduling of next pwm event
clocksync state: mcu_freq=16000000 last_clock=255318340232 clock_est=(24963.139 254861356260 16000058.093) min_half_rtt=0.000163 min_rtt_time=18706.657 time_avg=24963.139(843.461) clock_avg=254861356260.456(13495425735.401) pred_variance=821461.677
Dumping serial stats: bytes_write=37074254 bytes_read=5036039 bytes_retransmit=34 bytes_invalid=0 send_seq=664029 receive_seq=664026 retransmit_seq=27263 srtt=0.012 rttvar=0.003 rto=0.025 ready_bytes=0 stalled_bytes=65
Dumping send queue 100 messages
Sent 0 24989.625705 24989.620945 61: seq: 17, queue_step oid=1 interval=1705 count=411 add=0, queue_step oid=1 interval=1732 count=99 add=0, queue_step oid=2 interval=8035 count=36 add=12, queue_step oid=1 interval=1765 count=68 add=0, queue_step oid=1 interval=1809 count=64 add=0, queue_step oid=2 interval=8539 count=45 add=23, queue_step oid=1 interval=1830 count=87 add=1, queue_step oid=1 interval=1912 count=80 add=1, queue_step oid=1 interval=2018 count=94 add=1
Sent 1 24989.628225 24989.620945 63: seq: 18, queue_step oid=2 interval=9635 count=36 add=33, queue_step oid=1 interval=2150 count=71 add=1, queue_step oid=1 interval=2243 count=74 add=2, queue_step oid=2 interval=10885 count=31 add=49, queue_step oid=1 interval=2358 count=65 add=3, queue_step oid=1 interval=2518 count=128 add=3, queue_step oid=2 interval=12480 count=25 add=75, queue_step oid=1 interval=2933 count=77 add=4, queue_step oid=2 interval=14460 count=19 add=117
Sent 2 24989.639647 24989.637287 59: seq: 19, queue_step oid=1 interval=3274 count=72 add=6, queue_step oid=2 interval=16769 count=16 add=191, queue_step oid=1 interval=3744 count=57 add=9, queue_step oid=2 interval=20040 count=11 add=311, queue_step oid=1 interval=4304 count=47 add=14, queue_step oid=1 interval=5014 count=37 add=23, queue_step oid=2 interval=23755 count=8 add=525, queue_step oid=1 interval=5912 count=30 add=40
Sent 3 24989.642207 24989.637287 64: seq: 1a, queue_step oid=2 interval=28159 count=7 add=950, queue_step oid=1 interval=7190 count=21 add=72, queue_step oid=2 interval=35585 count=4 add=1776, queue_step oid=1 interval=9116 count=6 add=186, queue_step oid=2 interval=43354 count=2 add=3345, queue_step oid=1 interval=9742 count=21 add=71, queue_step oid=2 interval=31984 count=2 add=-3470, queue_step oid=2 interval=27565 count=3 add=-141
Sent 4 24989.644567 24989.637287 59: seq: 1b, queue_step oid=2 interval=23435 count=2 add=-4887, queue_step oid=1 interval=12122 count=4 add=407, queue_step oid=2 interval=17545 count=6 add=-234, queue_step oid=1 interval=12955 count=16 add=86, queue_step oid=2 interval=16226 count=3 add=-815, queue_step oid=2 interval=14042 count=8 add=0, queue_step oid=1 interval=14957 count=3 add=914
Sent 5 24989.654936 24989.652536 60: seq: 1c, queue_step oid=2 interval=14354 count=9 add=235, queue_step oid=1 interval=16416 count=3 add=868, queue_step oid=1 interval=23286 count=7 add=335, queue_step oid=2 interval=16839 count=3 add=726, queue_step oid=2 interval=16866 count=9 add=51, queue_step oid=1 interval=26078 count=3 add=1736, queue_step oid=2 interval=15274 count=12 add=-266
Sent 6 24989.657496 24989.652536 64: seq: 1d, queue_step oid=1 interval=46077 count=2 add=25541, queue_step oid=1 interval=83058 count=1 add=0, queue_step oid=2 interval=11499 count=16 add=-167, set_next_step_dir oid=1 dir=1, queue_step oid=1 interval=1505137 count=1 add=0, queue_step oid=2 interval=8952 count=21 add=-86, queue_step oid=2 interval=7256 count=27 add=-48, queue_step oid=2 interval=6021 count=40 add=-27, queue_step oid=2 interval=4998 count=51 add=-16
Sent 7 24989.660056 24989.652536 64: seq: 1e, queue_step oid=2 interval=4226 count=35 add=-7, queue_step oid=2 interval=4028 count=35 add=1, queue_step oid=2 interval=4112 count=38 add=9, queue_step oid=2 interval=4507 count=44 add=17, queue_step oid=1 interval=188948 count=1 add=0, queue_step oid=2 interval=5300 count=38 add=29, queue_step oid=1 interval=55660 count=2 add=-904, queue_step oid=1 interval=57112 count=2 add=-3420, queue_step oid=2 interval=6225 count=26 add=27
Sent 8 24989.671571 24989.669091 62: seq: 1f, queue_step oid=1 interval=20080 count=7 add=-307, queue_step oid=2 interval=6714 count=13 add=65, queue_step oid=1 interval=18263 count=6 add=460, queue_step oid=2 interval=7165 count=13 add=17, queue_step oid=1 interval=19577 count=2 add=-5445, queue_step oid=2 interval=7759 count=17 add=17, queue_step oid=1 interval=11525 count=12 add=-106, queue_step oid=2 interval=8846 count=5 add=193
Sent 9 24989.673971 24989.669091 60: seq: 10, queue_step oid=1 interval=10837 count=11 add=126, queue_step oid=2 interval=9165 count=11 add=34, queue_step oid=2 interval=10570 count=6 add=162, queue_step oid=1 interval=10913 count=5 add=-392, queue_step oid=1 interval=9465 count=6 add=-205, queue_step oid=2 interval=10938 [... cut ...]
Read 24991.551949: 'N431772 G1 X171.771 Y121.29 E1276.5775*107\n'
Read 24991.557588: 'N431773 G1 X170.747 Y122.035 E1276.60277*100\n'
Read 24991.563194: 'N431774 G1 X169.49 Y122.035 E1276.62786*91\n'
Read 25023.465876: 'N431775 M105*20\n'
gcode state: absolute_coord=True absolute_extrude=True base_position=[0.0, 0.0, 0.0, 8315.086449999999] last_position=[169.49, 122.035, 29.274, 9591.71431] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=1.0 speed=100.0
Reactor garbage collection: (23944.669809199, 24967.685003652, 0.0)
```