Klipper3d/klipper #2546: Klipper + Bigtreetech tft35 (touchscreen mode)

ice2009, 2020-02-20:
Hello :) 

I have question it is possible use bigtreetech tft35 with touchscreen mode for klipper ? Because i connect this display to board and display says printer not connected. Any people solved this problem ?

Thank's

jakep82, 2020-02-20:
Touch screens are not supported in Klipper.

ice2009, 2020-02-20:
Yes but btt firmware send gcode commands to printer

jakep82, 2020-02-20:
This has been asked and answered numerous times.  This screen is not supported in Klipper.

KevinOConnor, 2020-02-21:
Looks like a dup of #2231. 

-Kevin

KafrouniChris, 2020-02-21:
Even the emulation doesn't work right? Would there be an easy fix?

jakep82, 2020-02-21:
@KafrouniChris A developer would need to add support, and nobody is currently working on this.
 See this comment.  https://github.com/KevinOConnor/klipper/issues/2231#issuecomment-566371883

DeviousPenguin, 2020-02-23:
> Even the simulation of the lcd12864 doesn't work right? Would there be an easy fix?

Why wouldn't the lcd12864 simulation not work?  I'm looking to run a TFT24 which is similar to the TFT35 in lcd12864 mode, but if it doesn't work then I may order a different screen.

jakep82, 2020-02-24:
> Why wouldn't the lcd12864 simulation not work?

This is answered at the link I provided above.

DeviousPenguin, 2020-02-24:
Thanks, I just assumed the lcd12864 simulation would be OK, looks like the simulation is not perfect, I'll have to try updating the firmware on the TFT24 and see it can be patched.

KevinOConnor, 2020-03-05:
Lets close this if preference to the existing ticket #2231 .

-Kevin

closed 2020-03-05 by KevinOConnor