Klipper3d/klipper #2547: Stepper X issue on multiple printers - how to debug?

vladbabii, 2020-02-21:
Some background

Firmware - v0.8.0-327-g15f21be5

(A) bedslinger
- mks gen 1.4
- 4 x 2208 in uart mode (x,y,z,z2) lv8729 (e)
- 2.5 A steppers on all of them
- bltouch as probe, using bed mesh (auto created)
- optical sensors on x and y 
- klipper version v0.8.0-327-g15f21be5
- connected via usb to a pi 3 (using it's own official power brick) - rasbian lite from august 2019
- 24v
- mosfets for hotend and bed
- built: early 2018
- last klipper update: a couple of days ago, sd card broke on pi and had to reinstall everything (and updated firmware)

(B) small corexy
- skr 1.3
- 2 x 2208 in uart mode (x,y) + lv8729  (z,e)
- 2.4 A stepper on e, rest are 1.7 A
- z probe using an endstop on the frame, triggered  by bed (fixed position)
- mechanical endstops on x and y
- klipper version v0.8.0-327-g15f21be5
- connected via usb to a pi 4 (using it's own official power brick) - latest rasbian lite
- 24v
- mosfets for hotend and bed
- built: started about 4 weeks ago,  finished a couple of days ago, installed everything on a new pi 4 then upgraded to newer firmware about 3 or 4 days ago

(C) big corexy toolchanger 
- a couple of skr v 1.1  for x, y and z, msk gen 1.4 for hotend and bed
- tmc 2209 in uart mode for x, y, z, z2, z3 (revering now to 5160s for and y in spi to check), 2208 in standadlone mode for toolchanger, lv8729 for e
- 0.9 A, 2.5 Steppers for x and y, big 2.5A steppers on Z (e3d super whooper), 1.7A steppers on E, 1.6 A steppers on toolchanger
- built: started on june 2019, working partially in december 2019
- 12 and 24v depending on board
- connected to a laptop running x86 ubuntu

Short version
(A)
1. start print
2. print goes fine for a while
3. X stepper stops working (no movement), only the Y stepper moves
Tried 10 prints, all failed

(B)
1. start print
2. print goes fine for a while (random period of time, 1 layer or 6 layers at 2.4 mm)
3.  X stepper goes forward and back a few degrees, Y and Z work fine
4. Tried 10 prints, 4 failed

(C)
1. start print
2. print goes fine for a while (random period of time, 1 layer or 6 layers at 2.4 mm)
3.  X stepper goes forward and back a few degrees, Y and Z work fine
4. Tried 5 prints, 3 failed

For all 3 printer in the logs i see nothing out of the ordinary. Tmc dump gives a short on A or B, but that would mean stepper is non-functional - but after restart the stepper works fine for a while.

(1) How can i debug this better? I will allocate some time this evening to gather logs for all printers and post them in this thread but there's anything else I can do please let me know.

(2) Was there any recent changes regarding stepper movement, tmc drivers ? I want to try reverting to older versions and test the printers again.

Long version - timeline
On (A) the pi sd card broke and i had to reinstall everything. I used newer version of klipper (software and firmware) and raspbian. This printer has been working for about a year without major issues (belt tensioning, screw tightening, cleaning - normal wear and tear). I am using the same slicer with the same settings.

1. start print
2. print goes fine for a while
3. X stepper stops working (no movement), only the Y stepper moves

Ok, i said bad wiring... checked cables with multimeter - they seem fine, replaced them anyway - issue persists
Then I checked tmc stepper settings and lowered voltages - no change

If I jog the printer from octoprint interface or from terminal with G0 X works with zero issues. Once i start printing the issue appears again, usually on a alyer above 1 mm and i have to restart to make things work again.

Next, i ordered a new stepper which I got today.

I connected the X stepper to another skr mini board and played with octoprint moving it left/right - no issues whatsoever. I made a script to run commands for 2 hours - all ok. Started a print.. at about 1.5 mm height it broke again...

Ok, so printer broke, I moved to printer (B) to print some stuff, and I updated it since i needed the macro override functionality. The second print i got the same issue - X stepper stops working correctly (moves back and foward a couple of degrees - maybe 1-2 degrees)

Now i started to suspect that my first printer doesn't have a hardware issue, since having the same hardawre issue on both printers with different components at the same time is highly unlikely.

On (B):

1. start print
2. print goes fine for a while (random period of time, 1 layer or 6 layers at 2.4 mm)
3.  X stepper goes forward and back a few degrees, Y and Z work fine

A couple of times the print finished, but it's random...

I then updated printer (C) and it works like (B). This is when i decided to post here, maybe you can give me some ideas on what to test. There is nothing in a log that's special between the two runs.

I will  
- add more logs in the next 24 hours
- replace stepper x on the (A) printer
- if issue persists with replaced stepper: 
  - get some logs from (A) printer with current fw version
  - downgrade to previous version and try printing
  - add those logs as well

Thank you for your time, hope I'm not missing something very obvious here.

jakep82, 2020-02-21:
An issue with 2208 drivers was fixed yesterday.  Might be related.

https://github.com/KevinOConnor/klipper/commit/e03f75e7583bd510cc06908fe38b8f676a09f6d3

jakep82, 2020-02-21:
See https://github.com/KevinOConnor/klipper/issues/2516 for more info.  You appear to have the same issue.

vladbabii, 2020-02-22:
Thank you @jakep82

vladbabii, 2020-02-22:
A weird issue is that TMC_DUMP was reporting a short on A on the X stepper.

ld3300, 2020-02-22:
From the TMC2208 Datasheet: 

> 2-Phase stepper motors up to 2A coil current (peak)

I tried to run a NEMA17 motor from my box of motors that I had purchased some years ago as an extra extruder with a TMC2208 driver and it would sometimes work for a bit, then start underextruding, give up for a bit, start working again, etc.  Turned out the stepper had too high of a coil current to work, when I replaced it with a new one purchased online the problem went away.

vladbabii, 2020-02-24:
@ld3300 The (A) printer has been working for a year, printing about 16-20 hours per day in average, at high speeds. Now it's failing at 1/3 of those speeds while i can move the gantry with one hand. I am aware some stepper and driver combinations will not work well, but what i have worked very well until last update. What threw me off initially was the tmc short on A error. I updated (B) to latest klipper and it works fine (no shorts, no errors).

ld3300, 2020-02-24:
cool, glad you aren't experiencing my mistake, though I wish I could help out more.

closed 2020-02-28 by klipper-gitissuebot