Klipper3d/klipper #2548: BTT Relay Module Support

kurtisane, 2020-02-21:
Hi guys,
I'm running Klipper in combination with the SKR PRO and I have the Relay Module from BigTreeTech. 
I was wondering if there is a feature in Klipper to control the Relay Module that is connected to the MCU ?

Regards

ld3300, 2020-02-22:
You can define a pin like is described in example_extras.cfg

kurtisane, 2020-02-23:
> You can define a pin like is described in example_extras.cfg

I am not sure which section you mean. I searched for the terms "relay"

ld3300, 2020-02-23:
[https://github.com/KevinOConnor/klipper/blob/e03f75e7583bd510cc06908fe38b8f676a09f6d3/config/example-extras.cfg#L1198](https://github.com/KevinOConnor/klipper/blob/e03f75e7583bd510cc06908fe38b8f676a09f6d3/config/example-extras.cfg#L1198)

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like the original question was answered (one can use an output_pin config section to configure an external relay).

-Kevin

closed 2020-03-21 by KevinOConnor