Klipper3d/klipper #2550: Longer prints always fail (Lost communication with MCU?)

VanKurt, 2020-02-22:
Whenever I try to make alonger print (> 2hours), the print fails. The printer just stops.
In the klippy.log I don't see anything suspicious. The octoprint log however seems to be full of 

> | Recv: // Lost communication with MCU 'mcu'

which can't be good, right?
Is there any more info on what exactly is going wrong to be gained from the logs? I really would like to fix this thing :-)

[octoprint.log](https://github.com/KevinOConnor/klipper/files/4238850/octoprint.log)
[klippy.log](https://github.com/KevinOConnor/klipper/files/4238849/klippy.log)

vladbabii, 2020-02-22:
1. Check power issues on the pi
2. Check power on the board
3. Usb cable can become loose due to vibrations - make sure it's held in place somehow on both ends
4. If you have a relay between the pi usb and the board, try removing it
5. If the usb cable is not shielded properly and you have other wires (especially bed or hotend heater) running paralell to it it can also cause it
6. If nothing seems to work, try a new/different sd card in the pi - if it has issues it can cause freezes

What board are you using and how is it connected?

vladbabii, 2020-02-22:
Also, once you get that error, open a shell to the pi and run this command

```
dmesg
```

The last couple of lines will give you a linux error message if something broke on the usb connection - if there are any issues there it's not a klipper issue.

KevinOConnor, 2020-02-22:
Also, see the FAQ: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors

-Kevin

VanKurt, 2020-02-22:
I'm currently trying to replace all parts in question (power supplies, cables etc). Let's see if that helps :-)
Since this is clearly not a Klipper issue, I'm closing this.

VanKurt, 2020-02-26:
In the mean time I replaced a lot of stuff:

- The old Raspi 3B is gone, replaced with a new Pi 4
- New, official Raspberry 4 power supply
- New SD card
- Using latest OctoPi image (0.17 instead of 0.16)
- New USB to printer cable (also removed VCC pin from cable to make sure printer is only powered by its own supply, not via USB)

What I did not replace so far (things might get expensive):

- New Creality Ender 3 power supply
- New Creality Ender 3 mainboard

Still the print failed after 8 hours :-(

Before I do something stupid (like thow the printer out of the window or going back to marlin/SD card printing without Klipper and Octoprint), is there any more information to be gained from the logs?

Here are some more details:

- the display on the printer started to show some weird stuff after a few minutes. It seems to have gotten worse over time. Attached is a photo of the "final stage" this morning, when the print had failed.
[image]
 - As recommended above, I called the dmesg command (see attached file)
[dmesg.txt](https://github.com/KevinOConnor/klipper/files/4253685/dmesg.txt)
- Also the latest klippy.log and octoprint.log are attached
[klippy.log](https://github.com/KevinOConnor/klipper/files/4253688/klippy.log)
[octoprint.log](https://github.com/KevinOConnor/klipper/files/4253690/octoprint.log)
- I have a USB Webcam attached to my PI. May that somehow cause trouble?

Arksine, 2020-02-26:
You seem to be having the same problem reported in #1919. A search for the error in dmesg.txt reveals that it could potentially be a kernel issue.  Specifically,  raspberrypi/linux#1187 and raspberrypi/linux#2406 appear to be related. 

You could try disconnecting the webcam, or any of the solutions outlined in the issued above. 
 Alternatively, if you have a spare x86 pc/laptop sitting around, you could try using it as a dedicated Klipper host.

vladbabii, 2020-02-26:
For what is worth, i never had an usb connectivity error on my printers connected to x86/x64 laptops. Only to PIs

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like the conversation has concluded.  Many people successfully run Klipper on USB connections that stay up for multiple months - as far as we know, the failing USB connections on some printers is due to a hardware issue of some kind.

-Kevin

closed 2020-03-21 by KevinOConnor

joekochevar, 2020-05-08:
I'm also having this issue, but I noticed a common thread among a few of these, which is the plugin "theSpaghettiDetective," paired with USB webcams. I'm not positive, but I have a feeling that this plugin might be interacting poorly with Klipper.

I deactivated that plugin and downgraded my USB bus to 1.1 speeds and haven't had the issue in 26 hours and counting (where it previously failed at the 13 hour mark without fail). Will try reverting to USB 2.0 while keeping TSD deactivated next.

darkpoetry, 2020-11-10:
Not to reopen, but just observing that connecting a cellphone webcam to the same computer driving the printers also caused this havoc for me. Best to use a dedicated charger for those.