Klipper3d/klipper #2551: STM32 (SKR Pro)@24v + TMC5160 Extruders do not work

BoHoehne, 2020-02-22:
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/4239042/klippy.log.zip)

I am having the problem, that the extruder does not start.
If it starts it stalls at low speeds or small movements. 

When i feel the titan extruder gear, it is moving very weakly, too weakly to transport filament.
If i slightly push the main gear slightly, the extruder starts working for a while, only to stop at the next retract.

I tried stepsticks from bigtreetech and watterott. Both deliver the same result.

Any help is welcome.

[image]

BoHoehne, 2020-02-22:
I just tried a 2208 in standalone mode on the extruder.

It worked flawlessly.

The problem points to the TMC integration in klipper.

BoHoehne, 2020-02-22:
Just to summarize:

SKR Pro v1.1 (STM32F4xx)
The SKR Pro v1.1 is a brand new board (1 week old).
PSU is 24v
Titan extruder

I tested the following stepper drivers:
TMC2208 (standalone mode) behaving almost normal but stalling later in print (end of 1st layer), needs a push to recover to normal operation
TMC2100 not moving at all
TMC5160 stalling on every retract and needs a push to recover operation
DRV8825 not moving at all 

On every test i checked for filament friction by hand. The force needed was low and the hotend was not clogged. I also check for heat creeps (all metal hotend), which did not occur.
Extruder sockets 1-3 behaving the same (all tested).

Anyone seen this ?

BoHoehne, 2020-02-22:
If someone running a SKR-Pro v1.1 could share his/her config file, that would help too.

theopensourcerer, 2020-02-22:
I've got a SKR Pro v1.1 on my bench (12v right now) and have tested DRV8825, 2208 & 2130s and all see to be working OK. I use the example config file in (generic-bigtreetech-skr-pro.cfg) and the jumper settings from here:

https://drive.google.com/file/d/1AQ9loH8QoqeO-hCVQ90UwKXFT7Eu3tHw/view

Sorry but don't have time to start system up right now and share config. Maybe tomorrow if you still need it.

BoHoehne, 2020-02-22:
Thanks, i used that config as well.
I replaced the stepper motor and the titan extruder.
No change.

Anyone running a SKR Pro v1.1 @ 24V ?

KevinOConnor, 2020-02-22:
Try giving the code at PR #2534 a test to see if it improves things.  Also, please run `DUMP_TMC STEPPER=extruder` at startup and after the event - that will provide extra information in the logs.

-Kevin

BoHoehne, 2020-02-22:
I tested it just now.
It did not change anything.

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/4239555/klippy.log.zip)

BoHoehne, 2020-02-22:
Btw. in an earlier test, i also changed the stepper drivers x <-> e0. 
X worked as expected, extruder didn't.

BoHoehne, 2020-02-23:
This is a little bit embarrassing for me.

Last night i had a little one man electronics tinkering party. 
I disassembled the extruder, dismounted the stepper motor, disassebled the wire loom, etc.

My problem was actually a mixture of two problems: 
1. A broken titan extruder casing (the lower side) which allowed the main gear to evade when pressure was applied.
2. A short curcuit in the wire loom connecting the stepper motor, basically the wire was broken.

I then pulled the latest Klipper version and did a SUCCESSFUL testprint.

32bit + tmc5160 + klipper = WOW !!!

Klipper was not the problem, thanks to everyone for the assistance.

closed 2020-02-23 by BoHoehne