Klipper3d/klipper #2552: Probblem with klipper + octoprint

ice2009, 2020-02-22:
Hello :) 

I installed new klipper + octoprint today but i have probblem with installation if i use command /dev/serial/by-id/* raspberry says ls: cannot access '/dev/serial/by-id/*': No such file or directory. Can you help me ?

Thank's

ice2009, 2020-02-22:
Here is log file 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4239827/klippy.log)

ld3300, 2020-02-22:
what does it show if you type `lsusb`?

Achim01, 2020-02-23:
Try this:
ls -l /dev/serial/by-id/
I think you forgot a space. (before and/or after -l)

stelgenhof, 2020-02-23:
Are you sure the cable is properly connected? This error message can indicate no serial port is available. You can try to connect the USB cable and restart your RPi.

Spacemarine2018, 2020-02-23:
Look at the system log with the command dmesg
Then unplug and replug the cable and do again dmesg

Every plug or unplug operation should show up in dmesg.

closed 2020-03-21 by KevinOConnor