Klipper3d/klipper #2553: Bed levelling moves nozzle out of range

VanKurt, 2020-02-22:
I have set up the levelling as follows:

> mesh_min: 55.0,30.0
> mesh_max: 240.0,220.0
> probe_count: 5,5
The bed levelling then fails:

> Recv: // probe at 106.000,38.000 is z=0.630000
> Recv: // probe at 106.000,38.000 is z=0.627500
> Recv: // probe at 152.250,38.000 is z=0.597500
> Recv: // probe at 152.250,38.000 is z=0.600000
> Recv: // probe at 198.500,38.000 is z=0.587500
> Recv: // probe at 198.500,38.000 is z=0.587500
> Recv: // probe at 244.750,38.000 is z=0.630000
> Recv: // probe at 244.750,38.000 is z=0.635000
> Recv: !! Move out of range: 291.000 38.000 5.000 [0.000]
> Recv: !! Move out of range: 291.000 38.000 5.000 [0.000]

The x value 291mm is clearly out of the max range of 240mm.
Why is the bed mesh levelling trying to go there?

[klippy.log](https://github.com/KevinOConnor/klipper/files/4239888/klippy.log)

VanKurt, 2020-02-22:
I just realized that the coordinates are relative to the probe, not the nozzle. Ugh!
That explains the weird behavior. But it also makes it very difficlult to get the correct min/max/x/y values. Currently I'm moving the printer head to the four corners of my bed, then I use M114 to get the X/Y values. Since M114 gets the coords of the nozze (and not the sensor probe), these values cannot be used directly, which is rather unintuitive...

BlackStump, 2020-02-22:
I agree with you the original relative to nozzle was more intuitive yet people complain enough to say it was wrong. you can please some people some of the time but not all people all of the time.
so it is what it is.

VanKurt, 2020-02-23:
> I agree with you the original relative to nozzle was more intuitive yet people complain enough to say it was wrong. you can please some people some of the time but not all people all of the time.
> so it is what it is.

I can't see any benefits of the new system. Now we have to worry about two coordinate systems instead of one. But well, you should never look in a gift horse's mouth ;-)

Are there any other settings that also refer to probe-units instead of nozzle-units? I wouldn't like to run into any more bad sursprises...

Arksine, 2020-02-23:
There was a lot of feedback that nozzle relative coordinates were confusing, this is particularly true of users transitioning from Marlin.  The primary benefit to specifying probe relative coordinates is that it allows the user to specifically define the mesh area on the print surface.

I understand that not everyone is happy with it, but unfortunately its impossible to make everyone happy.  FWIW its a good idea to check Config_Changes.md when you upgrade so there are no surprises.

As of right now none of the other modules are probe relative.  Bed Mesh is somewhat unique in that it generates the majority of its points, whereas the other modules require that you specify them.  IMO this makes the idea of changing their coordinates a bit less palatable.

KevinOConnor, 2020-03-21:
I'm going to close this issue as it looks like a solution was found for the original issue.

-Kevin

closed 2020-03-21 by KevinOConnor