Klipper3d/klipper #2554: Wait for user input

hermes3008, 2020-02-23:
Not an issue just a request add a Goode or command wait for user input like to manually install probe and confirm

hermes3008, 2020-02-23:
If it could use the screen while waiting like install probe then press knob or center button

ld3300, 2020-02-23:
Right now you can make a pause with an M117 notification and create a menu item with the functions you want that would include a resume command when operation completed.  Describe how else you would like it to function?  Maybe a G-Code that would take you to the menu screen you need?

StrikeEagleCC, 2020-02-24:
I was reading through the docs looking for this functionality, but hadn't found it yet. Thanks @ld3300 for the tip! Is there documentation for how to do this yet, or do you have an example you could share?

hermes3008, 2020-02-24:
Yes I would like have a G-Code that would take you a menu screen
The code go something like

M117 message
Pause

Or

Menu ?
Pause

ld3300, 2020-02-24:
I use this currently:
```
[gcode_macro SETUP_FILAMENTS]
gcode:
  M117 CHECK FILAMENTS
  RESPOND TYPE=command MSG="CHECK FILAMENTS"
  PAUSE 
```
I would also like to be able to use gcode macros to order up menu items.  I started to look through it, but haven't quite found the right path yet.  I also heard that there may be some major menu changes, so I'm not sure if the work would be beneficial long.

hermes3008, 2020-02-25:
As far as was messing with menu Comand I found one arguement so far (Do) i have not figured out the actions to apply for Do. but planing to find it in the code

I was thinking to bring up a hiden Menu item to keep user from moving head before continuing

ld3300, 2020-03-01:
#2575

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like a workaround was found and the conversation has concluded.

-Kevin

closed 2020-03-21 by KevinOConnor

NeoMatrixJR, 2020-06-04:
Can someone who's got this working post a sample config?  For me PAUSE doesn't seem to do anything....everything I set up just keeps on going.  Could really use a working example.

ld3300, 2020-07-07:
hi @NeoMatrixJR put the command CLEAR_PAUSE somewhere in your startup and make sure it is called anytime printing is resumed from octoprint as opposed to resuming from the LCD.  You can actually call it any time after you have resumed a print to make sure Klipper is up to date on the pause state.

matti125, 2021-09-09:
I do not have a display on my printer, I control it throught mainsail/console. II do not seem to be able to use PAUSE in a start-the-print -macro, so that I could prompt the user to either do bed mesh calibration (which requires manual probe arming) or just continue with thw print. A PAUSE in a macro does not seem to stop processing the other macro steps in the same macro, which is understandable. But how then could I ask the user to choose when starting a print? Should I be able to do so without a display? Am I just missing how the PAUSE should work?