Klipper3d/klipper #2557: Voron 2.2 Nozzle Probe Issue

Chezz1989, 2020-02-23:
Hello dear community,

I have build an Voron 2.2.

I have been struggling with the correct function of gantry leveling for days. The function of homing is carried out without problems, but when I run the G32, the Z_Probe triggers without touching an obstacle. 1000 pages browsed, 100 times viewed the software ... tried etc.

I'm desperate.

thanks, people
Chris

[klippy.log](https://github.com/KevinOConnor/klipper/files/4241154/klippy.log)

hevilp, 2020-02-25:
Take a look at the horizontal_move_z: 5 and increase it.

Your probe is probalby triggered while horizontal movement. (Probe triggered prior to movement in your log)

Chezz1989, 2020-02-25:
i have increase it (try Z30, 100) but the problem is still there.

it seems like he's after homing X, Y, and Z and triggered he lift up a little bit and when he moved to the middle at this moment the probe triggered but there is nothing no heatbed no physical thing there!

Thank you for your efforts!

Chezz1989, 2020-02-29:
I have solved the problem! The reason was the voltagedrop to the probe... the cable was to long (2.5 meters)

closed 2020-02-29 by Chezz1989