Klipper3d/klipper #2558: Klipper exists on extrusion/retract after switch to manual stepper

Spacemarine2018, 2020-02-23:
Disclaimer: I'm building a pen-plotter, so I'm abusing the inteded functions quite a bit.

Since I'm only doing extrusions of a constant length while the head is standing still, I exchanged the [extruder] section with a [manual stepper]. This makes controlling it much easier to tune the movement.

When I'm using the manual stepper commands to move my extruder, klipper suddenly quits with the error message: MCU 'mcu' shutdown: Timer too close

However, using my oscilloscope with a current probe, I verified that the stepper does exactly the same movement as before with the old [extruder] section. So the generated step-rate should be identical as before. I even lowered the microsteps from 64 to 16 but still have the same problem. 64 micro-steps worked perfectly before.

Are manual stepper commands somehow more demanding on the MCU?
[klippy.log](https://github.com/KevinOConnor/klipper/files/4241296/klippy.log)

Arksine, 2020-02-23:
I can only make an assumption here because this doesn't include your configuration with using the extruder, but its likely that the extruder section was capping the speed/acceleration.  Your speeds/accelerations are way too high.

Spacemarine2018, 2020-02-23:
The extruder section contained exactly the same values. And as I wrote above:

"However, using my oscilloscope with a current probe, I verified that the stepper does exactly the same movement as before with the old [extruder] section."

Also I can extrude like 5 times and it only goes wrong after the 6th time (or on some other time). This is also independent of the microstep-count. If your assumption were correct, 64 microsteps should trigger this error much earlier than 16. But that is not the case.

KevinOConnor, 2020-02-24:
The Klipper software is not designed (nor tested) to work with absurd values in the config.  Set the config back to normal values and if you continue to have an issue then post the log from that session.

-Kevin

Spacemarine2018, 2020-03-02:
I can confirm that the problem is gone if I reduce both speed and acceleration by a factor of 10. However, If I use the old extruder section (no manual stepper) I can still use the high values without any problems.

It seems that movements by manual steppers are treated differently than movements on the extruder.

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like a solution was found for the original issue.

-Kevin

closed 2020-03-21 by KevinOConnor