Klipper3d/klipper #2559: The klippy host software is attempting to connect

UncleCCC, 2020-02-23:
Hello
I just tried to connect Klipper to my Ender3pro. The installation worked without problems according to the instructions. 
Unfortunately Klipper can not connect to the printer. What am I doing wrong?
I have attached the log, thanks for the help
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/4241781/klippy.log.zip)

StrikeEagleCC, 2020-02-24:
I was just about to open the same issue.
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/4243033/klippy.log.zip)

I also have an Ender3 Pro, and have been running klipper mostly successfully for many months. I updated last night, including re-flashing the board, and have been unable to connect since. I had a number of config incompatibilities that I sorted out, but when the problem persisted after that, I replaced my config with Ender3 example.

The only notable differences that I see between my log and @UncleCCC's is that I tried both /dev/ttyUSB0 and the port as reported by `ls /dev/serial/by-id/*` in the `[mcu]` section, and that I didn't wait as long before sending M112, so he has more timeout reports.

Perhaps relevant (but probably not helpful) is that prior to the update, if the printer was power cycled, the screen would stay blank and the printer unresponsive until the raspberry pi was restarted. I never investigated this behavior, and I've removed the log since then, so I don't have any debugging information. I also don't remember which combinations of restarting firmware/host i tried without success, but restarting the pi fixed it every time and wasn't a significant inconvenience.

netonjm, 2020-02-24:
The same problem here! I thought it was something related to my current printer (CR10SPro)... good to know!! 
Here my log details: https://gist.github.com/netonjm/b24cc3cbf511b9b10c58e61468086004

StrikeEagleCC, 2020-02-25:
@netonjm @UncleCCC , was this your first install, or were either of you updating? Can anyone with a working install for one of these creality boards comment with the version?

netonjm, 2020-02-25:
> @netonjm @UncleCCC , was this your first install, or were either of you updating? Can anyone with a working install for one of these creality boards comment with the version?
Solved! The problem was the baud rate. Now I’m working in the pin out! Thanks!

StrikeEagleCC, 2020-02-25:
Glad you got it working! What did you change? I think the default baudrate when compiling the firmware is 250000, and the same if you don't specify it explicitly in the config.

netonjm, 2020-02-25:
> Glad you got it working! What did you change? I think the default baudrate when compiling the firmware is 250000, and the same if you don't specify it explicitly in the config.

Yeah, for some reason I manually changed to match with my value previously set in Marlin to 115200 but it seems this doesn’t work here. Now working in the pinout, it seems there is no configuration example for CR10S Pro 😢

StrikeEagleCC, 2020-02-25:
@UncleCCC , did you happen to change your buad rate when you did `make menuconfig`?

netonjm, 2020-02-25:
[image]

yay! :-)

StrikeEagleCC, 2020-02-25:
Mine is fixed. Uninstalled with the script and reinstalled. I probably messed up something during the update.

StrikeEagleCC, 2020-02-26:
@UncleCCC , were you able to get yours connected?

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like the conversation has concluded.

-Kevin

closed 2020-03-21 by KevinOConnor