Klipper3d/klipper #2560: Non-accurate temperatures after updating Klipper and Z gantry slopping in probe_calibrate mode.

sinicolau6, 2020-02-24:
So i have 2 problems. I am trying to calibrate the Z-probe and every time i try to move it upwards with the testz z=+10 command, the z axis moves up, then it reaches its intended height then slopes down a bit. I tested it and it doesnt actually move downwards, its as if klipper is moving the Z axis a bit more than the intended height then lowering it to the good height.

The second problem is as follows:

After someone suggested i update klipper to fix the aformentioned problem, i did so and now my temperatures are stuck around 33 degrees and it takes forever for the Bed to cooldown in PID tune mode, and the Extruder overshoots when calibrating it in PID tune mode. Plus, the previously mentioned problem hasnt been fixed.

So i decided to go to Marlin and see if the problem was my printer(which worked perfectly before updating klipper) and to my surprise, where klipper got stuck showing My Extruder temperature at 32.9 degrees, Marlin showed the actual temp being 25.5.

StrikeEagleCC, 2020-02-25:
What printer are you using? Is it possible the config file is specifying the wrong type of thermocouple? What do you mean by "slopes down a bit"?

sinicolau6, 2020-02-25:
Im using the Ender 3 with the SKR E3 Mini board. It shoudnt be possible for the config to have the wrong thermocouple for both the nozzle and bed sincer it worked perfectly prior to updating, and i didn't change anything in the config file.

By sloping down I mean that it raises up, and instead of freezing in place once it gets to the desired point, the Z axis goes down a bit.

sinicolau6, 2020-02-25:
UPDATE: I let the printer cool down over night, and now i booted it together with Klipper, and it seems that Klipper automatically heats up the Nozzle and Bed to 33 degrees. I tried setting the temps to 20 from OctoKlipper and it seemingly doesnt go down at all.

l-Kage-l, 2020-02-25:
Here is the Klippy log.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4250658/klippy.log)

l-Kage-l, 2020-02-25:
Yeah, its probably what you said, that being normal for the testz funtion. But its really strange, because i flashed the previosuly installed Klipper and its reading Temps well. Just the new configuration is doing really really badly which is confusing me.

Coffee0297, 2020-02-25:
the E3 has issues with temp mesurement, a bad board design. look here https://github.com/KevinOConnor/klipper/issues/2194
i guess kevin are trying to fix it if he can find the issue.

-Tonny

l-Kage-l, 2020-02-25:
I've seen the thread but i dont think that explains how Klipper works with the previous version but doesnt with the newest one.

BlackStump, 2020-02-25:
After the fix to this problem https://github.com/KevinOConnor/klipper/issues/2194 ambient temp reads high, worse then it did prior, I thought it was fine before the fix.
My observations with the mini e3 is the hot bed especially seems to be read in steps, I believe it heats normally just the updates are stepped, you notice it on both heatup and cool down.
ie it heats normally the readings stop for a time then the temperature climbs a lot faster then pauses again then does the same faster climb in temp rinse repeat. It reaches the set temp fine.

l-Kage-l, 2020-02-25:
Yeah, that might be good for reaching the set temps, but with the updated version it still doesnt go below 33 degrees, as stated above. Its something with the Klipper software since it clearly worked on the previous version. I added the working and NOT working klipper versions for the SKR E3 Mini V1.3 over at #2194

KevinOConnor, 2020-03-02:
There was a recent fix (commit 31b2c3ea) to the stm32f1 code.  I recommend you update to the latest code and retry (see the FAQ for upgrade instructions).

-Kevin

l-Kage-l, 2020-03-02:
Yeah, I know. Thanks.

closed 2020-03-02 by l-Kage-l