Klipper3d/klipper #2562: Printer not ready

RaoulFabisch, 2020-02-25:
Hi,

I am getting the printer not ready error and don't know how to find out the reason. maybe you can find out in the log.,
[klippy.log](https://github.com/KevinOConnor/klipper/files/4250649/klippy.log)

StrikeEagleCC, 2020-02-25:
Did you happen to change the baud rate when you did `make menuconfig`? Have you verified your serial port is correct?

KevinOConnor, 2020-03-05:
Typically these types of connection issues are due to an incorrect configuration during "make menuconfig", or an unsuccessful flash.  My best advice is to rerun "make menuconfig", be sure to leave the "low-level configuration options" not selected, double check the values that are used, and then reflash the board.

-Kevin

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like the conversation has concluded.

-Kevin

closed 2020-03-21 by KevinOConnor