Klipper3d/klipper #2563: manual_stepper diag_pin for sensorless homing

kakou-fr, 2020-02-25:
Hi

I put a TMC2209 for a manual stepper driver (on a second MCU named mmboard), but I cannot find the good way to use sensorless homing, I tried (with a lot of other thing) : 

`[manual_stepper idler_stepper]
step_pin: mmboard:PB13
dir_pin: mmboard:PB14
enable_pin: !mmboard:PB12
step_distance: .040000
velocity: 100
accel: 80
#endstop_pin: !mmboard:PC3 # IR extruder endstop Z+
endstop_pin: tmc2209_idler_stepper:virtual_endstop
homing_retract_dist: 0

[tmc2209 manual_stepper idler_stepper]
uart_pin: mmboard:PB7
microsteps: 16
run_current: 0.800
hold_current: 0.800
interpolate: True
sense_resistor: 0.110
stealthchop_threshold: 250
diag_pin: ^mmboard:PC1
driver_SGTHRS: 150`

KevinOConnor, 2020-02-27:
As far as I know it would be configured as above - if it didn't, you'll need to attach the log.

-Kevin

kakou-fr, 2020-02-27:
Just added a klippy.log with some test

the error is always : 
error: Invalid pin description

KevinOConnor, 2020-02-27:
A config section that defines a new pin needs to be in the config file above the first use of that pin.  (So, move the tmc2209 config section above the corresponding manual_stepper config section.)

-Kevin

kakou-fr, 2020-02-29:
ok the config is loaded if I put the tmc2209 section before manual_stepper.
But, the manual stpeper don't stop when using STOP_ON_ENDSTOP.
Looks like the MANUAL_STEPPER STOP_ON_ENDSTOP don't support sensorless homing ?

KevinOConnor, 2020-02-29:
My understanding is that sensorless homing is finicky - you should read through the docs at https://www.klipper3d.org/Sensorless_Homing.html to tune the settings.  As far as I know, the code works as intended, but it may take a lot of tuning for the drivers to detect an event reliably (if possible at all).

-Kevin

kakou-fr, 2020-02-29:
ok i will try again. I already use it on 2 other printer for X/Y with no issue. I don't understantd why it is not working with manual_stepper

kakou-fr, 2020-03-01:
If I define : 
driver_SGTHRS: 250

When I test : 
SG_RESULT:  00000082 SG_RESULT=130

SG_RESULT is enought to trigger the endstop, but nothin is triggered (or the trigger state is not verified for manual_stepper)

Arksine, 2020-03-01:
If you haven't already, make sure that your board supports sensorless homing, and if so make sure that the diag pins are correctly jumpered.

kakou-fr, 2020-03-01:
I tried to connect the pin on X+/Y+ and then on 2 pins on the EXP1.
The board is a SKR MINI V1.1.
Uart iis connected to EXP1 (and works)

KevinOConnor, 2020-03-02:
I'm not aware of any issues with the sensorless homing code.  I'd triple check pins and routing.

-Kevin

kakou-fr, 2020-03-02:
ok i will write a sample printer.cfg to only test these stepper with stepper_x/y in place of manual_stepper

kakou-fr, 2020-03-02:
I tried with this config, and I can sensorless home XY :
`# Y : MMU2S idler with the 5 ball bearings
[tmc2209 stepper_x]
uart_pin: mmboard:PB7
microsteps: 16
run_current: 0.800
hold_current: 0.800
interpolate: True
sense_resistor: 0.110
stealthchop_threshold: 100
diag_pin: ^mmboard:PC13 # Y+
driver_SGTHRS: 75

[stepper_x]
step_pin: mmboard:PB13
dir_pin: mmboard:PB14
enable_pin: !mmboard:PB12
step_distance: .040000
position_endstop: 0
position_min: 0
position_max: 100
#endstop_pin: !mmboard:PC3 # IR extruder endstop Z+
endstop_pin: tmc2209_stepper_x:virtual_endstop
homing_retract_dist: 0
homing_speed: 100

# X : MMU2S color selector
[tmc2209 stepper_y]
uart_pin: mmboard:PC12
microsteps: 16
run_current: 1.000
hold_current: 0.400
interpolate: True
sense_resistor: 0.110
stealthchop_threshold: 250
diag_pin: ^mmboard:PB6 # X+
driver_SGTHRS: 150

[stepper_y]
step_pin: mmboard:PC6
dir_pin: mmboard:PC7
enable_pin: !mmboard:PB15
# 80 step/mm
step_distance: .0025
#endstop_pin: !mmboard:PC0 # switch endstop on the left Z-
endstop_pin: tmc2209_stepper_y:virtual_endstop
homing_retract_dist: 0
position_endstop: 0
position_min: 0
position_max: 100
homing_speed: 35`

With this config : 
`
# Y : MMU2S idler with the 5 ball bearings
[tmc2209 manual_stepper idler_stepper]
uart_pin: mmboard:PB7
microsteps: 16
run_current: 0.800
hold_current: 0.800
interpolate: True
sense_resistor: 0.110
stealthchop_threshold: 100
diag_pin: ^mmboard:PC13 # Y+
driver_SGTHRS: 75

[manual_stepper idler_stepper]
step_pin: mmboard:PB13
dir_pin: mmboard:PB14
enable_pin: !mmboard:PB12
step_distance: .040000
velocity: 100
accel: 500
#endstop_pin: !mmboard:PC3 # IR extruder endstop Z+
endstop_pin: tmc2209_idler_stepper:virtual_endstop
homing_retract_dist: 0

# X : MMU2S color selector
[tmc2209 manual_stepper selector_stepper]
uart_pin: mmboard:PC12
microsteps: 16
run_current: 1.000
hold_current: 0.400
interpolate: True
sense_resistor: 0.110
stealthchop_threshold: 250
diag_pin: ^mmboard:PB6 # X+
driver_SGTHRS: 150

[manual_stepper selector_stepper]
step_pin: mmboard:PC6
dir_pin: mmboard:PC7
enable_pin: !mmboard:PB15
# 80 step/mm
step_distance: .0025
velocity: 35
accel: 500
#endstop_pin: !mmboard:PC0 # switch endstop on the left Z-
endstop_pin: tmc2209_selector_stepper:virtual_endstop
homing_retract_dist: 0
`
and this command, homing doesn't work (I try different value of driver_SGTHRS) : 
`[...]
Send: MANUAL_STEPPER STEPPER=idler_stepper MOVE=70
[...]
Send: MANUAL_STEPPER STEPPER=idler_stepper MOVE=-95 STOP_ON_ENDSTOP=1
[...]
Recv: !! Failed to home manual_stepper idler_stepper: Timeout during homing
[...]`

KevinOConnor, 2020-03-03:
On a closer look, it appears the recent commit feb15eaf broke sensorless homing for manual_stepper.

-Kevin

kakou-fr, 2020-03-03:
looks like it works with this patch

`79a80,83
>         # Notify start of homing/probing move
>         endstops = self.rail.get_endstops()
>         self.printer.send_event("homing:homing_move_begin",
>                                 [es for es, name in endstops])
96a101,107
>         # Signal homing/probing move complete
>         try:
>             self.printer.send_event("homing:homing_move_end",
>                                     [es for es, name in endstops])
>         except CommandError as e:
>             if error is None:
>                 error = str(e)`

KevinOConnor, 2020-03-21:
This should be fixed as of commit 961f6862.

-Kevin

kakou-fr, 2020-03-21:
yes thks

closed 2020-03-21 by kakou-fr