Klipper3d/klipper #2567: MCU 'mcu' shutdown: ADC out of range then heating

Anorebo, 2020-02-27:
Hi! I configure Klipper on a bundle of arduino Mega2560 and Ruramps4d. When trying to calibrate the PID or just heating the hotend and bed, I get the error: "MCU ' mcu' shutdown: ADC out of range". The error always occurs at the same values of the heater temperature, hotend approximately 246 degrees, bed - 75. How can I understand the reason for the shutdown?

[klippy.zip](https://github.com/KevinOConnor/klipper/files/4259482/klippy.zip)

ld3300, 2020-02-28:
What happens  if you remove the verify heater sections?

Anorebo, 2020-02-28:
Hi! I started without this section and got an error, then tried to fix it using the parameters of this section

ld3300, 2020-02-28:
you may have broken code.  Definitely try a reinstall to be up to date with the master branch

Anorebo, 2020-02-29:
Hi, David! I understand what the problem was. My story is this: I use the popular in Russia motherboard Ruramps4d. but it is designed to work with arduino Due and reference voltage for analog pins 3.3 V. unfortunately, I only have non-original Due, which iimeyut problems with running the firmware.  So to study the Klipper, I connected the arduino Mega, which has a reference voltage of 5V, which led to an error in the temperature measurement. When I realized this, I started looking for a way to tell Klipper the correct reference voltage, tried using the adc_voltage parameter, it seems, but it didn't work for my type of thermistors (EPCOS 100K B57560G104F). I had to change thermistor.py adding the correction factor adc = 5.0*adc / 3.3 to the calc_temp function. After that, the temperature of the heaters was calculated correctly. But there is a problem that I described earlier. Looking at the code more closely, I realized that my correction is not enough and I still need to add a coefficient to the temperature conversion function  calc_adc (self, temp) r=3.3*r/5.0. I did it, and the Klipper worked as it should.
I don't like this solution, but I like Klipper, and it allowed me to continue studying on my hardware. But it would be nice if you could specify adc_voltage for any type of sensors? Thank you, sorry for my English

ld3300, 2020-03-01:
ahh, well, you are smarter tan me.  I am glad you got it working,  It sounds like there may need to be a new boards file made or something, as there are other 3,3v hosts in kliper.

jalanjarosz, 2020-03-01:
David,
The issue is a combination of shield and MCU. I had the same issue, but in reverse, with a 3v3 volt Adafruit Grand Central and a modified RAMPS v1.6

When you have the MCU Vcc at one voltage and the thermostat ref voltage is different, then you have temperature issues.

My case Vcc was 3v3 and Thermistor ref voltage was 5v.  Sounds like with the RURamps his Vcc is 5v for the MEGA, but the thermistor ref voltage is 3v3.  the RURamps schematic would need to be checked to confirm.

Anorebo, check the schematic of the RURamps to see if the Thermistor rev voltage is being pulled from the 3v3 leg of the MCU. If it does, then the same issue would exist in other FW as it does in Klipper. I’d guess the software patch you created is the way to go.

James

KevinOConnor, 2020-03-02:
The thermistor measuring code does not depend on voltage - it measures the relative resistance of the thermistor as compared to the resistance of the pullup.  Said another way, the voltage "drops out of the equations".

If you are indeed mixing 3.3v and 5v in some way, then I'd recommend against that - it is likely to permanently damage one of the chips.

-Kevin

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like a workaround was found for the original issue and the conversation has concluded.

-Kevin

closed 2020-03-21 by KevinOConnor