Klipper3d/klipper #2568: bed_mesh samples

boboxx, 2020-02-27:
I just setup a capacitive proximity Switch (LJC18A3-B-Z/BX NPN NO) on a Ender3. I copied from one of the sample config but samples seems to be ignored and I only get one sample per point. Am I missing something?

this is my definition:
[bed_mesh]
speed: 50
horizontal_move_z: 5 #height (in mm) head should move prior to starting probing
samples: 3
mesh_min: 50,30
mesh_max: 215,215
probe_count: 4,4
fade_start: 1.0
fade_end: 10.0

[klipper.log](https://github.com/KevinOConnor/klipper/files/4262737/klipper.log)

Arksine, 2020-02-27:
The `samples` option should go in the [probe] section.  Alternatively you can specify the samples in the call to BED_MESH_CALIBRATE: 

`BED_MESH CALIBRATE SAMPLES=3`

boboxx, 2020-03-01:
Thank you Eric, that solved the problem :)

closed 2020-03-01 by boboxx