Klipper3d/klipper #2569: Unusual Firmware issues - Constant extruder retraction

Leviathan220, 2020-02-27:
[klippy.log.copy.txt](https://github.com/KevinOConnor/klipper/files/4264524/klippy.log.copy.txt)

I'm not sure what to call it. I cleanly reinstalled a completely fresh system to try to get rid of the issue. I tried turning on pressure_advance and running the test print. Ever since then my printer has been stuck in that mode. I have reinstalled octoprint, I have tried raspbian with DWC2 just in case octoprint was the issue. I have reflashed the klipper firmware to my printer. The extruder will start trying to retract after the first layer. It doesn't matter what slicer profile I use. It doesn't matter what gcode file I print with. Even ones I have printed before with zero issue. Klipper doesn't seem to want to go back to the way it was. The only thing I have left to do is try swapping to the Marlin Firmware to see if it's just the firmware isn't getting flashed properly or something.

ld3300, 2020-02-28:
Try installing a clean version without DWC2.  You are unlikely to get help with a modified version of the firmware as no one will be able to tell what the changes are.

Leviathan220, 2020-02-28:
I already stated in my comment that I did in fact install a clean version without the DWC2 I only installed it to try both in case one was the issue. Neither a clean install nor the DWC2 solved this issue. I am currently trying to test Marlin. Perhaps the firmware didn't actually flash as it says it did. The arduino IDE should tell me for certain.

Leviathan220, 2020-02-28:
My klipper.log was uploaded from one of the attempts. It might have been from the DWC2 install idk. It doesn't matter the issue persists with a clean install I've flashed back and forth about 4 times now trying 2 with the DWC2 and 2 with octoprint. Nothing solved the issue. I would also like to add that I completely reflashed the SD card each time so nothing would carry over between tests.

ld3300, 2020-02-28:
can you upload a log with a clean install?

KevinOConnor, 2020-02-29:
This sounds like a hardware issue.  You should rerun the steps described at: https://www.klipper3d.org/Config_checks.html

FYI, the only state Klipper saves is in the config file.

-Kevin

Leviathan220, 2020-03-04:
I've swapped to marlin atm. I doubt whatever issue I was experiencing would likely happen again if I switch back I will close this issue for now. If I come across anything else I'll let you know.

closed 2020-03-04 by Leviathan220