Klipper3d/klipper #2570: FYSETC Cheetah v1.1 Timeout on serial connect

fernandocc, 2020-02-27:
Similar issue as other FYSETC Cheetah owners, used the generic-fysetc-cheetah-v1.1.cfg and followed the firmware compile instructions as described.

Octoprint is trying to connect to /tmp/printer configured as described in the instructions.

Any suggestions, saw some previous tickets with similar issue but were closed without clear resolutions.

Thanks

[klippy.log](https://github.com/KevinOConnor/klipper/files/4264629/klippy.log)

KevinOConnor, 2020-02-29:
There was an error just fixed (commit 31b2c3ea) - you should grab the latest code, recompile the micro-controller code, and retry (see the FAQ for full upgrade instructions).

-Kevin

fernandocc, 2020-03-03:
Thanks, pulled the latest code and recompiled, but no luck, same errors as in the log provided.
Using UART1 in the menu config options just in case that matters.

fernandocc, 2020-03-07:
Got it figured out, the cheetah board backfeeds 5V on the USB VCC pin, put some tape on it.

closed 2020-03-07 by fernandocc