Klipper3d/klipper #2571: Default Menu does not derive bed size for Control movement

raiscan, 2020-02-28:
Hi! Currently, Klippy's default menu structure does not take stepper_x, stepper_y and stepper_z's "position_max" and "position_min" into account, resulting in insufficient control on beds larger than the defaults and the head crashing on beds that are smaller.

I appreciate this can be changed by the end user altering the menu cfg but surely these fields can be used to derive the control menu's bounds?

Thanks for all the hard work!

ld3300, 2020-02-28:
Have you had instances with the head crashing?  The menu is calling the G1 gcode and should respond the same way as it would if entered from the command line.  It would be concerning if it is causing G1 to react differently.  

I think the menu awareness could be a nice feature, but would require some menu reworking.  I've been playing around with the menu, I will try to figure it out, but it will likely be over my head.

KevinOConnor, 2020-02-29:
This looks like a dup of #1856 .

-Kevin

KevinOConnor, 2020-03-11:
I'm going to close this in preference to the existing #1856 .

-Kevin

closed 2020-03-11 by KevinOConnor