Klipper3d/klipper #2572: Dual Z Home

Klinenator, 2020-02-29:
Any chance of a dual z home option like prusa has?

vladbabii, 2020-02-29:
If you're talking about dual endstops, one for each Z, i belive you just need to set an endstop_pin for each z stepper motor instead of just the first one.

kakou-fr, 2020-02-29:
auto-adjust dual Z with probe ?? 

[stepper_z]
step_pin: P2.3
dir_pin: P0.22
enable_pin: !P0.21
step_distance: .00125
endstop_pin: probe:z_virtual_endstop
position_min: -3
position_max: 230
homing_speed: 5

[stepper_z1]
step_pin: P2.0
dir_pin: !P0.5
enable_pin: !P0.4
step_distance: .00125

[z_tilt]
z_positions:
    -80,20
    295,20
points:
    -20,110
    110,110
horizontal_move_z: 10
retries: 5
retry_tolerance: 0.05

vladbabii, 2020-03-03:
Sorry, you were pretty vague in your description. z_tilt should be used to sync the two z steppers. If you have issues with the steppers getting out of sync you might want to add anti-backlash nuts to your z screws. They help quite a bit with this.

kakou-fr, 2020-03-06:
z_tilt is useful when you work on your printer and you desync z or when the weight of the extruder desync z when the printer is off

Klinenator, 2020-03-10:
I would like something like

[stepper_z]
step_pin: P2.3
dir_pin: P0.22
enable_pin: !P0.21
step_distance: .00125
endstop_pin: probe:z_virtual_endstop
position_min: -3
position_max: 230
homing_speed: 5

[stepper_z1]
step_pin: P2.0
dir_pin: !P0.5
enable_pin: !P0.4
step_distance: .00125
endstop_pin: probe:z1_virtual_endstop

so i have two different z endstops and z homes to the top until its own endstop is triggered

vladbabii, 2020-03-10:
You either have 2 end stops or one probe. I don't think you can have two probes and i can't imagine how a probe could probe 2 different points at the same time..

Klinenator, 2020-03-10:
Sorry I have two end stops

KevinOConnor, 2020-03-21:
Klipper supports homing of multiple Z stepper motors.  If you have two endstops, then specify the endstop pins in the stepper_z and stepper_z1 config sections.

-Kevin

swatti87, 2020-03-22:
Slightly related. A question: Can i home both X and Y at the same time somehow? I have sensorless homing drivers but for that to work reliably i need to home a biiit slow, so running both at the same time would be sweet.

closed 2020-04-08 by KevinOConnor