Klipper3d/klipper #2573: Feature request: Run a gcode command and store output in a macro variable or some internal location

vladbabii, 2020-02-29:
I want to run a gcode command and store output in a macro variable or some internal location which i can parse with other macros to automate things.

Use case examples:  
* delayed timer running TMC_DUMP to check driver status and do some things based on that (like stopping/pausing print when a stepper issue is detected). I could then use the jinja templates to parse the output string and do something meaningfull with the information
* before printing run a self_check of all TMC drivers and see if there are any issues 

Thank you

KevinOConnor, 2020-02-29:
See https://www.klipper3d.org/Command_Templates.html#variables

-Kevin

ld3300, 2020-03-01:
To add to Kevin's notes you can use the same `SET_GCODE_VARIABLE MACRO=start_probe VARIABLE=bed_temp VALUE=123xxx` command from within other macros besides the one that contains the variable itself (in this case macro start_probe).  You can even run the SET_GCODE_VARIABLE command directly from the command line or a g-code file.  I also use the variables in if / else logic statements quite often to make decisions, like which extruder has filament available, etc.

vladbabii, 2020-03-02:
I know how to set a variable with SET_GCODE_VARIABLE.

Let's say i have a random gcode command like PROBE_ACCURACY

This outputs on the serial connection something like
```
Recv: // PROBE_ACCURACY at X:110.000 Y:110.000 Z:4.900 (samples=10 retract=2.000 speed=5.0
Recv: // probe at 110.000,110.000 is z=0.327500
Recv: // probe at 110.000,110.000 is z=0.430000
Send: M105
Recv: // probe at 110.000,110.000 is z=0.425000
Recv: // probe at 110.000,110.000 is z=0.442500
Recv: // probe at 110.000,110.000 is z=0.430000
Recv: // probe at 110.000,110.000 is z=0.437500
Recv: // probe at 110.000,110.000 is z=0.462500
Recv: // probe at 110.000,110.000 is z=0.417500
Recv: // probe at 110.000,110.000 is z=0.415000
Recv: // probe at 110.000,110.000 is z=0.445000
Recv: // probe accuracy results: maximum 0.462500, minimum 0.327500, range 0.135000, average 0.423250, median 0.430000, standard deviation 0.034570
```

I would write to write a macro that runs PROBE_ACCURACY, gets the full output then parses some values and outputs some data based on the values is read. This way i would be able to read ```maximum 0.462500, minimum 0.327500, range 0.135000``` and give the user a feedback that the probe is good quality, bad quality or unusable.

Or i could parse the mesh output and run SAVE_CONFIG automatically or tell the user that the front-left corner is too high compared to the rest, use the bed leveling screws to fix it and re-run the command...

I don't think I have any way to do this now.

My example with TMC_DUMP was to have a macro that
1. runs TMC_DUMP on all steppers, one by one (doable now with jinja)
2. get output of each command (this is what i need)
3. find the status line (doable with junja, afaik)
4. if there are any issues with any stepper then run a different macro (doable now)

Maybe an example of the desired command would be better :
```
SET_GCODE_VARIABLE MACRO=<macro_name> VARIABLE=<name> MACROTORUN=<name of macro>
```

for example to get output of G28 i would do
```
SET_GCODE_VARIABLE MACRO=storage_macro VARIABLE=homing_output MACROTORUN=HOME_ALL<name of macro>
```

and the macro would be defined as
```
[gcode_macro HOME_ALL]
gcode:
  G28 X0 Y0
  G28 Z0
```

KevinOConnor, 2020-03-03:
There is currently no ability to capture the human readable responses from a g-code command.  FWIW, my high-level advice would be to write a Python "extras" module instead of using Jinja2 to accomplish your goals.

-Kevin

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like the conversation has concluded.  I suspect a Python "extras" module is the place to implement more detailed tools.

-Kevin

closed 2020-03-21 by KevinOConnor