Klipper3d/klipper #2574: Confused, ignorant, or just plain stupid?

Primer-Merc, 2020-03-01:
Howdy all, I'm either one or all of these three and I almost have no hair left to pull out. I don't know if this is a problem or a configuration issue. I've been doing lots and lots of reading on Klipper setup and I would rather not bother anyone with this , but I am so close to having this fully setup I had to.

My Hardware : 
Cartesian Printer -Artillery Genius
Slicer = Cura 4.5
Raspberry Pi 4  --> MKS Gen L V1.0
3D Touch = BL Clone

My environment:
Z-Min endstop used for homing Z (factory wired)
Z-Max endstop used for probing Z (factory wired)
3D Touch = +2.8mm when the pin is stowed

My print start G-code:
G28 ; Home the axes
G90 ; Use absolute coordinates
G92 E0
G29 ; bed mesh level via the macro in the printer.cfg
G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
G1 X2 Y20 Z0.3 F5000.0 ; move to start-line position
G1 X2 Y200.0 Z0.3 F1500.0 E15 ; draw 1st line
G1 X2 Y200.0 Z0.4 F5000.0 ; move to side a little
G1 X2 Y20 Z0.4 F1500.0 E30 ; draw 2nd line
G92 E0 ; reset extruder
G1 Z1.0 F3000 ; move z up little to prevent scratching of surface

My problem. 

When the printer starts printing, the nozzle is basically sitting on the bed. Little to no material is being extruded. I have performed all the bed leveling steps including the paper method. After I do the paper test, and move the tool down 5.5 mm the z offset in the printer.cfg is z_offset = 0.700 . So I'm so confused why the nozzle is so close to the bed when the print starts, with a .700 offset. The only way to get a decent print is to add a z-offset to the slicer. I'm sure either I have something configured wrong or I don't understand the behavior. Or...I'm just stupid. 

You may see a "Extrude only move too long" error in the log. This can be ignored,  G92 E0 was accidentally removed from my start up and the problem above started before it was removed. 

[klippy.zip](https://github.com/KevinOConnor/klipper/files/4271279/klippy.zip)

BTW I'm really enjoying how this setup is compared to Marlin. 99% of the documentation is spot on from a " I understand what this means" point of view.  I appreciate all the work that goes into to this even if my issue can't be resolved within Klipper. 

Thanks again! Kyle

ld3300, 2020-03-01:
The Z offset doesn't go in the BLtouch section.  That offset should be the height distance between the probe when triggered and the nozzle.

You should use `SET_GCODE_OFFSET Z=` in your config file to tune your offset.

Your config is telling the printer that your z-endstop is at 0.00, so the offset isn't being applied.  Setup your BLtouch to be the z endstop using the instructions in this section:  [probe](https://github.com/KevinOConnor/klipper/blob/31b2c3ea7a21a45c975bc6847c39b82d3c458af2/config/example-extras.cfg#L557)

Primer-Merc, 2020-03-01:
Hi David,

Thanks for the response. Just so I'm clear, "I" / Klipper cannot use the factory Z endstop for homing and the BL-Touch for leveling? The reason I wanted this functionality was just in case the BLTouch ever failed I would have a backup to stop the nozzle from crashing. 

The command SET_GCODE_OFFSET Z= , Should that be under a specific section?

While doing more research I did find this.   https://www.reddit.com/r/klippers/comments/d4pwg5/how_do_zendstop_and_zprobe_together_affect_the/

 #relative_reference_index:
#   A point index in the mesh to reference all z values to. Enabling
#   this parameter produces a mesh relative to the probed z position
#   at the provided index.

I'm not sure how to use it though. This is where my ignorance comes into play :-(  I'm hoping someone here has an example of what needs to be in this field. Also gonna try to ask the fella on reddit who seems to have it working. 

Kyle

Primer-Merc, 2020-03-01:
Hey David,

Thank you again for the help ! Your (SET_GCODE_OFFSET Z=) suggestion led me to that reddit page and after some thought, I entered this into my printer.cfg   relative_reference_index: 4   .
And that's all it took. Because I use a 3x3 mesh and home to the center of the bed, the #4 point on the mesh is the value to use. I just finished up a small print and it looks great, and I still need to setup pressure advance so I guess it can only get better ( knocks on wood). 

Again thanks for the help!

Kyle

closed 2020-03-01 by Primer-Merc