Klipper3d/klipper #2577: Skr mini e3 sensorless homing

ice2009, 2020-03-01:
Hello :) 

I install new board for my ender 3 (skr mini e3 v1.2 with tmc2209) Does anybody have printer.cfg file with sensorless homing ? Can you share please?

Thank's

Magoo0876, 2020-03-01:
Use these lines instead of the ones from the standard config. Be careful because the printer is highly modified and positions do not fit a standard Ender 3.

[stepper_x]
step_pin: PB13
dir_pin: !PB12
enable_pin: !PB14
step_distance: .003125
endstop_pin: tmc2209_stepper_x:virtual_endstop
position_endstop: 0
position_max: 250
homing_speed: 50
homing_retract_dist: 0

[tmc2209 stepper_x]
uart_pin: PB15
diag_pin: PC0
microsteps: 64
run_current: 0.90
hold_current: 0.5
stealthchop_threshold: 250
driver_SGTHRS: 120

[stepper_y]
step_pin: PB10
dir_pin: !PB2
enable_pin: !PB11
step_distance: .003125
endstop_pin: tmc2209_stepper_y:virtual_endstop
#endstop_pin: ^PC1
position_endstop: 0
position_max: 290
homing_speed: 50
homing_retract_dist: 0

[tmc2209 stepper_y]
uart_pin: PC6
diag_pin: PC1
microsteps: 64
run_current: 1.0
hold_current: 0.500
stealthchop_threshold: 250
driver_SGTHRS: 120
#255 is most sensitive

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like the original question was answered.

-Kevin

closed 2020-03-21 by KevinOConnor